realsenseai/librealsense #12839: Segment tree (foreground) from background with RGB and **complicated** depth data

LopezJER, 2024-04-08:
Hi, 

We have collected RGB and depth images of banana plant specimens using Realsense D435i. Now, we're aiming to create a program that can segment the foreground tree specimen from the background. We figured that since the depth image provides distances of objects within the image, we can set a threshold distance to mask the background, effectively isolating the plant specimens closer to the camera.

However, our actual captured depth images depict highly variable depths within the plant (see below):

Sample 1:

[image]

Sample 2:

[image]

Sample 3:

[image]

For context:
In the field, we have used Android phones to interface with the Realsense camera via the Android OS build of RealSense SDK 2.0. We have captured the RGB image, raw 16-bit depth data, grayscale 8-bit depth data without performing any alignment. We performed alignment of the raw depth data to the RGB image as a post-processing step, using code adapted from @gwen2018 in https://github.com/IntelRealSense/librealsense/issues/12020#issuecomment-1675211260. From our side, the alignment appears successful, but the tremendously varying depths make the task seem intractable.

Here are the relevant files:
Our depth-to-color alignment code:
[align.py.txt](https://github.com/IntelRealSense/librealsense/files/14904283/align.py.txt)

Sample specimen frames (includes color frame in PNG and NPY, raw depth frame in BIN and NPY, and 8-bit grayscale depth frame in PNG):

Specimen 1: 
[specimen1.zip](https://github.com/IntelRealSense/librealsense/files/14904297/specimen1.zip)

Specimen 2:
[specimen2.zip](https://github.com/IntelRealSense/librealsense/files/14904303/specimen2.zip)

Specimen 3:
[specimen3.zip](https://github.com/IntelRealSense/librealsense/files/14904304/specimen3.zip)

### **Given these observations and challenges, we have two main questions:**

**1. Is it feasible to develop a general-purpose application for foreground segmentation with the kind of images we have provided? If so, could you suggest an approach or reference examples?**

**2. If our current approach isn't viable, could you recommend alternative methods for achieving accurate segmentation for our deep learning preprocessing needs? I'm open to anything!**

This program is intended to be part of the preprocessing pipeline for deep learning, where our model is trained on the segmented plant specimens instead of the raw PNG image.

MartyG-RealSense, 2024-04-08:
Hi @LopezJER  The depth images look quite good aside from the incorrect depth coloring.  Comparing the images, I suspect that there is a direct relationship between accuracy and the amount of sunlight in the scene, with accuracy worsening as brightness increases.

You could try maximizing the **Laser Power** value with `rs.option.laser_power` by setting it to its maximum value of '360'.  Doing so should make the camera's dot pattern projection more visible to the camera and aid depth analysis, though the pattern may still be weakly visible in such bright light.  An example of setting Laser Power with Python code is at https://github.com/IntelRealSense/librealsense/issues/8592#issuecomment-800257572

An alternative if you have the project budget to do so may be to consider using the **D435if** camera model, which is a D435i equipped with light-blocking filters for improved depth quality in scenarios where there are light reflections present (leaves are light-reflective).

https://www.intelrealsense.com/depth-camera-d435if/

A cheaper way to test whether the problem is with light reflection from the leaves would be to obtain an inexpensive **linear polarizing filter** product and place it over the lenses of your D435i, as described at https://github.com/IntelRealSense/librealsense/issues/10353#issuecomment-1083391059

LopezJER, 2024-04-08:
Cool. Thanks for the laser_power suggestion @MartyG-RealSense. I hadn't considered the light reflectiveness as a factor at this stage of the project. I'll implement it when I get back to work, and let you know the result.

I'll also let my team know about the proposed alternative camera model and polarizing filter to suit our use case.

MartyG-RealSense, 2024-04-19:
You are very welcome, @LopezJER - thanks very much for the update.  As you suggest, I will close the issue.  Thanks again!

closed 2024-04-19 by MartyG-RealSense