realsenseai/librealsense #12840: how to enable firmware logs through c++?

eilat-inbal, 2024-04-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.15.00.02 | 
| Operating System & Version |   {Win (10)   | 
| Platform                                 | PC  |
| SDK Version                            |  2 }                          | 
| Language                            |  {C++ }                          | 

### Issue Description
i want to enable FW logs (in order to fix a no frames received error).
this works when i use the real sense viewer through this button on the buttom right:
[image]

how can I do the same thing through c++ code?

MartyG-RealSense, 2024-04-08:
Hi @eilat-inbal  A RealSense user who also experienced No Frames Received in the Viewer until firmware logging was enabled resolved their particular issue by using a USB 3.2 cable with their camera.  That case can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4402721245587/comments/9640524921491

If your camera is already detected in the Viewer as USB 3.2 and you are still experiencing the problem, you may be able to retrieve the log messages with the **RS2_CAMERA_INFO_DEBUG_OP_CODE** instruction.  The RealSense SDK does not have an equivalent of the Viewer's firmware logging activation button, though a C++ tutorial called 'FluentRealSense' has an example of code for printing the debug code.

```
const char* debug_opCode() const
{
return get_info(RS2_CAMERA_INFO_DEBUG_OP_CODE);
}
```

https://www.codeproject.com/Articles/1233892/FluentRealSense-The-First-Steps-to-a-Simpler-RealS

eilat-inbal, 2024-04-09:
thank you @MartyG-RealSense for your very quick response. unfortunately, I'm not able to use the USB 3.2 cable....
does the RealSense viewer not use the SDK to communicate with the camera?
what does this button of "enable FW logs" do? and why does it solve the no frames received error?
I tried calling it once after starting the camera, and it did not work. I'm still getting the same error.
do I have to call it more than once? or maybe in a different timing?
thanks!
Inbal

MartyG-RealSense, 2024-04-24:
Case closed due to no further comments received.

closed 2024-04-24 by MartyG-RealSense

tothambrus11, 2024-10-16:
Here is the section of realsense-viewer that is run when the firmware logs are enabled: https://github.com/IntelRealSense/librealsense/blob/master/common/output-model.cpp#L30-L103

tothambrus11, 2024-10-17:
This is how you can enable the firmware logs in C++. I don't know if this busy-looping will be an issue or we should suspend the thread for some time to get the logs again, but this seems to work well. 

This is just the result of me stripping down what the realsense-viewer was doing, so there might be a better approach.

```cpp
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2019-24 Intel Corporation. All Rights Reserved.

#include <exception>
#include <fmt/core.h>
#include <iosfwd>
#include <iostream>
#include <librealsense2/hpp/rs_frame.hpp>
#include <librealsense2/hpp/rs_internal.hpp>
#include <librealsense2/hpp/rs_pipeline.hpp>
#include <librealsense2/hpp/rs_types.hpp>
#include <thread>

std::string nice_string(std::vector<uint8_t> data)
{
  std::stringstream stringStream;
  for (auto &elem : data) {
    stringStream << std::setfill('0') << std::setw(2) << std::hex << static_cast<int>(elem) << " ";
  }
  return stringStream.str();
}
void print_firmware_logs(rs2::device dev)
{

  auto logger = dev.as<rs2::firmware_logger>();
  if (!logger) {
    fmt::print("Firmware logger is not supported on this device\n");
  }

  logger.start_collecting();
  fmt::print("FW Logger started collecting...\n");

  try {
    auto message = logger.create_message();
    while (logger.get_firmware_log(message)) {
      // fmt::print("{}\n", nice_string(message.data()));
    }
  } catch (const std::exception &ex) {
    fmt::print("Failed to fetch fw logs: %s\n", ex.what());
  }
}

// Hello RealSense example demonstrates the basics of connecting to a RealSense device
// and taking advantage of depth data
int main()
{
  try {
    // Create a Pipeline - this serves as a top-level API for streaming and processing frames
    rs2::pipeline pipeline{};

    // enable firmware logs

    // Configure and start the pipeline
    auto profile = pipeline.start();

    std::jthread const myThread([device = profile.get_device()]() { print_firmware_logs(device); });

    while (true) {
      // Logger and is initialized successfully (can parse logs)

      try {
        // Block program until frames arrive
        rs2::frameset const frames = pipeline.wait_for_frames(100);

        // Try to get a frame of a depth image
        rs2::depth_frame const depth = frames.get_depth_frame();

        // Get the depth frame's dimensions
        const auto width = depth.get_width();
        const auto height = depth.get_height();

        // Query the distance from the camera to the object in the center of the image
        const float dist_to_center = depth.get_distance(width / 2, height / 2);

        // Print the distance
        std::cout << "The camera is facing an object " << dist_to_center << " meters away \n";
      } catch (std::exception &e) {
        fmt::print("CAMERA ERROR: {}\n", e.what());
      }
    }
  } catch (const rs2::error &e) {
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    "
              << e.what() << '\n';
    return 1;
  } catch (const std::exception &e) {
    std::cerr << e.what() << '\n';
    return 1;
  }
}
```

MartyG-RealSense, 2024-10-17:
Thanks so much @tothambrus11 for sharing your solution!