realsenseai/librealsense #12841: RealSense Can't list camera, D435 for windows 11

yanite, 2024-04-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Windows 11   | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.54.2 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSense Can't list camera, D435 for windows 11

driver show:
Device USB\VID_8086&PID_0B07&MI_00\b&374ad5a0&1&0000 is configured.

Driver name: oem5.inf
Class GUID: {ca3e7ab9-b4c3-4ae6-8251-579ef933890f}
Driver Date: 03/22/2021
Driver version: 6.1.160.22
Driver Provider: Intel
Driver part: RS400_DEPTH.NTamd64
Driver level: 0xFF0001
Matching device ID: USB\VID_8086&PID_0B07&MI_00
Low-level driver: usbvideo.inf:USB\Class_0e:00FF2005
Device updated: true
Parent device: USB\VID_8086&PID_0B07\816513020237

Device USB\VID_8086&PID_0B07&MI_00\b&374ad5a0&1&0000 is having trouble booting.

Driver name: oem5.inf
Class GUID: {ca3e7ab9-b4c3-4ae6-8251-579ef933890f}
Service: usbvideo
Low-level filter: WdmCompanionFilter
High level screening process:
Problem: 0x0
Problem status: 0xC00000E5

windows event log

MartyG-RealSense, 2024-04-08:
Hi @yanite  Can you check the camera privacy settings in Windows please to ensure that permission for applications to access the camera is set to On. Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

yanite, 2024-04-08:
> 嗨，您能否检查Windows中的相机隐私设置，以确保应用程序访问相机的权限设置为开。可以在 [#3381 上](https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789)找到这样做的说明（评论）

Yes, I checked it carefully and it's correct

[image]

yanite, 2024-04-08:
Very strange, windows web can display the camera correctly and open it

MartyG-RealSense, 2024-04-09:
Does it make a difference if you download and run the **Intel.RealSense.Viewer.exe** file in 'Assets' file list of the SDK 'Releases' page to launch the 2.54.2 version of the Viewer if you have not used this standalone Viewer already, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

[image]

yanite, 2024-04-10:
> 如果您还没有使用过这个独立的查看器，请下载并运行 SDK“版本”页面的“资产”文件列表中的 **Intel.RealSense.Viewer.exe** 文件以启动 2.54.2 版本的查看器，这有什么不同吗？
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2
> 
> ![image](https://private-user-images.githubusercontent.com/41145062/320858734-d34c9437-4e79-40df-a2bb-d4eaf5d69f6e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI3NjY2MDQsIm5iZiI6MTcxMjc2NjMwNCwicGF0aCI6Ii80MTE0NTA2Mi8zMjA4NTg3MzQtZDM0Yzk0MzctNGU3OS00MGRmLWEyYmItZDRlYWY1ZDY5ZjZlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDEwVDE2MjUwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZiYWI5NDA5ZTA1YTA1ODgyZmY2OWVlNDMzZmU5M2M3NDY3OTYyZjlmMjk0ZWVmMGU5NDY3NTdjNDQxNzBlOWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.lrQJavpUGVfgxs1D9qlX_i6uifFISu8rLceB219--MY)
> 
> 我还注意到您提到了**驱动程序版本 6.1.160.22 Intel 2021/3/22**。这是实感通用 Windows 平台 （UWP） 驱动程序。此驱动程序已过时，无需安装在 Windows 计算机上即可在 Windows 中使用实感。安装它可能会在启动某些实感程序时导致错误。
> 
> https://www.intel.com/content/www/us/en/download/19236/intel-realsense-d400-series-universal-windows-platform-uwp-driver-for-windows-10.html
> 
> 如果已在 Windows 计算机上安装了 UWP 驱动程序，则以下链接中有有关删除它的说明。
> 
> https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323/comments/4431239847443

Yes, I removed the Intel driver, as standard, including the option to delete the checkout, the Microsoft driver I am currently using, and then downloaded the standalone Intel RealSense View, but he still doesn't show anything, is there any other way?

[image]

[image]

MartyG-RealSense, 2024-04-11:
Do you know what firmware driver version is installed in your camera, please?  If you do not know then you can find it using the tool **rs-enumerate-devices** in the Windows Command Prompt interface, launching it with the command below.

**rs-enumerate-devices -s**

yanite, 2024-04-12:
> 您知道您的相机中安装了什么固件驱动程序版本吗？如果您不知道，则可以使用 Windows 命令提示符界面中的 **rs-enumerate-devices** 工具找到它，并使用以下命令启动它。
> 
> **rs-枚举设备 -s**

I can't access it on Windows, so I can only use it on my other embedded board

firmware version 5.15.1
[image]

MartyG-RealSense, 2024-04-12:
Thank you.  Your firmware version **5.15.1.0** is the correct one for use with SDK 2.54.2.

As the camera works correctly on a Linux computer then the USB cable likely does not have a problem.

When you installed the RealSense SDK on Windows, did you use the **Intel.RealSense.SDK-WIN10** installer program (which is on the same file list as Intel.RealSense.Viewer.exe) or did you instead build the SDK from source code on Windows with CMake?

yanite, 2024-04-12:
> Intel.RealSense.SDK-WIN10

Thank you very much for your reply, I've tried both, and it doesn't work, 1. I downloaded Intel.RealSense.SDK-WIN10 from github and it's not ok, 2. I can't compile the code myself. Is there a way to flash back the old firmware?

MartyG-RealSense, 2024-04-12:
You could downgrade the firmware in the RealSense Viewer on your Linux computer.

1.  Visit the link below and download a **.bin** file for the firmware version that you want to downgrade to.

2.  Go the the Viewer's **More** option near the top of its options side-panel and select **Update Firmware** from its drop-down menu.

3.  A file selection window will pop up.  Navigate to the location that you downloaded the .bin file to and select the file.  The firmware downgrade will begin.

Please bear in mind that RealSense cameras only allow a maximum of 20 downgrades before downgrading is permanently locked and then you can only update to a firmware version newer than the one that is currently installed in the camera.

yanite, 2024-04-13:
downgrade fail  

 13/04 23:50:14,046 INFO [548176843120] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 13/04 23:50:16,049 INFO [548176843120] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.0/video4linux/video0
 13/04 23:50:16,050 INFO [548176843120] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.0/video4linux/video1
 13/04 23:50:16,050 INFO [548176843120] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.3/video4linux/video2
 13/04 23:50:16,067 INFO [548176843120] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 13/04 23:50:18,069 INFO [548176843120] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.0/video4linux/video0
 13/04 23:50:18,070 INFO [548176843120] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.0/video4linux/video1
 13/04 23:50:18,071 INFO [548176843120] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.3/video4linux/video2
 13/04 23:50:18,083 INFO [548176843120] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 13/04 23:50:20,085 INFO [548176843120] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.0/video4linux/video0
 13/04 23:50:20,086 INFO [548176843120] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.0/video4linux/video1
 13/04 23:50:20,086 INFO [548176843120] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.3/video4linux/video2
 13/04 23:50:20,114 INFO [548176843120] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 13/04 23:50:21,134 INFO [546861736304] (ds-device-common.cpp:70) entering to update state, device disconnect is expected
 13/04 23:50:21,174 ERROR [546861736304] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 13/04 23:50:21,236 WARNING [546861736304] (ds-device-common.cpp:97) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 13/04 23:50:21,510 INFO [546861736304] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 13/04 23:50:21,511 INFO [546861736304] (context.cpp:336) Found 3 RealSense devices (mask 0xfe)
 13/04 23:50:21,517 INFO [546861736304] (environment.cpp:130) Found 3 unreachable streams, 1089 extrinsics deleted
 13/04 23:50:21,519 ERROR [546861736304] (librealsense-exception.h:52) acquire: Cannot open '/dev/video11 Last Error: Device or resource busy
 13/04 23:50:21,538 WARNING [546861736304] (rs.cpp:312) null pointer passed for argument "device"
fish: Job 1, 'realsense-viewer' terminated by signal SIGSEGV (Address boundary error)

If there's a problem with this thing, where do I send it to fix it? In China

MartyG-RealSense, 2024-04-13:
The error **Failed to read busnum/devnum** is a very rare one that usually occurs when the camera is used with a low power single-board computing device whose USB port cannot provide sufficient power to the camera for it to operate.  What computer / computing device are you using Windows on, please?

If the camera still works normally on your Linux computer then that would mean that the camera product is not faulty.  For example, there may be an issue with the USB port on your Windows computer that prevents the camera from being detected by RealSense applications.

Intel do not offer a repair service for RealSense cameras.  If there is a verifiable fault with the camera then you can enquire to the retailer that you purchased it from regarding the possibility of returning it to them for a replacement if you purchased the camera in the last 12 months.

MartyG-RealSense, 2024-04-19:
Hi @yanite  Do you require further assistance with this case, please?  Thanks!

yanite, 2024-04-26:
> 嗨，您需要进一步协助此案吗？谢谢！

thank you.

Of course, if there was a better way I would try it, except to reinstall the system ^_^. But I can't think of any more ways to do it at the moment, I'm looking at the code to see if I can find another way

MartyG-RealSense, 2024-04-27:
Please do update me when you have news to report from your investigation.  Good luck!

MartyG-RealSense, 2024-05-17:
Case closed due to no further comments received.

closed 2024-05-17 by MartyG-RealSense

yanite, 2024-05-26:
> 嗨，您需要进一步协助此案吗？谢谢！

windows 11 need start  Windows Camera Frame Server ，