realsenseai/librealsense #12844: corrupted frame

guotonglucky, 2024-04-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   {Client ( Win 10) / Server ( Jetson Nanno B01 Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                        | 
| Platform                                 | PC / NVIDIA Jetson |
| SDK Version                            |  { 2.50.0 }                          | 
| Language                            |  { C++ }                          | 
| Segment			|  { others }                          | 

### Issue Description
* Hi, I built realsense from source with Ethernet on Jetson Nano. Open **Intel.RealSense.Viewer.exe(2.50.0)** on win10, click Add Network Device->Ok, and toggle Depth and RGB sensors on:
* 
[image]

* The following problems occur:
09:39:18 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\ipDeviceCommon\RsUsageEnvironment.cpp:48 - rtsp://10.10.71.59:8554/Stereo Module/:corrupted frame!!!: data size is 2847474191 frame size is 1991

guotonglucky, 2024-04-10:
The depth is incorrect over the Internet.

MartyG-RealSense, 2024-04-10:
Hi @guotonglucky  The range of supported resolution and FPS combinations is more limited on an ethernet connection.  Which resolution and FPS are you using, please (for example, 1280x720 depth at 6 FPS).

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#36-software-limitations

guotonglucky, 2024-04-11:
Hello, 
* **1280 × 720 color at 6   FPS, 1280 × 720 depth at 6   FPS. The depth map is only detectable when I'm near the camera. When I get a little bit further away, it doesn't change. Same as with realsense-viewer.exe.**
[image]

* **640   × 480 color at 30 FPS, 640   × 480 depth at 30 FPS. It's normal.**
[image]
`
int main()
{

	cv::Mat color_mat, depth_colormap;
	cv::Mat_<uint16_t> depth_mat;
	int src_img_width = 1280, src_img_height = 720, max_fps = 6;
	//int src_img_width = 640, src_img_height = 480, max_fps = 30;

	std::string server_ip = "10.10.71.59";

	rs2::context ctx;
	rs2::net_device dev(server_ip);
	dev.add_to(ctx);
	std::cout << "Connected" << std::endl;

	rs2::pipeline pipe(ctx);

	rs2::config cfg;
	cfg.enable_stream(RS2_STREAM_COLOR, -1, src_img_width, src_img_height, RS2_FORMAT_ANY, max_fps);
	cfg.enable_stream(RS2_STREAM_DEPTH, -1, src_img_width, src_img_height, RS2_FORMAT_ANY, max_fps);
	pipe.start(cfg);

	//rs2::align align(RS2_STREAM_COLOR);
	//rs2::hole_filling_filter hole_filling_filter;

	const char* depth_win = "depth_Image";
	const char* color_win = "color_Image";
	cv::namedWindow(depth_win);
	cv::namedWindow(color_win);
	while (true)
	{
		rs2::frameset frameset = pipe.wait_for_frames();
		//frameset = align.process(frameset);

		auto depth_frame = frameset.get_depth_frame();
		auto color_frame = frameset.get_color_frame();

		//depth_frame = hole_filling_filter.process(depth_frame);

		const int w_depth = depth_frame.as<rs2::video_frame>().get_width();
		const int h_depth = depth_frame.as<rs2::video_frame>().get_height();

		const int w_color = color_frame.as<rs2::video_frame>().get_width();
		const int h_color = color_frame.as<rs2::video_frame>().get_height();

		cv::Mat(cv::Size(w_depth, h_depth), CV_16UC1, (void*)depth_frame.get_data()).copyTo(depth_mat);
		cv::convertScaleAbs(depth_mat, depth_colormap, 0.3);
		cv::applyColorMap(depth_colormap, depth_colormap, cv::COLORMAP_JET);

		if (src_img_width == 1280 && src_img_height == 720)
		{
			cv::Mat(cv::Size(w_color, h_color), CV_8UC2, (void*)color_frame.get_data()).copyTo(color_mat);
			cv::cvtColor(color_mat, color_mat, cv::COLOR_YUV2BGRA_YUY2);
		}
		else
		{
			cv::Mat(cv::Size(w_color, h_color), CV_8UC3, (void*)color_frame.get_data()).copyTo(color_mat);
		}

		cv::imshow(depth_win, depth_colormap);
		cv::imshow(color_win, color_mat);
		cv::waitKey(30);
	}

	return 0;
}
`

guotonglucky, 2024-04-11:
And I can't wait to use your LRS-net.

MartyG-RealSense, 2024-04-11:
I cannot see a solution for this issue of short range at 1280x720 depth unfortunately because a higher FPS speed cannot be set.  

The proposed LRS-Net update proposed at https://github.com/IntelRealSenseArchive/lrs-net/tree/apuzhevi_lrsnet was cancelled a long time ago, though the sour e code remains available and Intel do plan to introduce a new networking interface in an upcoming RealSense SDK release.

guotonglucky, 2024-04-12:
Thank you very much.

closed 2024-04-28 by MartyG-RealSense