realsenseai/librealsense #12846: Flickering Issue on realsense d450

Hrithik1049, 2024-04-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I have been experiencing flickering of video while using realsense d450 camera. Is there any settings issue? or something wrong with the hardware.

https://github.com/IntelRealSense/librealsense/assets/117892842/89b27f26-2290-4830-a4cc-f721b72bd227

MartyG-RealSense, 2024-04-10:
Hi @Hrithik1049  Are you enabling the **HDR** function or rapidly changing the exposure value, please?

Hrithik1049, 2024-04-10:
I am running on default settings. Regarding HDR and exposure.

1. HDR is disabled
2. Auto exposure is enabled 
For all the settings, please refer to the settings file I shared in the first post.

Also, wanted to ask why are we getting completely black frames.

MartyG-RealSense, 2024-04-10:
The json defines 640x480 resolution at 30 FPS for depth and IR.  However, when the camera is started the resolution and FPS in the json is ignored.  

If a custom stream configuration (cfg) has not been defined in a program script then the program will instead automatically apply the default stream configuration of the particular RealSense camera model that is being used.  So instead of using 640x480 at 30 FPS, the program is likely using **848x480** on D435i at 30 FPS for depth and IR.  That may not account for the flickering effect on the IR stream though.

If the camera is being used outdoors then it is possible that if the camera faced directly into the sun at some point then its IR sensors could become saturated with light and cause exposure to drop to its minimum, resulting in a black frame.  If auto-exposure is enabled then the camera should automatically recover once the camera is turned away from facing the light source directly.  

Does it still flicker if auto-exposure is disabled and manual exposure is used instead?

MartyG-RealSense, 2024-04-19:
Hi @Hrithik1049  Do you require further assistance with this case, please?  Thanks!

Hrithik1049, 2024-04-22:
Yes, We saw that as soon as we turned off auto exposure some of the flickering went away. we also saw a setting in the code for anti-flicker exposure can you provide some more information on this.

MartyG-RealSense, 2024-04-22:
https://github.com/IntelRealSense/librealsense/issues/10733#issuecomment-1488246925 has information about this mode.

I used the SDK's **rs-sensor-control** tool to check whether Auto Exposure Mode was supported on various models of 400 Series cameras.  For most of them it stated that it was not supported, but on D455 it was supported.  So it should work with D450, as this is the circuit board used inside a D455 camera.

[image]

In a previous case about this instruction, I suggested trying the C++ code below to set it.  It resulted in an error about an invalid value but that may be because that particular case used the D435 model which doesn't support this option.  So please try this code.

```
rs2::pipeline pipe;
rs2::pipeline_profile selection = pipe.start();
rs2::device selected_device = selection.get_device();
auto depth_sensor = selected_device.first<rs2::depth_sensor>();
depth_sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_MODE, 1.f); // Enable auto-exposure anti flicker mode
```

Hrithik1049, 2024-04-29:
Hello @MartyG-RealSense . I ran the code you suggested but got this error. 
Intel RealSense D455 error calling rs2_set_option(options:0x7fe3cc32a150, option:Auto Exposure Mode, value:1):
    object doesn't support option #21

Hrithik1049, 2024-04-29:
I ran the rs-sensor-control and it generated this output for stereo camera sensor 
Sensor supports the following options:

  0: Backlight Compensation is not supported
  1: Brightness is not supported
  2: Contrast is not supported
  3: Exposure
       Description   : Depth Exposure (usec)
       Current Value : 33000
  4: Gain
       Description   : UVC image gain
       Current Value : 16
  5: Gamma is not supported
  6: Hue is not supported
  7: Saturation is not supported
  8: Sharpness is not supported
  9: White Balance is not supported
  10: Enable Auto Exposure
       Description   : Enable Auto Exposure
       Current Value : 1
  11: Enable Auto White Balance is not supported
  12: Visual Preset
       Description   : Advanced-Mode Preset
       Current Value : 0
  13: Laser Power
       Description   : Manual laser power in mw. applicable only when laser power mode is set to Manual
       Current Value : 150
  14: Accuracy is not supported
  15: Motion Range is not supported
  16: Filter Option is not supported
  17: Confidence Threshold is not supported
  18: Emitter Enabled
       Description   : Emitter select, 0-disable all emitters, 1-enable laser, 2-enable laser auto (opt), 3-enable LED (opt)
       Current Value : 0
  19: Frames Queue Size
       Description   : Max number of frames you can hold at a given time. Increasing this number will reduce frame drops but increase latency, and vice versa
       Current Value : 16
  20: Total Frame Drops is not supported
  21: Auto Exposure Mode is not supported
  22: Power Line Frequency is not supported
  23: Asic Temperature is not supported
  24: Error Polling Enabled
       Description   : Enable / disable polling of camera internal errors
       Current Value : 1
  25: Projector Temperature is not supported
  26: Output Trigger Enabled
       Description   : Generate trigger from the camera to external device once per frame
       Current Value : 0
  27: Motion Module Temperature is not supported
  28: Depth Units
       Description   : Number of meters represented by a single depth unit
       Current Value : 0.001
  29: Enable Motion Correction is not supported
  30: Auto Exposure Priority is not supported
  31: Color Scheme is not supported
  32: Histogram Equalization Enabled is not supported
  33: Min Distance is not supported
  34: Max Distance is not supported
  35: Texture Source is not supported
  36: Filter Magnitude is not supported
  37: Filter Smooth Alpha is not supported
  38: Filter Smooth Delta is not supported
  39: Holes Fill is not supported
  40: Stereo Baseline
       Description   : Distance in mm between the stereo imagers
       Current Value : 95.1615
  41: Auto Exposure Converge Step is not supported
  42: Inter Cam Sync Mode
       Description   : Inter-camera synchronization mode: 0:Default, 1:Master, 2:Slave, 3:Full Salve, 4-258:Genlock with burst count of 1-255 frames for each trigger, 259 and 260 for two frames per trigger with laser ON-OFF and OFF-ON.
       Current Value : 0
  43: Stream Filter is not supported
  44: Stream Format Filter is not supported
  45: Stream Index Filter is not supported
  46: Emitter On Off
       Description   : Alternating emitter pattern, toggled on/off on per-frame basis
       Current Value : 0
  47: Zero Order Point X is not supported
  48: Zero Order Point Y is not supported
  49: LDD temperature is not supported
  50: Mc Temperature is not supported
  51: Ma Temperature is not supported
  52: Hardware Preset is not supported
  53: Global Time Enabled
       Description   : Enable/Disable global timestamp
       Current Value : 1
  54: Apd Temperature is not supported
  55: Enable Mapping is not supported
  56: Enable Relocalization is not supported
  57: Enable Pose Jumping is not supported
  58: Enable Dynamic Calibration is not supported
  59: Depth Offset is not supported
  60: Led Power is not supported
  61: Zero Order Enabled is not supported
  62: Enable Map Preservation is not supported
  63: Freefall Detection Enabled is not supported
  64: Receiver Gain is not supported
  65: Post Processing Sharpening is not supported
  66: Pre Processing Sharpening is not supported
  67: Noise Filtering is not supported
  68: Invalidation Bypass is not supported
  69: Digital Gain is not supported
  70: Sensor Mode is not supported
  71: Emitter Always On
       Description   : Emitter always on mode: 0:disabled(default), 1:enabled
       Current Value : 0
  72: Thermal Compensation
       Description   : Toggle thermal compensation adjustments mechanism
       Current Value : 1
  73: Trigger Camera Accuracy Health is not supported
  74: Reset Camera Accuracy Health is not supported
  75: Host Performance is not supported
  76: Hdr Enabled
       Description   : HDR Option
       Current Value : 0
  77: Sequence Name
       Description   : HDR Option
       Current Value : 0
  78: Sequence Size
       Description   : HDR Option
       Current Value : 2
  79: Sequence Id
       Description   : HDR Option
       Current Value : 0
  80: Humidity Temperature is not supported
  81: Enable Max Usable Range is not supported
  82: Alternate IR is not supported
  83: Noise Estimation is not supported
  84: Enable IR Reflectivity is not supported
  85: Auto Exposure Limit is not supported
  86: Auto Gain Limit is not supported
  87: Auto Rx Sensitivity is not supported
  88: Transmitter Frequency is not supported
  89: Vertical Binning is not supported
  90: Receiver Sensitivity is not supported
  91: Auto Exposure Limit Toggle is not supported
  92: Auto Gain Limit Toggle is not supported
  93: Emitter Frequency
       Description   : Controls the emitter frequency, 57 [KHZ] / 91 [KHZ]
       Current Value : 0

MartyG-RealSense, 2024-04-29:
In the rs-sensor-control listing that you kindly provided, item 21 states for your D450 that **Auto Exposure Mode is not supported**.  I tested with my D455, and also with a D457 that is detected as D455 when used with a USB cable and both were listed as supporting Auto Exposure Mode.  So it is possible that a camera has to be specifically identified as D455 in order for Auto Exposure Mode to be accessible.

When you run rs-sensor-control, what name is your D450 module identified by when it first starts, please?

Hrithik1049, 2024-04-29:
So in realsense viewer, it identifies as a D455 camera.

MartyG-RealSense, 2024-04-29:
Which librealsense SDK version and camera firmware driver version are you using, please?

Hrithik1049, 2024-05-10:
[image]

MartyG-RealSense, 2024-05-23:
Case closed due to no further comments received.

closed 2024-05-23 by MartyG-RealSense

Hrithik1049, 2024-06-11:
Hello @MartyG-RealSense So i tried with different combinations of latest firmware and librealsense. None of them seem to work. For the latest firmware and librealsense I am getting the following error. 
`Intel RealSense D455 error calling rs2_set_option(options:0x562845757840, option:Fisheye Auto Exposure Mode, value:0):
    object doesn't support option Fisheye Auto Exposure Mode`

MartyG-RealSense, 2024-06-11:
Hi @Hrithik1049  The D455 does not have a fisheye sensor.  This is only on the retired RealSense T265 Tracking Camera model.  Which program are you running, please?  If it is a T265-only program then RealSense 400 Series camera models such as D455 would not be able to be used with that program.

Hrithik1049, 2024-06-11:
This is the code which we are using 			
```
				rs2::stream_profile left_profile = mD435I_PipelineProfile.get_stream(RS2_STREAM_INFRARED, stream_index);
				mIntrinsics = left_profile.as<rs2::video_stream_profile>().get_intrinsics();

				rs2::region_of_interest rs_roi;
				rs_roi.min_x=5;
				rs_roi.min_y=5;
				rs_roi.max_x=mImageWidth-5;
				rs_roi.max_y=mImageHeight-5;
				
				std::this_thread::sleep_for(std::chrono::milliseconds(100));
				Sensor = mD435I_PipelineProfile.get_device().first<rs2::depth_sensor>();	
				
				Sensor.as<rs2::roi_sensor>().set_region_of_interest(rs_roi);
				
				Sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 0);
				
				Sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
				
				Sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_MODE,0.f);
```

MartyG-RealSense, 2024-06-11:
https://github.com/IntelRealSense/librealsense/issues/5308#issuecomment-560132907 recommends performing an `if (sensor->is<roi_sensor>())` check to confirm that the sensor supports setting the ROI.

Hrithik1049, 2024-06-11:
For some reason, I was not able to run the command you gave tried in realsense Viewer. I was able to define the ROI.

MartyG-RealSense, 2024-06-11:
I reviewed the video at the top of this case again.  I observed that the timing of the flickering is irregular, which suggests that it is not occurring on every frame.  

Also, the flickering stops at the 14 second point out of 17 on the video.

You were using a custom json configuration file, 'D435iSettings'json'.  I tested the json with the infrared stream in the RealSense Viewer on a D435 and a D455 but the infrared image did not flicker like yours did.

Hrithik1049, 2024-06-13:
yes, the flickering is not consistent but it is significant. We have been able to reproduce the flickering when we point at a bright light. Indoors we were able to reproduce the flickering to a moderate scale. Can you repeat the experiment and see if it's happening with your setup ?

MartyG-RealSense, 2024-06-13:
I was not able to replicate the flickering with my D455.  I do not have access to a D450 module though and so cannot test for the difference that a lack of front glass in front of the lenses might make when a D450 is pointed directly at a strong light source.

 In general, when RealSense cameras are pointed directly at strong light such as the sun then the infrared sensors can become saturated with light and cause problems with the image.  If auto-exposure is enabled then the image should auto-correct if the camera is turned away from directly facing the light or the front of the camera is covered over for several seconds.

Hrithik1049, 2024-06-17:
So there is no way to handle such flickers?

MartyG-RealSense, 2024-06-17:
You could try placing a thin film of **linear polarization filter** product over the lenses on the outside of the D450 to negate light glare when directly facing strong light sources.  Any polarization filter should work so long as it is linear, so you can find the filter inexpensively by searching stores such as Amazon for **linear polarizing sheet**