realsenseai/librealsense #12850: Jfrog repo is not available

AndrewBloom, 2024-04-11:
On Android the jfrog repository is used on native_example and on the documentation. url "https://egiintel.jfrog.io/artifactory/librealsense"

It's not reachable, giving a landing page with the following content:

JFrog's Free-Tier subscription has been deactivated.
The following options are available:

Upgrade to a paid account by visiting [MyJFrog](https://my.jfrog.com/)
For Java, Conan and Docker users, JFrog has free community or OSS editions available in our [developer downloads](https://jfrog.com/community/open-source/) area
If you’re an [OSS contributor](https://jfrog.com/community/giving-back/), speaker or JFrog Guru, please [contact us](https://jfrog.com/contact-us/) to explore alternative paths for your projects

MartyG-RealSense, 2024-04-12:
Hi @AndrewBloom  Intel no longer use the JFrog repository or perform any RealSense testing on Android.

AndrewBloom, 2024-04-12:
Hi Martin, please check and correct these files then:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/native_example/app/build.gradle

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/native_example/readme.md

MartyG-RealSense, 2024-04-12:
Hi @AndrewBloom  Intel are not planning on updating the RealSense JFrog support but you are welcome to submit changes via a Pull Request (PR) yourself.

AndrewBloom, 2024-04-12:
@MartyG-RealSense I'm not sure we understand each other.
The JFrog is currently used in master branch, the latest version of your code, on the Android wrappers. The example that uses does not compile (obviously).
So you have an example that does not work. Is that clear?

MartyG-RealSense, 2024-04-12:
Thank you for the clarification.  Most of the Android wrapper has not been updated though for over 3 years, so if there is an example that needs fixing then the wrapper's users are welcome to submit fixes themselves via a PR.

AndrewBloom, 2024-04-12:
I see, thanks for the clarification.

MartyG-RealSense, 2024-04-19:
Hi @AndrewBloom  Do you require further assistance with this case, please?  Thanks!

AndrewBloom, 2024-04-22:
Hi @MartyG-RealSense , thanks for asking. I solved it specifically for my case, so it works for me but I will not submit a Pull Request. 
I hope that Intel will reconsider Android support in the future because in my opinion it's a valuable platform for CV projects.

MartyG-RealSense, 2024-04-22:
I understand, thanks very much for the update.  I'm pleased to hear that you have a solution for your particular case.

closed 2024-04-22 by MartyG-RealSense