realsenseai/librealsense #12851: IMU - roll can fluctuate between 10 degree difference. Is that normal?

ok-kewei, 2024-04-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   |5.14.0 | 
| Operating System & Version |   Ubuntu 20.04 | 
| Platform                                 |Laptop  |
| ROS Version                            |  Noetic                          | 
| Language                            |  {ROS, Python}                          | 

I run this imu filter madgwick (roslaunch imu_filter_madgwick imu_filter_madgwick.launch ) and convert them quaternion_to_euler: roll, pitch yaw.  
But the roll value fluctuated a lot, it shows between 30 degree to 40 degree. The gap is like 10 degrees. Is that normal? 

Thanks a lot for your help. :)

MartyG-RealSense, 2024-04-12:
Hi @ok-kewei  Raw RealSense IMU data can be sensitive to change and 'noisy'.

The IMU can be calibrated using the Python tool at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration

https://www.intelrealsense.com/wp-content/uploads/2020/07/IMU_Calibration_Tool_for_Intel_RealSense-Depth_Cameras_Whitepaper.pdf

ok-kewei, 2024-04-16:
ok, got it. Thanks. :)

MartyG-RealSense, 2024-04-28:
Hi @ok-kewei  Do you require further assistance with this case, please?  Thanks!

ok-kewei, 2024-04-28:
Hi @MartyG-RealSense, thanks for your reply. You may close the case. :)

closed 2024-04-28 by MartyG-RealSense