realsenseai/librealsense #12858: Failed to resolve the request

eladpar, 2024-04-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | () | 
| Operating System & Version |  ubuntu 20 | 
| Platform                                 | jetson xavier nx  |
| SDK Version                            |  { l2.54.1}                          | 
| Language                            |  {C++ / Python }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
So I have  2 cameras connected, one is d435 and the second is d435i
sometimes ( especially when i change between a program that uses C++ sdk and a program that uses python sdk)
I get an error with the depth image only of the d435i ( the other images are fine)
reboot usually fixes this problem
[image]
Any Ideas?

MartyG-RealSense, 2024-04-17:
Hi @eladpar  The **Failed to resolve the request** error means that the camera was unable to provide the 848x480 resolution, most likely because it was set to 30 FPS and it could not provide 30 FPS at that moment.

The most likely cause for this was that a camera was mis-detected as being on a USB 2.1 connection instead of USB 3.2.  In USB 2.1 mode, the D435 will only support 848x480 resolution at the minimum of 6 FPS.  Requesting a higher FPS such as 30 on a 2.1 connection would cause the failed request error.

I also note the information box in the top corner of the Viewer window that you are using firmware driver 5.13.0.50 with SDK version 2.54.1.  Using an older firmware than the one recommended for a particular SDK version can cause errors.  The recommended firmware version for use with 2.54.1 is **5.15.0.2**.

You also have a red **RealSense UDEV Rules are missing!** notification in the top corner of the Viewer.  The udev rules are important because they handle USB devices and can dictate whether a camera works correctly or not.  

I would recommend looking in the top corner of the Viewer to see whether this message repeatedly appears.  If it does then you may need to install the RealSense udev rules.

eladpar, 2024-04-21:
> Hi @eladpar The **Failed to resolve the request** error means that the camera was unable to provide the 848x480 resolution, most likely because it was set to 30 FPS and it could not provide 30 FPS [...]

can it be mis-detected as a USB 2.1 connection instead of USB 3.2, and still show 3.2 on the viewer?
I am familiar with this problem but usually in the realsense viewer it will be written as a 2.1 usb connection

MartyG-RealSense, 2024-04-21:
If the Viewer says 3.2 then it is a 3.2 connection.  If the connection is mis-detected as 2.1 then it will always have 2.1 beside the camera name.

closed 2024-05-04 by MartyG-RealSense