realsenseai/librealsense #12861: try to install pyrealsense at raspian

naorwaiss, 2024-04-17:
```
navkit@raspberrypi:~/librealsense-2.48.0/build $ cmake -G "Unix Makefiles" \
-DCMAKE_C_COMPILER=/usr/bin/gcc \
-DCMAKE_CXX_COMPILER=/usr/bin/g++ \
-DCMAKE_MAKE_PROGRAM=/usr/bin/make \
-DCMAKE_BUILD_TYPE=Release \
-DFORCE_RSUSB_BACKEND=ON \
-DBUILD_PYTHON_BINDINGS:bool=true \
-DBUILD_EXAMPLES=true \
-DBUILD_GRAPHICAL_EXAMPLES=true \
-DBUILD_WITH_CUDA:bool=false \
..
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_COMPILER_ENV_VAR
CMake Error: Could not find cmake module file: /home/navkit/librealsense-2.48.0/build/CMakeFiles/3.27.4/CMakeCXXCompiler.cmake
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_C_COMPILER_ENV_VAR
CMake Error: Could not find cmake module file: /home/navkit/librealsense-2.48.0/build/CMakeFiles/3.27.4/CMakeCCompiler.cmake
CMake Error at /usr/local/share/cmake-3.27/Modules/CMakeTestCompilerCommon.cmake:1:
  Parse error.  Expected a command name, got bad character with text "".
Call Stack (most recent call first):
  /usr/local/share/cmake-3.27/Modules/CMakeTestCXXCompiler.cmake:12 (include)
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
navkit@raspberrypi:~/librealsense-2.48.0/build $

```

can help me with the cmake ??

MartyG-RealSense, 2024-04-18:
Hi @naorwaiss  The best available reference for installing librealsense **and** the pyrealsense2 wrapper on Raspberry Pi is the guide at the link below, which is a few years old now at the time of writing this.

https://github.com/datasith/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

naorwaiss, 2024-04-21:
i have problem with the install of the openGL

```
navkit@raspberrypi:~ $ sudo -H pip3 install pyopengl_accelerate==3.1.3rc1
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
navkit@raspberrypi:~ $ 

```

the installation not work here

MartyG-RealSense, 2024-04-21:
Does OpenGL install if you install the latest version with the command below instead of specifying a specific version?

`sudo -H pip3 install pyopengl_accelerate`

naorwaiss, 2024-04-21:
sudo apt-get install python-opengl
sudo -H pip3 install pyopengl
sudo -H pip3 install pyopengl_accelerate==3.1.3rc1

this try line is problamatic and stack all the installation process 

```
navkit@raspberrypi:~ $ sudo apt-get install python-opengl
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python-opengl
navkit@raspberrypi:~ $ 

```

this is the first line the seccent is broke - try with  --break-system-packages - but gave me warring 

```
(newenv) navkit@raspberrypi:~ $ sudo -H pip3 install pyopengl_accelerate
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
(newenv) navkit@raspberrypi:~ $ sudo -H pip3 install pyopengl_accelerate --break-system-packages
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyopengl_accelerate in /usr/local/lib/python3.11/dist-packages (3.1.7)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
(newenv) navkit@raspberrypi:~ $ deactivate

```

MartyG-RealSense, 2024-04-22:
Does the python-opengl install work if you insert -y into the installation command?

`sudo apt-get install -y python-opengl`

If that still does not work, an alternative installation method is to use **pip install**.

`sudo pip3 install pyopengl`

closed 2024-05-04 by MartyG-RealSense