realsenseai/librealsense #12863: Depth Quality Tool for Intel RealSense Cameras Execution error

ky047, 2024-04-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | ( D400 ) | 
| Firmware Version                   | (2.54.2) | 
| Operating System & Version |   (window 10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.54.2 }                          | 
| Language                            |                           | 
| Segment			|  others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello
I'm going to calibrate the RealSense D435
During the calibration process, run Intel RealSense Viewer
I've confirmed that the camera is working normally
I want to run the Depth Quality Tool for Intel RealSense Camera to see if it meets performance standards
The Depth Quality Tool does not open!!
Is there a way for me to solve the problem?
The graphic card is the latest version

MartyG-RealSense, 2024-04-18:
Hi @ky047  Have you got the depth stream enabled in the RealSense Viewer before you launch the Depth Quality Tool, please?  If you have then you should stop the depth stream in the Viewer, otherwise the Depth Quality Tool will not open.  This is because the Viewer has already claimed the right to access the depth stream.

ky047, 2024-04-19:
Where should I disable the depth stream by any chance?
Can you tell me by picture or image?

ky047, 2024-04-19:
Is it correct to deactivate the red box in the picture?
Red box is disabled, but depth quality tool is not running and is in infinite loading state
[image]
[image]

Depth quality tool loading screen

MartyG-RealSense, 2024-04-19:
When the depth stream is enabled, the red icon next to 'Stereo Module' is blue.  Clicking on the blue icon disables the depth stream and turns its color back to red.  You can be certain that the depth stream is disabled in the Viewer by closing the Viewer program completely. 

Yes, having the UWP driver installed can cause problems.  It is not required to use RealSense cameras with Windows or to perform calibration.  If the RealSense Viewer can run normally though then I would think that UWP is not affecting the Depth Quality Tool, as it is very similar to the Viewer.

ky047, 2024-04-21:
Yes UWP driver was a problem
Depth Quality has been enabled after deleting the UWP driver
Do I not need a UWP driver to proceed with the camera calibration?
Can I start calibration with camera one chip calibration?

ky047, 2024-04-22:
I finished the calibration using Dynamic Calibrator
At first, the Z accuracy exceeded 2%, so I was worried, but after calibration, it was adjusted to 0.5%
If I'm upgrading the camera firmware
Does the camera get disconnected or not working? (I haven't upgraded it yet. I can't proceed because I'm afraid it might break down during the upgrade)

MartyG-RealSense, 2024-04-22:
The camera resets during the firmware update process at around the 30% point in the update process and then attempts to re-connect to the camera.  In the majority of cases this update completes successfully.  The camera does have a recovery mechanism if something goes wrong with the firmware update.

ky047, 2024-04-22:
Can you tell me where the recovery mechanism is?

MartyG-RealSense, 2024-04-23:
If a camera is in a 'recovery' state after a failed firmware update then it may be able to be recovered using the **rs-fw-update** firmware reocvery tool at the link below by adding an -r flag to the firmware update command.

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

For example, if you wanted to get the camera out of a recovery state and at the same time update to firmware 5.15.1.0, you would download the firmware .bin file for 5.15.1.0 from the firmware releases page at the link below.

https://dev.intelrealsense.com/docs/firmware-releases

Then you would launch the rs-fw-update tool with a command like the one below:

`rs-fw-update -r -f Signed_Image_UVC_5_15_1_0.bin`

ky047, 2024-04-24:
Is there no recovery solution in the realsense viewer?
Additionally, if I click rs-fw-update
It turns off automatically
I used window 10 and did it at the top connected to the camera
I used rs-fw-update.exe as shown in the manual
Is there a problem I need to take care of?

MartyG-RealSense, 2024-04-24:
A camera that is in recovery mode will typically display an identity like the one below in the RealSense Viewer.  Whilst there is not a recovery option like in rs-fw-update, you can attempt to perform a firmware update in the Viewer on a camera that is in recovery mode.

[image]

Some of the RealSense SDK tools and examples, including rs-fw-update, have to be launched from the Windows **Command Prompt** interface instead of double-clicking on them.  Information about this launch method and using rs-fw-update's recovery mode can be found at https://github.com/IntelRealSense/librealsense/issues/8500#issuecomment-791540583

ky047, 2024-04-24:
ntel.Realsense.CustomRW.exe-
Even with this command, I can't downgrade the camera
Is there any way?

MartyG-RealSense, 2024-04-24:
The only option left other than purchasing a new D435 would be to purchase a circuit board called the **Vision Processor D4 V1** (one of the two circuit boards inside the D435 and the one that the firmware is stored on) and then open up the camera casing and swap the D4 V1 board.  The downgrade counter should then be at zero again with the new board.

If your D435 camera is less than a year old since purchase then opening up the casing would likely invalidate the D435's sales warranty though.  And if you do choose to order a new D4 board then make sure that you order the V1 version, or V2 (the one with an IMU component) if V1 is not available.  Versions V3 to V5 will not be compatible with the other circuit board inside your D435 (the D430 depth module).

ky047, 2024-04-25:
I had no choice but to update the new version 5.16.0.1
It's more compatible than I thought, so it's working fine
Your answer was very helpful
We can close this case
Thank you!

MartyG-RealSense, 2024-04-26:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update.  As you suggested, I will close this case.  Thanks again!

closed 2024-04-26 by MartyG-RealSense