realsenseai/librealsense #12867: D435i + problem with topic imu (orientation) + urdf

LauGasp, 2024-04-18:
Good afternoon, 

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |    Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |        5.15.0-102-generic                                   | 
| Platform                                 | Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz  |
| SDK Version                            | 2.54.2.0   | 
| ROS Distro			         |  noetic                                          | 
| RealSense ROS Wrapper Version   |  2.3.2                                                  | 

I'm using the urdf model of the D435i camera for Gazebo simulation. 
I added the imu (libgazebo_ros_imu_sensor.so) plugin and compared the frame transformations of the camera's urdf model with those of the real camera (I viewed both in rviz), so they are the same.
However, when I checked the imu topic published by the real camera:

```
^Cheader: 
  seq: 53787
  stamp: 
    secs: 1713445870
    nsecs: 239586830
  frame_id: "camera_imu_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 9.315534407505766e-05
  y: 0.0016925366362556815
  z: 1.1077881026722025e-05
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: 0.21089705027836053
  y: -9.85674596718617
  z: -0.301333504009737
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
---
```

And comparing it with the imu from the camera on Gazebo:

```
header: 
  seq: 2770
  stamp: 
    secs: 109
    nsecs: 258000000
    
  frame_id: "d435i_gyro_optical_frame"
orientation: 
  x: -2.3165440301105467e-06
  y: -0.003588130480113342
  z: -0.0003836942958122569
  w: 0.9999934890252931
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.0004599363701062937
  y: -0.0009509356986391131
  z: 6.470457888301394e-06
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.038919107240427644
  y: 0.023893370053852094
  z: 10.61458893750547
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

I notice that the gravity orientation is different (real camera: -9.8 in y and Gazebo: 9.8 in z). What could be causing this problem? How do I solve it?

what I added to gazebo.urdf:
```
 <!-- IMU -->
        <xacro:if value="${unite_imu_method == 'copy' or unite_imu_method == 'linear_interpolation'}">
        <sensor name="${camera_name}imu" type="imu">
          <always_on>true</always_on>
          <update_rate>100</update_rate>
          
          <topic>${topics_ns}/imu/sample</topic>
          <plugin name="{camera_name}imu" filename="libgazebo_ros_imu_sensor.so">
            <topicName>${topics_ns}/imu/sample</topicName>
            <bodyName>${camera_name}_link</bodyName>
            <updateRateHZ>10.0</updateRateHZ>
            <gaussianNoise>0.0</gaussianNoise>
            <xyzOffset>0 0 0</xyzOffset>
            <rpyOffset>0 0 0</rpyOffset> 
            <frameName>${gyro_optical_frame}</frameName>
          </plugin>
        </sensor>
      </xacro:if> 
```

MartyG-RealSense, 2024-04-19:
Hi @LauGasp  RealSense cameras use a different coordinate system to ROS, where Z = forward instead of up.  

RealSense cameras use the Y axis in the IMU to represent the vertical 9.8 gravity (the Accel value) instead of Z.  When a RealSense camera is used with the RealSense ROS wrapper, the wrapper automatically converts the RealSense axes to the ROS standard format axes.

LauGasp, 2024-04-20:
Hello, thank you so much @MartyG-RealSense. 
The pink arrow of the IMU in RViz is pointing vertically upwards after I read the data from the IMU topic (with linear_acceleration values x=0, y=-9.8 and z=0). 
This seems ok to me, because the method (kimera-vio-ros) I'm trying to test must be designed to operate in this specific coordinate system of the RealSense camera. I say this because the estimated odometry is working fine with this configuration.

However, when it comes to simulating in Gazebo with the realsense-ros models, it seems that the IMU is following the ROS standard format axes. In this case, I would have to convert the IMU data from the RealSense convention to the ROS convention.
I did the conversion below:

```
import rospy
from sensor_msgs.msg import Imu

def imu_callback(msg):
    modified_msg = Imu()
    modified_msg.header = msg.header

    modified_msg.orientation.x = -msg.orientation.y
    modified_msg.orientation.y = -msg.orientation.z
    modified_msg.orientation.z = msg.orientation.x  
    modified_msg.orientation.w = msg.orientation.w

 
    modified_msg.angular_velocity.x = -msg.angular_velocity.y
    modified_msg.angular_velocity.y = -msg.angular_velocity.z
    modified_msg.angular_velocity.z = msg.angular_velocity.x
  
    modified_msg.linear_acceleration.x = -msg.linear_acceleration.y
    modified_msg.linear_acceleration.y = -msg.linear_acceleration.z
    modified_msg.linear_acceleration.z = msg.linear_acceleration.x
 
    pub.publish(modified_msg)

if __name__ == '__main__':
    rospy.init_node('imu_modifier_axes')

    # Subscreva ao tópico de IMU
    rospy.Subscriber('/Pionner/D435i_camera/imu/sample', Imu, imu_callback)

    # Crie um publicador para a mensagem IMU modificada
    pub = rospy.Publisher('/Pionner/D435i_camera/imu/modified', Imu, queue_size=10)

    # Mantenha o nó em execução
    rospy.spin()
```
The thing is, the odometer isn't working properly at all, does the conversion make sense or do you have any idea what it might be?

MartyG-RealSense, 2024-04-21:
The IMU in the D435i can measure gyroscopic angle and acceleration but it is not able to track the camera's position and the distance travelled.

MartyG-RealSense, 2024-04-28:
Hi @LauGasp  Do you require further assistance with this case, please?  Thanks!

LauGasp, 2024-04-28:
Hi, @MartyG-RealSense, thanks for your attention! I know that the camera doesn't give me odometry, but it does give me gyro angle and acceleration measurements. I'm still facing the same problem and I don't know how to integrate the Kimera-vio-ros method with the Realsense D435i and the Gazebo. When I send the topics (infra1, infra2 and imu) from the URDF model, the estimated odometry (output) is very illogical and incorrect, which leads me to suspect that the input sent is not correct. This does not happen with the topics sent by the real camera.
Any idea what it might be?

MartyG-RealSense, 2024-04-28:
My knowledge of Kimera-VIO-ROS is limited, unfortunately.  A discussion about using a RealSense camera with it at https://github.com/IntelRealSense/realsense-ros/issues/1713 may provide helpful insights.

Instructions for using Kimera-VIO-ROS with a RealSense camera can be found at the link below if you have not seen them already.

https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/docs/hardware_setup.md

closed 2024-05-10 by MartyG-RealSense