realsenseai/librealsense #12869: Orin nano failed to read USB config descriptor with D455

ZombieFly, 2024-04-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   L4T 35.3.1 / Jetpack 5.1.1 (Ubuntu 20)  | 
| Kernel Version (Linux Only)    | 5.10.104-tegra                                      | 
| Platform                                 | NVIDIA Jetson Orin Nano  |
| SDK Version                            |  2.54.2                         | 
| Language                            |  python                         | 
| Segment			|  Robot                          | 

### Issue Description
I have a D455 which could be opend my Windows with realsense view.
I installed the librealsense on my orin nano using [libuvc_installation.sh](https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh), and I can see `Bus 002 Device 018: ID 8086:0b5c Intel Corp.` after running `lsusb | grep 8086`. There are four usb3.0 ports on my orin nano 8GB.After I plug the D455 into echo port, realsense view will reports `(enumerator-libusb.cpp:45) failed to read USB config descriptor: error = -1`.
[image]

MartyG-RealSense, 2024-04-20:
Hi @ZombieFly  The libuvc_installation.sh script does not include the RealSense examples and tools (such as RealSense Viewer) in its build.  Did you create a custom version of the script with **-DBUILD_EXAMPLES=TRUE -DBUILD_GRAPHICAL_EXAMPLES=TRUE** added on line 46 of the script, please? 

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

If you did not modify libuvc_installation.sh, can you tell me please how you added the RealSense Viewer tool to your Orin Nano?

ZombieFly, 2024-04-20:
Hi @MartyG-RealSense I tried to run `sudo make uninstall` and I saw `-- Uninstalling "/usr/local/bin/realsense-viewer"`.
After it, the command `realsense-viewer` became unusable. Now I modified the [line 46](https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46), added `-DBUILD_WITH_CUDA=true`(without add tools).Subsequently, I tried running the modified script.

During the build process, I still encountered the following output
``` bash
Scanning dependencies of target rs-data-collect
Scanning dependencies of target rs-gl
Scanning dependencies of target realsense-viewer
Scanning dependencies of target rs-rosbag-inspector
Scanning dependencies of target rs-depth-quality
Scanning dependencies of target rs-benchmark
```

``` bash 
[ 86%] Built target realsense-viewer
```
And after `make install`, there is a output:`-- Installing: /usr/local/bin/realsense-viewer`.
However, when I launch the realsense viewer with the D455 plugin connected, I still encounter the error: `(enumerator-libusb.cpp:45) failed to read USB config descriptor: error = -1`

ZombieFly, 2024-04-20:
The D455 camera appears to be functioning properly on Windows using the same cable.
[image]

MartyG-RealSense, 2024-04-21:
Is the camera able to be detected on your Orin Nano if you use the command **rs-enumerate-devices -s** to list a short summary of information about the camera?

ZombieFly, 2024-04-22:
`No device detected. Is it plugged in?`

MartyG-RealSense, 2024-04-22:
The next area that you could investigate is whether the Orin Nano is providing the camera with insufficient power to operate.  Intel provided guidance for the original Jetson Nano that strongly recommended enabling its **barrel jack** power connector if the Nano board had one, using the instructions at the link below.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

The Orin Nano also has a barrel jack (point 8 on the illustration below), so Intel's Nano power advice may also be relevant to Orin Nano.

[image]

MartyG-RealSense, 2024-04-28:
Hi @ZombieFly Do you require further assistance with this case, please?  Thanks!

ZombieFly, 2024-04-29:
I followed your advice and used the  barrel jack power connector, and enabled the 15W mode, but it still didn't work. Later, I purchased a USB hub with independent power supply (the device manufacturer told me that a single port can provide 3W output), but it still didn't solve the problem.

ZombieFly, 2024-04-29:
The USB hub I'm using is manufactured by UGREEN. It features four USB 3.0 outputs, and I'm using a 5V 2A power adapter to power it. It should provide enough supply for the D455. So I'm considering whether the issue might not be caused by insufficient power supply.

MartyG-RealSense, 2024-04-29:
If the problem is with the installation on Orin Nano rather than a camera or USB related issue then you could try installing the SDK from packages with the **JetsonHacks** website's 'installLibrealsense.sh' build script at the link below.

https://github.com/jetsonhacks/installRealSenseSDK/tree/master?tab=readme-ov-file#installlibrealsensesh

If you use the installLibrealsense.sh script, do not use the 'buildLibrealsense.sh' source code installation script afterwards.

closed 2024-05-17 by MartyG-RealSense