realsenseai/librealsense #12870: Realsense-viewer crashes while opening on MacOS sonoma

Niranjini-Kathiravan, 2024-04-20:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Mac OS| sonoma

When I open the realsense-viewer without connecting the camera it's able to open but when I connect the camera the realsense-viewer crashes and it's not opening.
Can someone help me in resolving this issue?
Also is there any minimum system requirements for running realsense-viewer on MacOS?
The errors says it is segmentation fault. 
I tried reinstalling but still it's not working. 
Please do let me know if you have a solution for this issue.

MartyG-RealSense, 2024-04-21:
Hi @Niranjini-Kathiravan  I note that you have posted a comment at the other case at https://github.com/IntelRealSense/librealsense/issues/12727 about problems with realsense-viewer on MacOS Sonoma.

Official support in the RealSense SDK for MacOS ended in 2018, though the SDK has continued to work with Mac in subsequent years and the community of RealSense MacOS users investigates problems and shares solutions with each other.  

From MacOS Monterey onwards however problems became more common and there are unfortunately not clear solutions for Sonoma issues available at the time of writing this.  Discussions led by MacOS users such as https://github.com/IntelRealSense/librealsense/issues/12727 are the best chance of finding a resolution.

There are not stated minimum requirements for running realsense-viewer on MacOS.

MartyG-RealSense, 2024-04-28:
Hi @Niranjini-Kathiravan  Bearing in mind the advice in the comment above, do you require further assistance with this case please?  Thanks!

closed 2024-05-04 by MartyG-RealSense