realsenseai/librealsense #12872: Announcement: RealSense SDK version 2.55.1 now available

MartyG-RealSense, 2024-04-21:
Hi everyone,

Version **2.55.1** of the RealSense SDK (librealsense is now available.  The recommend firmware for this version is **5.16.0.1**

https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1

**RELEASE NOTES**
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2551

**API CHANGES**
https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2551

Petrox, 2024-04-21:
@MartyG-RealSense Thank you for the new release.

The release note here and in the package mentions 5.16.0.1 firmware and then links to https://dev.intelrealsense.com/docs/firmware-releases which has not been updated and the firmware is not available to download (yet).

Also the tools/fw-update/readme.md links to a quite old v5.11.6.250 firmware https://downloadcenter.intel.com/download/28870/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?product=128255

Also this page (the downloadcenter document) was not updated with the latest firmware yet, it's currently at 5.15.x.

MartyG-RealSense, 2024-04-21:
Hi @Petrox   There can be a short delay between the day of release of a new SDK version and the listing of its recommended firmware driver on the firmware releases page.  In the meantime you can update to the latest firmware with the 2.55.1 version of the RealSense Viewer tool or download the bin file for 5.16.0.1 directly to your computer using the link below.

https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-5.16.0.1.bin