realsenseai/librealsense #12876: How can I get an object's location coordinates with simple python code from D435i?

Tranthanhbao198, 2024-04-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (2.54.2.5684) | 
| Operating System & Version | Win 11 and mac os 14 Pro M1  | 
| Kernel Version (Linux Only)    |  NA                                        | 
| Platform                                 | PC |
| SDK Version                            |    | 
| Language                            | python}                          | 
| Segment			|  {Robot arm  }                          | 

### Issue Description
Hello, 
Currently I have my own project using D435i to observe objects, then get information about the object's location coordinates, then will use python code to take those location send to the robot arm's moving command to pick the object, i will do it with 2 levels. 1 is i will show all of the objects on the table, then get all of the object's location coordinates, 2: I will place the objects on the table one by one and the camera will observe and get the coordinates of the locations of those objects one by one. Please help me. thank you so much

Tranthanhbao198, 2024-04-23:
hi @MartyG-RealSense , thank you for your reply, but I didn't quite find what i was looking for there. can u give me more details. thank you

MartyG-RealSense, 2024-04-23:
Do you need further information about solvePNP, please?  If you do then the official OpenCV documentation has more here:

https://docs.opencv.org/4.x/d5/d1f/calib3d_solvePnP.html

closed 2024-05-26 by MartyG-RealSense