realsenseai/librealsense #12878: Left and Right RGB Images

suptimq, 2024-04-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description

Since I saw one rgb image ROS topic and the RealSense Viewer shows one rgb image only, I wondered if it is possible to save the left and right images somehow.

MartyG-RealSense, 2024-04-25:
Hi @suptimq  Only the left infrared sensor of RealSense cameras can provide an RGB image, and this feature is only supported on some camera models (D405, D415 and the D455-class models such as D455, D455f, D456 and D457).

These models can though show both the 'RGB from left infrared image' and the normal RGB image at the same time.

suptimq, 2024-04-26:
Hi @MartyG-RealSense Thanks for the quick response! I appreciate the information for D405 and D455-class models. However, could you provide information for D435-class models such as D435 and D435i?

I saw there are a left imager and a right imager https://www.intelrealsense.com/depth-camera-d435/.

MartyG-RealSense, 2024-04-26:
The left and right infrared sensors on the D435 / D435i / D435f / D435if models are monochrome and so do not support RGB from left IR mode.  You can stream an **RGB-like** infrared image from both the left and right sensors if you disable the dot-pattern projection to remove the dots overlaid on the image but it will be in Y8 format rather than RGB and both images will also be black and white. 

[image]

closed 2024-04-26 by suptimq

wenbowen123, 2024-09-06:
@MartyG-RealSense how to get the IR images from D405 using the python API? The folllowing will give an error "RuntimeError: Couldn't resolve requests". However, the same worked for D415.
```
config.enable_stream(rs.stream.depth, W, H, rs.format.z16, fps)
config.enable_stream(rs.stream.color, W, H, rs.format.bgr8, fps)
config.enable_stream(rs.stream.infrared, 1, W, H, rs.format.y8, fps)
config.enable_stream(rs.stream.infrared, 2, W, H, rs.format.y8, fps)

profile = pipeline.start(config)
```

MartyG-RealSense, 2024-09-07:
Hi @wenbowen123  **RuntimeError: Couldn't resolve requests** means that a custom configuration was requested that the camera was unable to provide.

What FPS are you setting Infrared 1 and 2 to, please?  For example, if they were set to '6' (which D415 supports) then this could cause the RuntimeError on D405 as it supports 5 FPS instead of 6.

Another cause of the error would be if the USB connection on the D405 was being detected as USB 2.1 instead of USB 3, as the right-side Infrared 2 stream is not supported on a USB 2.1 connection, only on USB 3.

Also, which resolution are you using in W and H please?

wenbowen123, 2024-09-16:
USB 2.0 seems to be the issue, thanks @MartyG-RealSense !