realsenseai/librealsense #12881: can I get depthmap from recorded bag file (left / right IR images) by using SDK?

sunglyoungKim, 2024-04-25:
Hello,

I am trying to make a depth map using a recorded bag file with RGB, left, and right IR images. If I can do it offline rather than on the D415 chipset, I can increase fps with high resolution. Of course, the drawback is that it would not be a real-time application.

I have looked up documentation and this issue board and found a hint that I can use the OpenCV disparity map. Of course, I can do it by using left and right IR images.

However, Realsense uses a modified disparity map. I wonder if we can calculate a depth map using the Realsense SDK or if I have to implement it by following the Realsense document for making a depth map.

I hope someone can answer my question, and thank you so much!

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | 2.54.2 | 
| Operating System & Version |    Linux (Ubuntu 22.04)  | 
| Kernel Version (Linux Only)    |  (6.5.0-28-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.5.42 }                          | 
| Language                            |  python/opencv }                          | 
| Segment			|  {object detection / tracking / segmentation }                          |

MartyG-RealSense, 2024-04-25:
Hi @sunglyoungKim  Are you aiming to generate a disparity map with Python and OpenCV, please?  If you are then the information at https://github.com/IntelRealSense/librealsense/issues/8572 may be helpful to you.

If it is a depth map that you are aiming for then the Python / OpenCV reference at the link below may suit your requirements.

https://support.intelrealsense.com/hc/en-us/community/posts/23846939164947-RealSense-D405-Depth-Map-Fidelity

sunglyoungKim, 2024-04-25:
Thank you for replying to my questions so quickly.

Yes, it seems like I should follow #8572.

I have found this C++ code, too. Is it the same code that #8572 mentions about converting C++ code into Python?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/depth-filter

sunglyoungKim, 2024-04-26:
@MartyG-RealSense 
Hello,
Maybe I am wrong, but based on the links and #8572, it seems like I have to use OpenCV to generate disparity and depth maps from left and right IR images.

After I got the depth map, is it possible I can encode it correctly and point the dataset into this object?
config.enable_stream(rs.stream.depth, depth_res[0], depth_res[1], rs.format.z16, fps)

So that I can use all the align frames, post-processing filters, and make point cloud.

Thank you!

MartyG-RealSense, 2024-04-27:
Once you have generated images in OpenCV, https://github.com/IntelRealSense/librealsense/issues/2634 advises about a method to convert back to the RealSense SDK's **rs2::frame** format with C++ code so that you can use the SDK's functions on the data.

closed 2024-05-02 by sunglyoungKim

MartyG-RealSense, 2024-05-03:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update!