realsenseai/librealsense #12883: How to map outdoor environment using realsense D455 camera

adinathkale01, 2024-04-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I am using realsense D455 camera for outdoor mapping by holding camera in hand , i installed relasense camera package and rtabmap, while mapping after some time 3D map showing red and new ID not updating , in that case what to do, what is correct way to map outdoor environment day and night.I just want to map environment by holding camera in hand

MartyG-RealSense, 2024-04-27:
Hi @adinathkale01  If the problem happening during the day then the camera may have froze if pointed directly at a strong light source such as the sun.  This occures because the infrared sensors become saturated with light.  

If this happens then, if auto-exposure is enabled, the camera should automatically recover when the camera is turned away from the light source or you cover the front of the camera with your hand for several seconds.

You should be able to also use the camera at night if the camera's infrared dot pattern projection is enabled.

adinathkale01, 2024-04-27:
does speed ,hand movement or sudden turn affect on mapping or not

MartyG-RealSense, 2024-04-27:
D455 has sensors that are good at capturing depth during fast turns or motion.  Quick turns are more likely to negatively affect the IMU (gyro and accel) rather than depth.

MartyG-RealSense, 2024-05-08:
Hi @ Do you require further assistance with this case, please?  Thanks!

closed 2024-05-17 by MartyG-RealSense