realsenseai/librealsense #12884: linux, using d435i in pycharm, possible usb issues

EthanUNC, 2024-04-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | d435i | 
| Firmware Version                   |  | 
| Operating System & Version | ubuntu 22.04.4  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 |linux |
| SDK Version                            | sdk   2.54.2                    | 
| Language                            |python                    | 
| Segment			|                      | 

### Issue Description
I used d435i in pycharm, but with the following error.

profile = pipeline.start(config)
RuntimeError: Couldn't resolve requests

I thought it was a usb permission problem, so I went through the following steps.
1.  I go into this directory: /etc/udev/rules.d
2.  I rewrote a file called 50-myusb.rules
    This is the content:
UBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b3a", GROUP="users", MODE="0666"
3.   I restarted linux: sudo reboot  
4.   Execute this command: sudo udevadm control --reload
In this way, I successfully called d435 before, but now failed again.

Run lsusb:
Bus 001 Device 005: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i

MartyG-RealSense, 2024-04-28:
Hi @EthanUNC  A **Couldn't resolve requests** error would suggest that a 'config' custom stream configuration was defined that the camera was not able to provide at the time that the script was run.  An example of why this could happen would be if the USB connection was detected as being USB 2.1 instead of the faster USB 3.2, since the list of supported resolution / FPS combinations is more limited on a USB 2.1 connection.

For example, if 1280x720 color at 30 FPS was requested and the USB connection was detected as 2.1 then the `Couldn't resolve requests` error would occur because only 6 FPS will be supported on a D435i at 1280x720, whilst 30 FPS would be supported at 640x480 resolution.

EthanUNC, 2024-04-30:
@MartyG-RealSense Thank you so much. I made a stupid question. Forgot to check my usb port!

closed 2024-04-30 by EthanUNC