realsenseai/librealsense #12886: How do i install Pyrealsense2-aarch64 into my Raspberry pi 4 8Gb version

Sharath-MRNoBoDy, 2024-04-29:
Products that iam using are as follows 

Camera                   | Intel  Realsense camera d457
OS                          | Ubuntu raspberry pi bookworm 
Platform                  | Raspberry pi 4b 8gm version
kernel                      | 5.15.0-1053-raspi

I was able to follow the IntelRealsense SDK tutorial that is available [here](https://github.com/IntelRealSense/librealsense) .
I am able to view the camera from the realsense Veiwer and SDK.

Now comes the error i get by installing pyrealsense2-aarch64  it throws a error saying 

```
pip install pyrealsense2-aarch64
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement pyrealsense2-aarch64 (from versions: none)
ERROR: No matching distribution found for pyrealsense2-aarch64
```
I have install 
python 3.10.12 
PIP 22.0.2

MartyG-RealSense, 2024-04-29:
Hi @Sharath-MRNoBoDy  Until recently the aarch64 Arm version of the pyrealsense2 pip packages only worked for Python 3.7 to 3.9 and you are using Python 3.10.  Using an unsupported Python 3 version with pip install would cause the message **ERROR: No matching distribution found for pyrealsense2-aarch64**

Since version 2.55.1 of the SDK was released a week ago (April 21 2024), Python 3.10 has been supported.

[image]

Have you attempted installation of the SDK and pyrealsense2 wrapper since 2.55.1 and the new aarch64 file were released, please?

Sharath-MRNoBoDy, 2024-04-29:
Yes i did try to install the above library and iam getting this error for it 

`pip install pyrealsense2
Defaulting to user installation because normal site-packages is not writeable
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1007)'))': /simple/pyrealsense2/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1007)'))': /simple/pyrealsense2/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1007)'))': /simple/pyrealsense2/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1007)'))': /simple/pyrealsense2/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1007)'))': /simple/pyrealsense2/
Could not fetch URL https://pypi.org/simple/pyrealsense2/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pyrealsense2/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1007)'))) - skipping
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2`

and also even the `pip install pyrealsense2-aarch64`

Sharath-MRNoBoDy, 2024-04-29:
yes i had tried to install this pyrealsense2 package but it still not able to resolve this issue on the RPI 4B 
when i run a program using the import of this library i get this error 
`
Traceback (most recent call last):
  File "/home/program.py", line 13, in <module>
    pipe.start(cfg)
RuntimeError: Couldn't resolve requests
`
iam using the usb type c 3.1 cable 
but the same cable is able to fetch the camera feed from the SDK but not able to view the feed when running a program

MartyG-RealSense, 2024-04-29:
**RuntimeError: Couldn't resolve requests** is not related to the SDK or Python version.  It occurs when a stream resolution / FPS combination has been requested that the camera cannot currently provide.  If a valid resolution / FPS has been defined in a script then a common reason for it is if the camera's USB connection type is being mis-detected as USB 2.1 instead of 3.1 or 3.2.  

A way to test for this possibility would be to set your streams to 640x480 and 30 FPS, as this configuration will work with both USB 2.1 and 3.1 / 3.2.  If 640x480 works then it is likely that your camera connection is being detected as USB 2.1 when running the script.

Sharath-MRNoBoDy, 2024-04-29:
@MartyG-RealSense I have specified to the same frame specs as you have mentioned but iam not able to fetch the live feed 

`cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

cfg.enable_stream(rs.stream.infrared, 640, 480, rs.format.y8, 30)`

I am trying to fetch 3 feed from the camera of RBG , Dept and Infrared 
i am running the same program in the my PC 
but the programs runs with out any for infrared feed but the frame gets stuck in the RGB and Dept feed

MartyG-RealSense, 2024-04-29:
Do you have any other programs running that are using the depth and RGB streams when you run your script?

Sharath-MRNoBoDy, 2024-04-30:
@MartyG-RealSense no , just the program getting the RGB, Dept, and Infrared stream 
other than this the RealSense Viewer is closed before running the program

MartyG-RealSense, 2024-04-30:
What happens if you comment out the infrared stream and just access depth and color on your Pi?

Sharath-MRNoBoDy, 2024-05-01:
the Dept and the RGB streams work with out any issues but when i add the infrared feed into the picture the it throws me 
`line 13, in <module>
    pipe.start(cfg)
RuntimeError: Couldn't resolve requests
`

Sharath-MRNoBoDy, 2024-05-01:
Even though I comment the NIR(Infrared) feed and other related snips 
The code is still giving me the Runtime error 

`
line 13, in <module>
    pipe.start(cfg)
RuntimeError: Couldn't resolve requests
`

But the same Code Snippet is able to show the RGB and Dept Feed when I connect the camera into the PC and run the code.
This tells the code is error free

MartyG-RealSense, 2024-05-01:
Does the script run if you remove the word **cfg** from the brackets of the pipe start line so that the script ignores the cfg lines and applies the camera's default configuration for each stream instead?

If the script does run with cfg removed, does the script work if you put cfg back in the brackets and remove the spaces before the ; semi-colon line end symbol that you have on the depth and infrared instructions?

`cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30); 
cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30) ; 
cfg.enable_stream(rs.stream.infrared, 640, 480, rs.format.y8, 30) ;`

Sharath-MRNoBoDy, 2024-05-03:
If I do that change then it shows this error 
`
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'
`
since there is not attribute of **Pipeline** in pyrealsense2 package it is showing me the above error 
And if i add the (cfg) in the pipeline then it wont run and shows Runtime error for the same line

MartyG-RealSense, 2024-05-03:
When removing **cfg** from the brackets, does the AttributeError still occur if you change the **import pyrealsense2 as rs** line to the one below?

`import pyrealsense2.pyrealsense2 as rs`

Sharath-MRNoBoDy, 2024-05-08:
When doing all this iam back to ground zero with a fresh boot to raspberry pi os i tried to build the wrapper from the [here](https://github.com/MartyG-RealSense/librealsense/tree/master/wrappers/python#installation) but getting all types of error from proxy server(connecting with the network team) to 

```
sudo apt-get install librealsense2-utils
sudo apt-get install librealsense2-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package librealsense2-utils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package librealsense2-dev

```

I followed the guid from here 
```
sudo apt-get install git libssl-dev libusb-1.0-0-dev libudev-dev pkg-config libgtk-3-dev -y
./scripts/setup_udev_rules.sh  
mkdir build && cd build  
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true && make -j$(($(nproc)-1)) && sudo make install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.39.2-1.1).
libusb-1.0-0-dev is already the newest version (2:1.0.26-1).
pkg-config is already the newest version (1.8.1-1).
libgtk-3-dev is already the newest version (1:3.24.38-2~deb12u1+rpt5+rpi1).
The following packages were automatically installed and are no longer required:
  libbsd0:armhf libhwloc-plugins:armhf libhwloc15:armhf libicu72:armhf liblzma5:armhf libmd0:armhf libpciaccess0:armhf libraspberrypi0 libstdc++6:armhf libtbb12:armhf libtbbbind-2-5:armhf
  libtbbmalloc2:armhf libudev1:armhf libx11-6:armhf libxau6:armhf libxcb1:armhf libxdmcp6:armhf libxext6:armhf libxml2:armhf libxnvctrl0:armhf ocl-icd-libopencl1:armhf zlib1g:armhf
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  libssl-doc
The following NEW packages will be installed:
  libssl-dev libudev-dev
0 upgraded, 2 newly installed, 0 to remove and 15 not upgraded.
Need to get 2,355 kB of archives.
After this operation, 12.9 MB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 libudev-dev arm64 252.22-1~deb12u1 [52.1 kB]
Get:2 http://archive.raspberrypi.com/debian bookworm/main arm64 libssl-dev arm64 3.0.11-1~deb12u2+rpt1 [2,303 kB]
Fetched 2,355 kB in 3s (677 kB/s)     
Selecting previously unselected package libssl-dev:arm64.
(Reading database ... 158125 files and directories currently installed.)
Preparing to unpack .../libssl-dev_3.0.11-1~deb12u2+rpt1_arm64.deb ...
Unpacking libssl-dev:arm64 (3.0.11-1~deb12u2+rpt1) ...
Selecting previously unselected package libudev-dev:arm64.
Preparing to unpack .../libudev-dev_252.22-1~deb12u1_arm64.deb ...
Unpacking libudev-dev:arm64 (252.22-1~deb12u1) ...
Setting up libssl-dev:arm64 (3.0.11-1~deb12u2+rpt1) ...
Setting up libudev-dev:arm64 (252.22-1~deb12u1) ...
Processing triggers for man-db (2.11.2-2) ...
-bash: ./scripts/setup_udev_rules.sh: No such file or directory
mkdir: cannot create directory ‘build’: File exists
CMake Warning:
  Ignoring extra path from command line:

   ".."

CMake Error: The source directory "/home" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```

Ill rectify with my Network Team and get back to you @MartyG-RealSense. 

I feel like my proxy is blocking Upgrades from happening

MartyG-RealSense, 2024-05-08:
The **distribution_linux.md** package installation instructions should not be used with computers with an Arm architecture such as Raspberry Pi as they are designed for use with x86 / x64 architecture computers such as desktop and laptop PCs.

A method of installing librealsense on Raspberry Pi that historically works well for Pi is **libuvc backend**.  This method can be performed with a simple build script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Sharath-MRNoBoDy, 2024-05-13:
Hi @MartyG-RealSense 
I iam not able to install pyrealsense2 in the same RPI 4B 
I am able to install numpy and other packages in to the same device but the pyrealsense2 package is not able to be found iam getting this error is i try to install the same 

```
root@ubuntu:/home/ubuntu/Desktop# source testpip/bin/activate
(testpip) root@ubuntu:/home/ubuntu/Desktop# pip3 install numpy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting numpy
  Downloading numpy-1.26.4-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (14.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 14.2/14.2 MB 11.0 MB/s eta 0:00:00
Installing collected packages: numpy
Successfully installed numpy-1.26.4
(testpip) root@ubuntu:/home/ubuntu/Desktop# pip3 install pyrealsense2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
(testpip) root@ubuntu:/home/ubuntu/Desktop# pip3 install pyrealsense2-aarch64
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement pyrealsense2-aarch64 (from versions: none)
ERROR: No matching distribution found for pyrealsense2-aarch64

```

MartyG-RealSense, 2024-05-13:
Hi @Sharath-MRNoBoDy  The **pip install pyrealsense2** command does not work on Raspberry Pi, unfortunately.  The wrapper must be built from source code instead.

The easiest way to do this on Pi may be to edit the libuvc backend procedure's build script to add commands to build the pyrealsense2 wrapper at the same time that librealsense is built.

Change line 46 from this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

to this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS:bool=true`

Sharath-MRNoBoDy, 2024-05-17:
I tried the command u have mentioned and its still not able to install the same

MartyG-RealSense, 2024-05-17:
A RealSense user at https://github.com/IntelRealSense/librealsense/issues/11506#issuecomment-1599168850 provided a detailed guide about how they set up librealsense and pyrealsense2 on a Pi 4 with Ubuntu.

Sharath-MRNoBoDy, 2024-05-22:
By following your advice to work with ubuntu 
I started to work on the Ubuntu in Raspberry pi4b 
I  was able to install and build the required libraries and packages in the RPI4B 

now I am back to same error stating while using the terminal SSH to RPI4B with UBUNTU 

````
Traceback (most recent call last):
  File "/home/ubuntu/Desktop/realsense/realsense/temp.py", line 15, in <module>
    frame = pipe.wait_for_frames(10000)
RuntimeError: Frame didn't arrive within 10000

and while using the same RPI Physically, Iam getting a different error message

MartyG-RealSense, 2024-05-22:
What happens if you take the 10000 out of the brackets?

` frame = pipe.wait_for_frames()`

Sharath-MRNoBoDy, 2024-05-22:
If i remove the 10000 out form the brackets 
then I get the 

`
RuntimeError : Frame didn't arrive within 5000 
`
that's the reason I increased to 10000

MartyG-RealSense, 2024-05-22:
'5000' means that new frames did not arrive from the camera during a 5 second period of time.  10000 means new frames did not arrive for 10 seconds.  

Does the behaviour of the camera improve if you unplug the micro-sized end of the USB cable from the base of the D457 camera, reverse the connector's orientation and plug it back into the camera (USB-C cables are two-way insertion at the micro-sized end).

Sharath-MRNoBoDy, 2024-05-24:
As you suggest to flip the cable and try iam still getting the same error. 
I also tried to execute a test program from  [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline) 

I am still getting the same error following with  while using ssh 
``` Traceback (most recent call last):
  File "/home/ubuntu/Desktop/sharath /realsense/realsense/test.py", line 11, in <module>
    frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000
```
and  while using it physically 
``` 
Line 6 , in <module>
pipeline.start()
RuntimeError: no device Connected
 ```

with the test program from [here](https://github.com/IntelRealSense/librealsense/issues/12886#issuecomment-2087965849) also tried to execute a test program from  [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline) while running the same code in the PC it is running but  when the same code is executed in the RPI 4B them iam getting both errors

MartyG-RealSense, 2024-05-24:
As you are using RSUSB = true (which should take care of any potential conflicts between librealsense and the kernel) and the scripts that you have tested work correctly on a PC, it is possible that the problem is with the Bookworm version of the Ubuntu OS on your Pi.  

Some RealSense users with a Pi use Ubuntu 22.04, like the user mentioned earlier in this discussion at https://github.com/IntelRealSense/librealsense/issues/11506#issuecomment-1599168850 who successfully used **wait_for_frames(10000)**

Sharath-MRNoBoDy, 2024-06-12:
Hi Marty-G
I tried the method u had mentioned but still not able to retrieve the data from the camera but instead it shows the no device is connected 

```
Traceback (most recent call last):
  File "/home/ubuntu/temp.py", line 12, in <module>
    pipe.start(cfg)
RuntimeError: No device connected
```
But iam still running the same code even changing to **wait_for_frames(5000)** to **wait_for_frames(10000)** like mention the last comment

MartyG-RealSense, 2024-06-12:
What happens if you use try_wait_for_frames() so that the **Frame didn't arrive** type of RuntimeError does not cause the program to exit running? 

`frame = pipe.try_wait_for_frames(10000)`

MartyG-RealSense, 2024-06-14:
As your code works on a desktop PC but not Pi, this indicates that the script is okay.  

As the Viewer and SDK works but pyrealsense2 does not, this indicates that the Bookworm OS is not the cause of the problem.

Although the `pip install pyrealsense2` installation method supports Arm devices such as Pi, other RealSense users with a Pi have consistently reported that the pip install method's Arm package does not work with Pi boards.  This suggests that a **source code** installation of pyrealsense2 on Pi might be required instead.

https://github.com/IntelRealSense/librealsense/issues/12886#issuecomment-2124257588 suggests to me that you have already tried building pyrealsense2 from source code though.  Is this correct, please?

Sharath-MRNoBoDy, 2024-06-18:
Yes @MartyG-RealSense  I did completely build the packages for the Pyrealsense2 form the source code but still the problem arises in both the Ubuntu and even on the Raspbian OS.
Even if do this 
```python3 
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2
>>> import pyrealsense2 as rs 
>>> import numpy 
```
though the Pyrealsense is getting imported in the pyhton3 command but when ever i try to run the programs it show device not connected and even in the SDK application it shows **Intel RealSense device Detected** and even preview the fedd form the sdk

MartyG-RealSense, 2024-06-18:
When the camera works with SDK applications such as the Viewer but it is not detected by the pyrealsense2 wrapper, it can unfortunately be very difficult to solve.

A Python 3.11 user at https://github.com/IntelRealSense/librealsense/issues/3062#issuecomment-1814386407 shared what worked in their particular case.

Sharath-MRNoBoDy, 2024-11-25:
we are writing a separate programs for to capture a snap of the 3 fields separately but the i want them to be auto clibrated rather than using the GUI to Calibrate the camera
guide with a camera correction in python program 
it would be really helpfull

MartyG-RealSense, 2024-11-25:
The RealSense SDK has an example of a Python automatic calibration program at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/auto_calibration_config.json

closed 2024-12-02 by Sharath-MRNoBoDy