realsenseai/librealsense #12889: Problem D345i

LauGasp, 2024-04-30:
Good afternoon,

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |    Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |        5.15.0-102-generic                                   | 
| Platform                                 | Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz  |
| SDK Version                            | 2.54.2.0 ??   | 
| ROS Distro			         |  noetic                                          | 
| RealSense ROS Wrapper Version   |  2.3.2                                                  | 

I had a problem with the realsense D435i that I never got. Whenever I ran roslaunch realsense2_camera rs_d435_camera_with_model.launch with <arg name="enable_gyro" default="true"/> and <arg name="enable_accel" default="true"/> and <arg name="unite_imu_method" default="linear_interpolation"/>, I was able to view the contents of the /camera/imu topic (with rostopic echo and rostopic hz). However, since yesterday, when I open realsense-viewer, the imu doesn't always appear and some errors appear.

```
... logging to /home/lauragaspar/.ros/log/cf822320-06ff-11ef-8d40-d380674235d4/roslaunch-Asus-ROG-GL504GW-31837.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://Asus-ROG-GL504GW:35211/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
[... 155 lines cut ...]
 30/04 16:06:51,920 ERROR [140294713915264] (backend-v4l2.cpp:586) Cannot access /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video3/uevent
 30/04 16:06:51,920 ERROR [140294713915264] (librealsense-exception.h:52) Unable to find dev_name from video_path: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video3 Last Error: No such file or directory
 30/04 16:06:51,920 INFO [140294713915264] (backend-v4l2.cpp:1030) Not a USB video device: Unable to find dev_name from video_path: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video3 Last Error: No such file or directory
 30/04 16:06:51,920 ERROR [140294713915264] (backend-v4l2.cpp:586) Cannot access /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video4/uevent
 30/04 16:06:51,920 ERROR [140294713915264] (librealsense-exception.h:52) Unable to find dev_name from video_path: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video4 Last Error: No such file or directory
 30/04 16:06:51,920 INFO [140294713915264] (backend-v4l2.cpp:1030) Not a USB video device: Unable to find dev_name from video_path: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video4 Last Error: No such file or directory
 30/04 16:06:51,920 ERROR [140294713915264] (backend-v4l2.cpp:586) Cannot access /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video5/uevent
 30/04 16:06:51,921 ERROR [140294713915264] (librealsense-exception.h:52) Unable to find dev_name from video_path: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video5 Last Error: No such file or directory
 30/04 16:06:51,921 INFO [140294713915264] (backend-v4l2.cpp:1030) Not a USB video device: Unable to find dev_name from video_path: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video5 Last Error: No such file or directory
 30/04 16:06:51,921 INFO [140294713915264] (backend-v4l2.cpp:753) Enumerating UVC video6 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.3/video4linux/video6
 30/04 16:06:51,921 INFO [140294713915264] (backend-v4l2.cpp:753) Enumerating UVC video7 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.3/video4linux/video7
 30/04 16:06:51,921 INFO [140294713915264] (context.cpp:116) ... /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.3/video4linux/video6
 30/04 16:06:51,921 INFO [140294713915264] (context.cpp:116) ... /sys/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/video4linux/video0
 30/04 16:06:51,921 INFO [140294713915264] (context.cpp:128) Found 2 RealSense devices (0xff requested & 0xff from device-mask in settings)
```
or
```
30/04 16:09:32,061 ERROR [139873620707200] (rs.cpp:237) [rs2_create_context_ex( api_version:25501, json_settings:{ ) UNKNOWN] failed to load configuration file (/home/lauragaspar/.realsense-config.json): [json.exception.parse_error.101] parse error at line 1, column 1: attempting to parse an empty input; check that your input string or stream contains the expected JSON
 30/04 16:09:32,061 ERROR [139873620707200] (rs.cpp:237) [rs2_delete_context( context:nullptr ) UNKNOWN] null pointer passed for argument "context"
 30/04 16:09:32,061 WARNING [139873620707200] (rs.cpp:274) null pointer passed for argument "context"
RealSense error calling rs2_create_context_ex(api_version:25501, json_settings:{):
    failed to load configuration file (/home/lauragaspar/.realsense-config.json): [json.exception.parse_error.101] parse error at line 1, column 1: attempting to parse an empty input; check that your input string or stream contains the expected JSON

```
I've tested it on 3 different computers and with 2 realsenses and usb cables... What do you think could be the problem?

MartyG-RealSense, 2024-04-30:
Hi @LauGasp  I note that you are using camera firmware driver **5.16.0.1**.  This is the correct firmware version when using RealSense Viewer version 2.55.1.  

However, this creates problems when using the camera with the RealSense ROS1 wrapper and librealsense SDK version 2.50.0.  This is because the recommended firmware for 2.50.0 is **5.13.0.50** and using newer firmwares such as 5.16.0.1 with 2.50.0 may cause errors.

As the firmware is stored within the camera, this unfortunately usually places one in the position where you have to either use the ROS1 wrapper or the Viewer, but cannot use the camera with both.  It will not be possible to switch back and forth between firmware versions, because the cameras have a firmware downgrade limit where the ability to downgrade to an older firmware is permanently locked after 20 downgrades.

Even if you have previously been able to use the 2.3.2 ROS1 wrapper with a newer firmware than 5.13.0.50, I would not recommend doing so.  The ROS1 wrapper is not dependent on streams being enabled in the Viewer, so ideally the Viewer should be closed when performing a ROS launch.  Please test whether you can roslaunch successfully when the Viewer is not open.

Something else that I would recommend against is having depth, RGB and Motion Module all enabled at the same time in the Viewer.  This is because for some RealSense users the Viewer has a known issue that has no fix where enabling these three streams can cause one of the stream types to stop working.  The trigger for the issue is the enabling of the Motion Module, so changing the order that depth, RGB and Motion Module are enabled in will not make a difference.  So to avoid this issue occurring I would recommend only enabling two stream types at a time (depth and Motion Module, depth and RGB, etc).

LauGasp, 2024-04-30:
Hi @MartyG-RealSense , I believe downgrading the firmware to 5.13.0.50 did the trick. Thank you so much for your help!

MartyG-RealSense, 2024-05-01:
You are very welcome, @LauGasp - it's excellent to hear that your issue was resolved by a firmware downgrade.  Thanks very much for the update!

closed 2024-05-08 by MartyG-RealSense