realsenseai/librealsense #12890: Process Killed RealSense SDK Mac M2

MhmRhm, 2024-05-01:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457 | 
| Firmware Version                   | 5.16.0.1 | 
| Platform                                 | M2 Mac |
| Operating System & Version | MacOS 14.4.1 | 
| SDK Version                            | 2.55.1 | 
| Language                            | C | 

### Issue Description

I have followed [this instructions](https://lightbuzz.com/realsense-macos/) to build RealSense SDK for Mac.
I also copied the **libusb-1.0.0.dylib** next to **rs-enumerate-devices** in _/librealsense/build/RELEASE/Release_.

The **rs-enumerate-devices** exits without any error output while run from terminal with sudo. 
When running from Xcode it crashes at `PTHREAD_CHECK (pthread_mutex_lock(mutex) );`

The following image shows the situation.

[image]

MartyG-RealSense, 2024-05-01:
Hi @MhmRhm  May I first confirm that you have set the position of a **mode switch** in the casing of your D457 to work with a USB cable, please?  By default the camera is set to work with a non-USB connection type called **GMSL/FAKRA**.  If you have not set the mode switch to the USB position yet then instructions are provided at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/14840675121043/comments/14843232242835

MhmRhm, 2024-05-01:
Dear @MartyG-RealSense thanks for your fast response. 

The switch is on USB.
On a freshly reset Mac, after doing `brew install librealsense` I get the same Killed message. 
In a Ubuntu 22.04 image on UTM everything works and I get the expected result:
(I must mention that in the tutorial for [Linux manual installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) I did not Build and apply patched kernel modules (Step 3 under Install librealsense2)

[image]

MartyG-RealSense, 2024-05-01:
I note that in the log at the bottom of your Mac screen you have **Failed to set power state** and **No device detected - is it plugged in?** errors, which can occur if the camera cannot be detected or cannot be accessed.  

On your Mac are you using the official USB A to C cable supplied with the camera or a USB C to C cable of your own choice, please?  If it is a C to C cable then problems are more likely to occur with RealSense cameras, especially on Mac.

MhmRhm, 2024-05-01:
Dear @MartyG-RealSense there are no USB A to C cables supplied with the camera. Only one GMSL/FAKRA cable. I am using an official USB Type C to C cable supplied with my iPad mini. Same cable that I used in the UTM image and it is working there. 
I also rotated the Type C cable on the device end. Apparently for D457 USB port, only one set of RX/TX pins implemented in Type C.

MartyG-RealSense, 2024-05-02:
Thanks very much @MhmRhm for the reminder that D457 does not come with a USB-C cable.  I use a cable from another RealSense camera with mine.

Yes, with the D457 model the micro-sized end of the USB cable has to be inserted into the camera base in one particular direction of the two insertion directions available in order for the camera to operate at USB 3.2 speed, otherwise it is detected as USB 2.1 when inserted in the other direction.

Aside from the known reliability issues when using a C to C cable with RealSense cameras, the cable that is chosen to use with the D457 should also be designed for data transfer and not just be a charging / tablet to computer device sync cable.

MhmRhm, 2024-05-03:
Dear @MartyG-RealSense thanks for your patience. I managed to test with a USB4 Gen3 40Gbps cable. Same outcome. In the attache image you can see that my virtual machine is recognizing the USB 3.2 but Mac failed to run the enumeration application.

[image]

MartyG-RealSense, 2024-05-03:
You could next check Sonoma's privacy and security settings to see whether there is a setting that might be blocking access to the camera.

https://support.apple.com/en-gb/guide/mac-help/mchlf6d108da/mac

MhmRhm, 2024-05-07:
In FaceTime app, I can select the RealSense camera and it is functional.
When I run realsense-viewer I get the following segmentation fault:

[image]
[image]

MartyG-RealSense, 2024-05-08:
This error has been previously reported with a RealSense camera on an Apple Silicon Mac at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4548413451539/comments/27039742754195

My research indicates that it is not a RealSense-only issue but also occurs on MacOS in non-RealSense software that uses GLFW graphics such as the game Minecraft.  In the small number of cases where a workaround was developed, it involved creating a custom 'forked' version of GLFW with the problematic code edited out, like the case at https://github.com/LWJGL/lwjgl3/issues/695

I appreciate that this is something that the majority of RealSense MacOS users may not want to do or be able to do.

MhmRhm, 2024-05-08:
Dear @MartyG-RealSense thank you so much for that. Hopefully, soon I will try the solution.

MartyG-RealSense, 2024-05-23:
Hi @MhmRhm  Do you have an update about your case that you can provide, please?  Thanks!

MhmRhm, 2024-05-24:
Dear @MartyG-RealSense I followed [this](https://lightbuzz.com/realsense-macos/) tutorial and was unable to connect. I am continuing in Linux VM on my Mac. Thank you for following up.

MartyG-RealSense, 2024-05-24:
A RealSense Mac user had success with using a type of Linux called **Asahi** on their Apple Silicon Mac and wrote an installation guide about the procedure that they used.

https://github.com/HasanTheSyrian/librealsense-hasan/blob/master/doc/installation_asahi.md

MartyG-RealSense, 2024-05-30:
Hi @MhmRhm  Do you require further assistance with this case, please?  Thanks!

MhmRhm, 2024-05-30:
Dear @MartyG-RealSense thank you for all the help so far. 
I believe I will not need more assistance.

MartyG-RealSense, 2024-05-30:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks very much for the update!

closed 2024-05-30 by MartyG-RealSense

MhmRhm, 2024-09-05:
Dear @MartyG-RealSense thanks for your great support. That is great news. I will try that as soon as I can. Thank you so much again.