realsenseai/librealsense #12892: context.cpp:129 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)

Manat09, 2024-05-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-05-01:
Hi @Manat09  Please first check the camera privacy settings in Windows to ensure that permission for applications to access the camera is set to On.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

If the privacy settings are already set to On, next try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera (USB-C cables are two-way insertion at the micro-sized end).

Manat09, 2024-05-01:
> Hi @Manat09 Please first check the camera privacy settings in Windows to ensure that permission for applications to access the camera is set to On. Instructions for doing so can be found at [#3381 (comment)](https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789)
> 
> If the privacy settings are already set to On, next try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera (USB-C cables are two-way insertion at the micro-sized end).

Thank you for answering! But it did not help anyway.

lipilian, 2024-05-01:
If you check latest version of librealsense2, I think they remove the support for L515

Manat09, 2024-05-01:
> If you check latest version of librealsense2, I think they remove the support for L515

Oh I see yeah, thank you very much!

zhanghua7099, 2024-05-02:
How to solve this problem...

MartyG-RealSense, 2024-05-02:
@lipilian is correct that L515 support was removed from the current latest RealSense SDK version 2.55.1 as it is a retired camera model.

If you are using Windows and need to use L515 then you can download and install the previous SDK version 2.54.2 which still had L515 support.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

Manat09, 2024-05-07:
> @lipilian is correct that L515 support was removed from the current latest RealSense SDK version 2.55.1 as it is a retired camera model.
>
> If you are using Windows and need to use L515 then you can [...]

Hello, can we install from this website also to Linux?

MartyG-RealSense, 2024-05-07:
Hi @Manat09  Building librealsense from its GitHub page would build the latest 2.55.1 version.

You may be able to install from packages instead of source code if you use the command **apt-cache showpkg librealsense2** to show a list of librealsense package versions and install the 2.54.2 version of the packages, as described at https://github.com/IntelRealSense/librealsense/issues/10526#issuecomment-1129748073

The full instructions for a package installation can be found at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

In the section of the instructions where it asks you to install the following two packages:

```
sudo apt-get install librealsense2-dkms
sudo apt-get install librealsense2-utils
```

use the 2.54.2 names of these packages obtained with 'apt-cache showpkg librealsense2'

MartyG-RealSense, 2024-05-13:
Hi @Manat09  Do you require further assistance with this case, please?  Thanks!

Manat09, 2024-05-13:
> Hi @Manat09  Do you require further assistance with this case, please?  Thanks!

Hi, thank you so much, now I didn't started the project with your methods. Later I will try and give you feedback.

MartyG-RealSense, 2024-05-13:
You are very welcome.  I look forward to your feedback later.  Good luck!

Manat09, 2024-05-15:
> Hi @Manat09 Building librealsense from its GitHub page would build the latest 2.55.1 version.
> 
> You may be able to install from packages instead of source code if you use the command **apt-cache showpkg librealsense2** to show a list of librealsense package versions and install the 2.54.2 version of the packages, as described at [#10526 (comment)](https://github.com/IntelRealSense/librealsense/issues/10526#issuecomment-1129748073)
> 
> The full instructions for a package installation can be found at the link below.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
> 
> In the section of the instructions where it asks you to install the following two packages:
> 
> ```
> sudo apt-get install librealsense2-dkms
> sudo apt-get install librealsense2-utils
> ```
> 
> use the 2.54.2 names of these packages obtained with 'apt-cache showpkg librealsense2'

Hi, after using this command: `sudo apt-get install librealsense2-utils=2.54.2-0~realsense0.5307 librealsense2-dkms=2.54.2-0~realsense0.5307`
I got next output: 
`E: Version '2.54.2-5ubuntu5~1.gbpe9f05c' for 'librealsense2' was not found
E: Version '2.54.2-5ubuntu5~1.gbpe9f05c' for 'librealsense2-dev' was not found`
Do you know how to solve it?
Thank you!

MartyG-RealSense, 2024-05-15:
If the packages cannot be found then there is not a way to solve it, unfortunately.  

The only option left available would be to first remove all RealSense packages from your computer with the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Then visit the SDK 'Releases' page for 2.54.2 at the link below and use the 'Assets' list at the bottom of the page to download the 'Source code (zip)' file to your computer.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

[image]

Finally, extract the librealsense source code folder from the zip and use the folder to build the source code with CMake using the instructions at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

Manat09, 2024-05-18:
> If the packages cannot be found then there is not a way to solve it, unfortunately.
>
> The only option left available would be to first remove all RealSense packages from your computer with the [...]

Thank you very much!!! Everything is working now. With this version I can use my camera l515.

MartyG-RealSense, 2024-05-18:
You are very welcome.  That's excellent news, @Manat09 - thanks very much for the update!

Manat09, 2024-05-26:
> Hi @Manat09  Do you require further assistance with this case, please?  Thanks!

No, currently I don't have problems, thank you!

MartyG-RealSense, 2024-05-26:
You are very welcome.  Thanks very much for the update. As you do not currently require further assistance, I will close this case.  Thanks again!

closed 2024-05-26 by MartyG-RealSense