realsenseai/librealsense #12894: pyrealsense2 no device found

szymonzmyslony, 2024-05-03:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |  D435i                         | 
| Firmware Version                   | 5.15.0.2                        | 
| Operating System & Version |  Ubuntu 22  
| Kernel Version (Linux Only)    | 5.15.122                                                       | 
| Platform                                 |   NVIDIA Jetson Orin Nano               |
| Librealsense SDK Version       |   2.54.1                                                   | 
| Language                               |  python        | 
| ROS Distro			         |  Humble                               | 
| RealSense ROS Wrapper Version   | 4.54.1                                          | 

### Issue
I installed the realsense library from source with ```-DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true ``` and completed all instructions based on [this comment](https://github.com/IntelRealSense/librealsense/issues/12831#issuecomment-2041140925). Realsense GUI and cmd tools work perfectly but the python wrapper doesn't detect (or work at all) on my Jetson device 

 Crossposting from [ros-repo](https://github.com/IntelRealSense/realsense-ros/issues/3091), closing old issue

MartyG-RealSense, 2024-05-03:
Hi @szymonzmyslony  The Jetson source code installation guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 recommends adding definitions for the PYTHONPATH to the end of the bashrc file after initiating the `sudo make install` command and then sourcing the file.

szymonzmyslony, 2024-05-03:
the binaries for Python wrapper do not exists anywhere (I also don't get any errors / warnings associated with it during build time), is there a way to find a workaround here ?

MartyG-RealSense, 2024-05-03:
Instead of building the Python wrapper at the same time as the librealsense SDK, you can build librealsense first without the wrapper and then install the wrapper separately from a **.whl** package with the command `pip install pyrealsense2`

Whilst pip install supports installation on Arm devices with an **aarch64**  wheel file, Jetson users have often found that a pip install does not work for them.  You are welcome to try it on your own Jetson, of course.  A Jetson user at https://github.com/IntelRealSense/librealsense/issues/12362#issuecomment-1795092552 apparently did manage to make it work.

furiousteabag, 2024-05-03:
We tried to use python3.9 with ` python3.9 -m pip install pyrealsense2==2.54.1.5217` and getting the same `No device connected` problem

MartyG-RealSense, 2024-05-04:
One of my Intel RealSense colleagues advised that if there is not a suitable official pyrealsense2 pip installation package file available then you can build the pyrealsense2 wrapper from source code and then upload it as your own Python package.  The link below has a guide for creating and uploading packages to PyPi.

https://www.turing.com/kb/how-to-create-pypi-packages

furiousteabag, 2024-05-04:
We have suitable pyrealsense2 pip installation `python3.9 -m pip install pyrealsense2==2.54.1.5217`, but it just doesn't work:

```
    pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected
```

MartyG-RealSense, 2024-05-04:
A RealSense Jetson user at https://github.com/IntelRealSense/librealsense/issues/12807#issuecomment-2032658916 who recently experienced `RuntimeError: No device connected` managed to get the pyrealsense2 wrapper working by using JetPack 5.1.3 and Python 3.8 after initially having the error with JetPack 6.

furiousteabag, 2024-05-04:
I see, but downgrading JetPack is not an option, because many other libraries and programs depend on it. What can I try to do to make it work on JetPack 6.0?

MartyG-RealSense, 2024-05-04:
The latest 2.55.1 version of the librealsense SDK supports JetPack 6.  The RealSense user at https://github.com/IntelRealSense/librealsense/issues/12807 had **RuntimeError: No device connected** with JetPack 6 and the pyrealsense2 wrapper, though at that time they were using librealsense 2.54.2 because the Jetpack 6 supporting 2.55.1 had not been released yet.  So it may be worth trying 2.55.1 with JetPack 6.

MartyG-RealSense, 2024-05-10:
HI @furiousteabag  Do you require further assistance with this case, please?  Thanks!

furiousteabag, 2024-05-10:
Hey Marty! Tomorrow we will be trying to install 2.55.1 with Jetpack 6 and pyrealsense2. Will keep you posted here!

MartyG-RealSense, 2024-05-10:
Thanks very much @furiousteabag for the update - good luck!

thomasaab, 2024-05-15:
Hello, I am having exactly the same problem. Were you able to solve it? I have version 2.55.1 installed, with Jetpack 6, the same camera with firmware 5.16.0.1. The command "rs-enumerate-devices" recognizes the device, but pyrealsense2 does not. I have already tried different solutions that I have found, but I really don't want to downgrade to Jetpack 5. 

Any help would be greatly appreciated!

furiousteabag, 2024-05-15:
@thomasaab We haven't tried yet with 2.55.1, but keep us posted if you will have some progress!

shankar-anantak, 2024-05-16:
Hello, I am using Orin dev kit with Jetpacck 6.

Ive been trying to build 2.55.1 with python bindings but have been unsuccessful:

```
Traceback (most recent call last):
  File "/home/ubuntu/dev/Pipelines/hawkeye/empties_grid/empties_grid.py", line 81, in <module>
    pipeline.start(config)
RuntimeError: No device connected
```

I've tried almost every solution referenced. Any assistance is appreciated

MartyG-RealSense, 2024-05-17:
Hi @shankar-anantak  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/12807 who had `RuntimeError: No device connected` with JetPack 6 downgraded to JetPack 5.1.3.  Are you able to downgrade your JetPack or is this not an option in your particular situation, please?

shankar-anantak, 2024-05-17:
> Hi @shankar-anantak A RealSense user at #12807 who had `RuntimeError: No device connected` with JetPack 6 downgraded to JetPack 5.1.3. Are you able to downgrade your JetPack or is this not an option in your particular situation, please?

Hi @MartyG-RealSense, unfortunately downgrading is not an option at the moment as JP6 contains updated versions for some of the other frameworks I need.

Again, any assistance you can provide is appreciated!

MartyG-RealSense, 2024-05-18:
Hi @shankar-anantak  Have you tried the JetsonHacks website's **buildLibrealsense.sh** librealsense source-code build script for Jetson boards?  The script includes the **-DBUILD_PYTHON_BINDINGS=bool:true** build flag for installing the pyrealsense2 wrapper at the same time as compiling librealsense.

https://github.com/jetsonhacks/installRealSenseSDK/tree/master?tab=readme-ov-file#buildlibrealsensesh

MartyG-RealSense, 2024-05-26:
Hi @shankar-anantak  Do you have an update about your problem that you can provide, please?  Thanks!

MartyG-RealSense, 2024-06-10:
Case closed due to no further comments received.

closed 2024-06-10 by MartyG-RealSense

Transformer-LM, 2024-07-23:
> > 你好@shankar-anantak#12807的 RealSense 用户已将`RuntimeError: No device connected`JetPack 6 降级至 JetPack 5.1.3。您能降级您的 JetPack 吗？或者这对您的特定情况来说不是一个选择？
> 
> 你好@MartyG-RealSense，不幸的是，目前无法降级，因为 JP6 包含我需要的其他一些框架的更新版本。
> 
> 再次感谢您提供的任何帮助！
> 
> 对于上下文：
> 
> Jetson Orin 开发套件、Ubuntu 22.04、JP6、Python3.10、Cuda 12
> 
> 我尝试过 LIBUVC、RSUSB 方法，使用和不使用内核补丁脚本，从源代码构建最新的 librealsesnse，更新 cmake 方法，将 .so 文件复制到系统各个位置并添加到路径等
您好，请问那您是如何解决这个问题的？选择降级jetson?

MartyG-RealSense, 2024-07-23:
For users with JetPack 6 and a RealSense camera equipped with an IMU, the RealSense MIPI driver at the link below needs to be installed in order for cameras with an IMU to be supported, such as D435i and D455.

 https://github.com/IntelRealSense/realsense_mipi_platform_driver

********

对于使用 JetPack 6 和配备 IMU 的 RealSense 相机的用户，需要安装以下链接中的 RealSense MIPI 驱动程序，以便支持配备 IMU 的相机，例如 D435i 和 D455。

https://github.com/IntelRealSense/realsense_mipi_platform_driver