realsenseai/librealsense #12896: ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2' on Ubuntu

LeVHoangduc, 2024-05-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515| 
| Firmware Version                   |  | 
| Operating System & Version |  Linux (Ubuntu 20.04)| 
| Kernel Version (Linux Only)    |  5.15                                         | 
| Platform                                 | PC  |
| SDK Version                            |   }                          | 
| Language                            |  python }                          | 
| Segment			|  Robot                          | 

### Issue Description
Here's what I did:
Step1: Download zip [librealsense](https://github.com/IntelRealSense/librealsense)
Step2: At the root folder
- mkdir build
- cd build
- cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true
- make -j4
- export PYTHONPATH=$PYTHONPATH:/home/dsp520/anaconda3/envs/graspness2/lib/python3.7
- sudo make install

And I run this code:
```
import pyrealsense2 as rs
import numpy as np
import cv2

pipe = rs.pipeline()
cfg  = rs.config()
cfg.enable_stream(rs.stream.color, 960,540, rs.format.bgr8, 30)
cfg.enable_stream(rs.stream.depth, 640,480, rs.format.z16, 30)
pipe.start()
while True:
    frame = pipe.wait_for_frames()
    depth_frame = frame.get_depth_frame()
    color_frame = frame.get_color_frame()

    depth_image = np.asanyarray(depth_frame.get_data())
    color_image = np.asanyarray(color_frame.get_data())
    depth_cm = cv2.applyColorMap(cv2.convertScaleAbs(depth_image,
                                     alpha = 0.5), cv2.COLORMAP_JET)

    gray_image = cv2.cvtColor(color_image, cv2.COLOR_BGR2GRAY)

    cv2.imshow('rgb', color_image)
    cv2.imshow('depth', depth_cm)

    if cv2.waitKey(1) == ord('q'):
        break
pipe.stop()
```

 I got an error:
[image]

I checked at the wrappers/python/pyrealsense2 folder, and this is an error:
[image]

I have tried to "import pyrealsense2 .pyrealsense2 as rs" but It was still error.

LeVHoangduc, 2024-05-04:
For more information.
I just have these files .so after building
[image]

MartyG-RealSense, 2024-05-04:
Hi @LeVHoangduc  You can confirm whether your PYTHONPATH is correct with the command **which python3**

If you cannot get the PYTHONPATH to work then you can get around this obstacle by copying the pyrealsense2 .so file highlighted in the image above and also the file **librealsense2.so** into the same folder that your Python script file is in.  Typically after having built the Python wrapper from source code, these two files will be found at the location below.

**librealsense > build > wrappers >python**

[image]

Which source code version of librealsense are you using, please?  Support for L515 was removed in the current latest 2.55.1 SDK, so source code version 2.54.2 or older should be used with your camera.  All librealsense versions and their source code folder can be found here:

https://github.com/IntelRealSense/librealsense/releases

LeVHoangduc, 2024-05-04:
Thanks a lot for your help! Could you help me to clarify sth after I do next step!
I used the latest current version to build it. I'll try to build with [v.54](https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2).2 now.
However, do I need to delete everything before rebuilding?
And is the installation process I mentioned above correct? That means, I'll go to the root folder and do these step:
```
mkdir build
cd build
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true
make -j4
export PYTHONPATH=$PYTHONPATH:/home/dsp520/anaconda3/envs/graspness2/lib/python3.7
sudo make install
```
After downloading the source code!

MartyG-RealSense, 2024-05-04:
If you used the source code folder for 2.55.1 and created a 'build' directory then return to the same build folder and input the CMake command below to uninstall that SDK and clean up CMake.  

`sudo make uninstall && make clean`

Then delete the whole 2.55.1 source code folder and build the SDK again from the 2.54.2 one.

***************

The installation process that you listed look okay, though I would recommend using the CMake build instruction below.

`cmake ../  -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=TRUE -DBUILD_PYTHON_BINDINGS:bool=true`

With this build method you do not need to apply a kernel patch script and it also builds the SDK in Release mode to take advantage of optimizations..

LeVHoangduc, 2024-05-06:
@MartyG-RealSense Thanks a lot for your support! I'm done with 2.54.2 version.

closed 2024-05-06 by LeVHoangduc