realsenseai/librealsense #12898: realsense-viewer doesn't detect D435

A-Robotics-mama, 2024-05-06:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|

Please provide a description of the problem
#### realsense-viewer doesn't detect D435
`$ uname -ar
Linux ubuntu 5.15.122-tegra #1 SMP PREEMPT Mon Dec 18 21:24:25 PST 2023 aarch64 aarch64 aarch64 GNU/Linux`

`$ lsusb | grep -i intel
Bus 002 Device 003: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i`
`$ sudo dmesg | grep -i realsense
[   14.686649] usb 2-3.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[   14.707023] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.4/2-3.4:1.0/input/input8
[   14.708316] usb 2-3.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)`
Please advise, both installations: from source and from git clone https://github.com/IntelRealSense/librealsense.git with the same result
[image]

MartyG-RealSense, 2024-05-06:
Hi @A-Robotics-mama  Is the camera detected on a **libuvc backend** source code build of the SDK if you have not attempted this build method already?

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

A-Robotics-mama, 2024-05-07:
Hi @MartyG-RealSense , I install libuvc but unfortunately result is the same
`$ lsusb | grep Intel
Bus 002 Device 003: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i`
`$ sudo dmesg | grep -i realsense
[   12.346534] usb 2-3.2: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[   12.374245] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.2/2-3.2:1.0/input/input11
[   12.374859] usb 2-3.2: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)`
`$ rs-enumerate-devices 
No device detected. Is it plugged in?`
I rebooted AGX but same result, seems to be a problem according to Arm64 architecture because in parallel system with NUC10 Intel i7 it works fine.

A-Robotics-mama, 2024-05-07:
thank you @MartyG-RealSense , issue resolved, maybe somebody will be interested
disconnect all cameras
git clone https://github.com/JetsonHacks/installRealSenseSDK.git
cd installRealSenseSDK
./installLibrealsense.sh
./buildLibrealsense.sh
realsense-viewer

MartyG-RealSense, 2024-05-08:
Thanks so much @A-Robotics-mama for sharing the method that worked for you.  As you achieved a successful outcome, I will close this case.  Thanks again!