realsenseai/librealsense #12900: No IMU data. Check connectivity

cxychen-hub, 2024-05-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | ( RealSense Viewer v2.54.2) | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.54.2 }                          | 
| Language                            |  { python }                          | 
| Segment			|  { others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

When I run the python 'rs-imu-calibration.py ' ,  there shows

Align to direction:  [ 0. -1.  0.]   Mounting screw pointing down, device facing out

Error: No IMU data. Check connectivity.

The whole information as follows.
[image]

cxychen-hub, 2024-05-07:
Firmware Version: 5.15.1

MartyG-RealSense, 2024-05-07:
Hi @cxychen-hub   Please check whether the USB cable's connector is moving in the USB port when you are changing the position and angle of the camera.  Such movement in the port could cause a disconnection of the camera.

Or does the error message occur even before you have started moving the camera?

cxychen-hub, 2024-05-07:
> Hi @cxychen-hub Please check whether the USB cable's connector is moving in the USB port when you are changing the position and angle of the camera. Such movement in the port could cause a disconnection of the camera.
> 
> Or does the error message occur even before you have started moving the camera?

Thanks. This error occur before I move the camera, it seems that the program reads the camera's parameters but no IMU data. And the camera always connects with the computer. I have tried for many times.

cxychen-hub, 2024-05-08:
> Do you have access to the RealSense Viewer tool? If you do, is the IMU data able to be displayed if you enable the **Motion Module** option in the Viewer's side-panel?

Yes, it can be displayed correctly in RealSence Viewer, Motion Module.

MartyG-RealSense, 2024-05-08:
If you are only going to be using the IMU in the RealSense Viewer then you may not need to perform an IMU calibration with the rs-imu-calibration.py script.  This is because in **Motion Module > Controls** an option called 'Enable Motion Correction' is usually enabled by default for RealSense camera models with an IMU, and this feature 'fixes' the raw data to help to correct inaccuracies.

[image]

cxychen-hub, 2024-05-09:
> If you are only going to be using the IMU in the RealSense Viewer then you may not need to perform an IMU calibration with the rs-imu-calibration.py script. This is because in **Motion Module > [...]

Thanks for your answer. I get it.

MartyG-RealSense, 2025-01-06:
Case closed due to no further comments received.

closed 2025-01-06 by MartyG-RealSense

TheMuhammadRafay, 2025-03-02:
@MartyG-RealSense What if I have to get the imu data outside of realsense-viewer? Especially when performing some computation in ROS Rviz? Would you like me to calibrate the D435i camera using rs-imu-calibration.py? Because I am getting the IMU data in realsense-viewer, but the above error in the python script persists. How to solve this error? or do I need to use the camera without running calibration script?

MartyG-RealSense, 2025-03-02:
Hi @TheMuhammadRafay  The calibration of the IMU is typically very good by default, so it should not be necessary to run the calibration script.  

Using the procedure below in realsense-viewer will provide an indication of your current IMu calibration health.

1.  Go to the Motion Module category in the options side-panel of realsense-viewer and expand it open, then expand open the sub-category **Controls**.

2.  Left-click on the blue icon beside 'Enable Motion Correction' to turn the icon blue, meaning that the motion data correction feature is Off and only unadjusted raw IMU values will be displayed.

[image]

3.  Enable the motion module and look at the Accel panel.  Hover the mouse cursor over the Accel axis-lines graphic to show the IMU values.  If your 'Y' accel value is close to its ideal value of 9.80 (for example, 9.75) then that will be an indication that you have a very good IMU calibration and should not need to re-calibrate it.

************

If your raw Y accel value is not close to 9.80 (such as 9.6) then you could try enabling motion data correction in your ROS launch to bring it closer to the ideal value by adding `enable_motion_correction:=true` to your ROS launch instruction,