realsenseai/librealsense #12901: How to create color point cloud from disparity

Toaster0617, 2024-05-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2... }                          | 
| Language                            |  {python/opencv}                          | 
| Segment			|  {Robot/VR}                          | 

Hi. I'm quite new to using pyrealsense2 library. I having trouble in convert disparity to color point cloud using opencv function (eg. reprojectImageTo3D function). 

I've already get my disparity value by using method in #8572 but my disparity value lies between 0 to 65535. So did I need to scale or convert my result so that it can be used for generate color point cloud by using opencv?

Here is the image I get
[image]

MartyG-RealSense, 2024-05-07:
Hi @Toaster0617  You could convert the RealSense 16-bit **uint16_t** 'pixel depth value' (which ranges from 0 to 65535) to OpenCV's range of 0-255 by converting the depth value to the OpenCV format **CV_16UC1**.  The https://github.com/IntelRealSense/librealsense/issues/8572 case that you quoted has an example of such a conversion in one of its scripts, at https://github.com/IntelRealSense/librealsense/issues/8572#issuecomment-805755799

`disparity = data.view(np.uint16).reshape((imageH, imageW))`

If disparity values are not vital to your project then an alternative approach for obtaining a colorized point cloud with OpenCV code could be the librealsense SDK's **opencv_pointcloud_viewer.py** pyrealsense2 example program.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

[image]

Toaster0617, 2024-05-07:
@MartyG-RealSense Thanks for your help! Does this mean that I only need to convert my data format to 16-bit and scale it to 0-255?

MartyG-RealSense, 2024-05-07:
My understanding is that after RealSense data is converted from 16-but uint16_t to an OpenCV format such as CV_16UC1 then its scale is automatically adjusted without you having to rescale the depth values yourself.

Toaster0617, 2024-05-07:
@MartyG-RealSense Thanks for your response! I can solve my problem now!

closed 2024-05-17 by MartyG-RealSense