realsenseai/librealsense #12904: Obtaining Actual Depth Values with the L515

CCCCCCCT, 2024-05-07:
I currently have a depth map of 320*240, and I also have an RGB image of 640*480. How can I obtain the actual depth values in the depth map captured with my L515?

I can directly see the depth values using an application, but now I want to obtain the depth values from the captured images. I think this is a simple problem, but I've been stuck on it for a while. Please help me resolve this confusion. I would be very grateful.

[image]

[image]

MartyG-RealSense, 2024-05-07:
Hi @CCCCCCCT  The RealSense SDK's **rs-hello-realsense** C++ example program demonstrates printing the actual real-world depth (distance) value in meters as a plain-text value.

https://github.com/IntelRealSense/librealsense/tree/master/examples/hello-realsense

If you have installed the RealSense SDK (librealsense) and its tools and examples then you can find a pre-built executable version of rs-hello-realsense at the following locations on your computer.

**Windows**
C: > Program Files (x86) > Intel RealSense SDK 2.0 >tools

**Linux**
 /usr/local/bin

If you are using the Python language then https://github.com/IntelRealSense/librealsense/issues/9939 provides an example of code for printing the distance value.

MartyG-RealSense, 2024-05-15:
If you are using **get_distance** then you do not need to multiply the values by the depth scale, as get_distance() outputs the real-world distance in meters.

CCCCCCCT, 2024-05-22:
Hello, I can now import photos, but I still can't use python code to get the distance in the photo. Now I can use the L515 to directly turn on the camera to check the distance inside the camera, but this is not what I want.

Now I use opencv to check the depth of the pixel in the photo, but I think this is not right because my depth map is a png file.

Thank you so much for helping out.
[image]

CCCCCCCT, 2024-05-22:
I am thinking whether I should read the raw file, because my Depth file is png

MartyG-RealSense, 2024-05-22:
Saving RealSense depth data to PNG causes a lot of the depth information to be lost, whilst a .raw file preserves the depth.

An example of Python code for exporting depth to .raw that was verified as correct by a RealSense team member is at IntelRealSense/realsense-ros#1312

For C++ users there is an example provided at #1485 (comment)

CCCCCCCT, 2024-05-22:
Thank you very much, I'll do it right away

closed 2024-06-03 by MartyG-RealSense