realsenseai/librealsense #12906: "Error: No RealSense device." when calibrating IMU on Orange Pi 5 Plus

MingyuPan2, 2024-05-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 22.04.4 LTS | 
| Kernel Version (Linux Only)    | 5.10.160-rockchip | 
| Platform                                 | Orange Pi 5 Plus |
| SDK Version                            | Intel RealSense Viewer v2.55.1 | 
| Language                            | python 3.10.12 | 

### Issue Description
Hi, I am trying to calibrate the IMU in the D435i using the given code in /librealsense/tools/rs-imu-calibration. 

Command: python3 rs-imu-calibration.py,

Output: 
waiting for realsense device...

Error: No RealSense device.
Traceback (most recent call last):
  File "/home/mingyupan/Documents/D435i/librealsense/tools/rs-imu-calibration/rs-imu-calibration.py", line 732, in <module>
    main()
  File "/home/mingyupan/Documents/D435i/librealsense/tools/rs-imu-calibration/rs-imu-calibration.py", line 721, in main
    if os.name == 'posix' and old_settings is not None:
UnboundLocalError: local variable 'old_settings' referenced before assignment

It seems that the "serial_no" was not recognized at all when the code was ran. However, in the realsense-viewer, I can view the IMU data perfectly normal, even with Stereo Module with Infrared 1,2 and RGB camera enabled all at the same time. What might be the issue?

MingyuPan2, 2024-05-07:
[image]

Here's what i see in my realsense viewer

MartyG-RealSense, 2024-05-08:
Hi @MingyuPan2  This error with **rs-imu-calibration.py** has not been previously reported. 

Can you confirm that the RealSense Viewer is closed when you launch the calibration script, please?

Some years ago the calibration script used to be important because an IMU calibration needed to be written to the camera hardware before an **Enable Motion Correction** option in 'Motion Module > Controls' would unlock.  This option 'fixes' raw IMU data to make it more accurate.  However, the motion correction option is now usually unlocked and enabled by default on IMU-equipped RealSense camera models, making the calibration script less important now.

[image]

MingyuPan2, 2024-05-08:
Thank you for the reply @MartyG-RealSense! Yes, the Viewer is closed when the script was launched. How much more accurate will the IMU be if i calibrate it? For now, my normal is about 9.74 for Mounting screw pointing down, device facing out. 

I also do not have the Gyro Sensitivity option in my viewer but can access everything else.
[image]

Can this be a hardware compatibility issue? I just don't understand why the Viewer works fine but the script won't.

MartyG-RealSense, 2024-05-08:
The ideal target value is 9.80, so 9.74 is a very good current calibration and you likely do not currently need to perform a calibration of the IMU.

The new variable gyro sensitivity feature requires SDK 2.55.1 and firmware 5.16.0.1.

[image]

Sometimes the IMU calibration script does have problems for unknown reasons and there is not a workaround or solution for it.  As your camera's IMU works fine with the Viewer, it is likely a software problem (SDK, kernel, etc) rather than a hardware one.

When using RealSense cameras with a computing device that has a Rockchip (RK) chipset, it is recommendable to build the RealSense SDK from source code with the **RSUSB backend** installation method.

MingyuPan2, 2024-05-08:
Thank you! I will try the RSUSB backend method tomorrow. 

I also installed ROS2 Iron and the ROS2 wrapper today based on the official instructions. The build process did not give any errors, but when i try ros2 run realsense2_camera realsense2_camera_node, or ros2 launch realsense2_camera rs_launch.py, I get the error:
[INFO] [1715197483.388950003] [camera.camera]: RealSense ROS v4.54.1
[INFO] [1715197483.389225624] [camera.camera]: Built with LibRealSense v2.54.1
[INFO] [1715197483.389279582] [camera.camera]: Running with LibRealSense v2.54.1
 08/05 15:44:43,414 WARNING [281472409657728] (d400-factory.cpp:1195) DS5 group_devices is empty.
[WARN] [1715197483.414158992] [camera.camera]: No RealSense devices were found!

Since my kernel version is 5.10.160-rockchip, is this another software compatibility issue because of the kernel version?

MartyG-RealSense, 2024-05-08:
It could be because it is currently a non-RSUSB installation of librealsense, yes.

You could try performing a reset of the camera at launch with ros2 launch or ros2 run to see whether that makes a difference.

**ROS2 RUN**
`ros2 run realsense2_camera realsense2_camera_node -p initial_reset:=true`

**ROS2 LAUNCH**
`ros2 launch realsense2_camera rs_launch.py initial_reset:=true`

MartyG-RealSense, 2024-05-10:
The error message in your opening comment in this discussion indicates that there is a problem with the **old_settings** instruction in lines 721 and 722 of the rs-imu-calibration script.  What happens if you comment out these two lines and then run the script, please?

MingyuPan2, 2024-05-10:
I tried it a few days before. It just gives the same error without the Traceback (...):

waiting for realsense device...

Error: No RealSense device.

MartyG-RealSense, 2024-05-10:
You already have a very good calibration of 9.74 against an ideal of 9.80.  That value is not likely to degrade over time.  So at this point I would recommend putting the calibration attempts on hold, because what you already have is fine.

MingyuPan2, 2024-05-10:
Forgot to mention, using (sudo rs-enumerate-devices) can give the following output which looks promising...:

~/Documents/D435i/librealsense/build/Release$ sudo rs-enumerate-devices 
Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	238222074578
    Firmware Version              : 	5.15.1
    Recommended Firmware Version  : 	5.16.0.1
    Physical Port                 : 	6-1.4-7
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	224223132975
    Firmware Update Id            : 	224223132975
    Dfu Device Path               : 	

Stream Profiles supported by Stereo Module
 Supported modes:
    STREAM      RESOLUTION     FORMAT      FPS
    Infrared 1  1280x800       Y16         @ 25/15 Hz
    Infrared 1      |          Y8          @ 30/15 Hz
    Infrared 1  1280x720       Y8          @ 30/15/6 Hz
    Infrared 1   848x480       Y8          @ 90/60/30/15/6 Hz
    Infrared 1   848x100       Y8          @ 300/100 Hz
    Infrared 1   640x480       Y8          @ 90/60/30/15/6 Hz
    Infrared 1   640x400       Y16         @ 25/15 Hz
    Infrared 1   640x360       Y8          @ 90/60/30/15/6 Hz
    Infrared 1   480x270       Y8          @ 90/60/30/15/6 Hz
    Infrared 1   424x240       Y8          @ 90/60/30/15/6 Hz
    Infrared 2  1280x800       Y16         @ 25/15 Hz
    Infrared 2      |          Y8          @ 30/15 Hz
    Infrared 2  1280x720       Y8          @ 30/15/6 Hz
    Infrared 2   848x480       Y8          @ 90/60/30/15/6 Hz
    Infrared 2   848x100       Y8          @ 300/100 Hz
    Infrared 2   640x480       Y8          @ 90/60/30/15/6 Hz
    Infrared 2   640x400       Y16         @ 25/15 Hz
    Infrared 2   640x360       Y8          @ 90/60/30/15/6 Hz
    Infrared 2   480x270       Y8          @ 90/60/30/15/6 Hz
    Infrared 2   424x240       Y8          @ 90/60/30/15/6 Hz
    Depth       1280x720       Z16         @ 30/15/6 Hz
    Depth        848x480       Z16         @ 90/60/30/15/6 Hz
    Depth        848x100       Z16         @ 300/100 Hz
    Depth        640x480       Z16         @ 90/60/30/15/6 Hz
    Depth        640x360       Z16         @ 90/60/30/15/6 Hz
    Depth        480x270       Z16         @ 90/60/30/15/6 Hz
    Depth        424x240       Z16         @ 90/60/30/15/6 Hz
    Depth        256x144       Z16         @ 300/90 Hz

Stream Profiles supported by RGB Camera
 Supported modes:
    STREAM      RESOLUTION     FORMAT      FPS
    Color       1920x1080      RGB8        @ 30/15/6 Hz
    Color           |          RAW16       @ 30 Hz
    Color           |          Y8          @ 30/15/6 Hz
    Color           |          BGRA8       @ 30/15/6 Hz
    Color           |          RGBA8       @ 30/15/6 Hz
    Color           |          BGR8        @ 30/15/6 Hz
    Color           |          YUYV        @ 30/15/6 Hz
    Color       1280x720       RGB8        @ 30/15/6 Hz
    Color           |          Y8          @ 30/15/6 Hz
    Color           |          BGRA8       @ 30/15/6 Hz
    Color           |          RGBA8       @ 30/15/6 Hz
    Color           |          BGR8        @ 30/15/6 Hz
    Color           |          YUYV        @ 30/15/6 Hz
    Color        960x540       RGB8        @ 60/30/15/6 Hz
    Color           |          Y8          @ 60/30/15/6 Hz
    Color           |          BGRA8       @ 60/30/15/6 Hz
[... 20 lines cut ...]
    Color           |          YUYV        @ 60/30/15/6 Hz
    Color        424x240       RGB8        @ 60/30/15/6 Hz
    Color           |          Y8          @ 60/30/15/6 Hz
    Color           |          BGRA8       @ 60/30/15/6 Hz
    Color           |          RGBA8       @ 60/30/15/6 Hz
    Color           |          BGR8        @ 60/30/15/6 Hz
    Color           |          YUYV        @ 60/30/15/6 Hz
    Color        320x240       RGB8        @ 60/30/6 Hz
    Color           |          Y8          @ 60/30/6 Hz
    Color           |          BGRA8       @ 60/30/6 Hz
    Color           |          RGBA8       @ 60/30/6 Hz
    Color           |          BGR8        @ 60/30/6 Hz
    Color           |          YUYV        @ 60/30/6 Hz
    Color        320x180       RGB8        @ 60/30/6 Hz
    Color           |          Y8          @ 60/30/6 Hz
    Color           |          BGRA8       @ 60/30/6 Hz
    Color           |          RGBA8       @ 60/30/6 Hz
    Color           |          BGR8        @ 60/30/6 Hz
    Color           |          YUYV        @ 60/30/6 Hz

Stream Profiles supported by Motion Module
 Supported modes:
    STREAM      FORMAT         FPS
    Accel       MOTION_XYZ32F  @ 200/100 Hz
    Gyro        MOTION_XYZ32F  @ 400/200 Hz

MartyG-RealSense, 2024-05-11:
I note that you are using librealsense 2.55.1.  The 2.54.2 version of librealsense should be used with wrapper 4.54.1.  There is not a ROS wrapper for 2.55.1 at the time of writing this.

The **control_transfer returned** warning indicates that there may be a communication problem between the camera and computer, likely related to the USB system (the port or the cable).

Sometimes streams are able to be published successfully in RViz even if these warnings are generating in the log continuously, so please check RViz to see if the streams are displayed.

MingyuPan2, 2024-05-16:
Hi @MartyG-RealSense , thank you for the information! Yes, i can see the streams in RViz such as pointcloud, depth, rgb, infra&2. And the terminal still sometimes output errors such as:

[WARN] [1715886051.829293254] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1]  16/05 15:00:08,336 WARNING [281472317186432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11.

--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Interestingly, whenever I try (rs-enumerate-devices), it always gives the following output: 

 16/05 15:05:43,107 ERROR [281473402728576] (handle-libusb.h:125) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 16/05 15:05:43,107 ERROR [281473446666304] (uvc-sensor.cpp:412) acquire_power failed: failed to set power state
 16/05 15:05:43,107 ERROR [281473446666304] (rs.cpp:237) [rs2_create_device( info_list:0xaaaac1441d30, index:0 ) UNKNOWN] failed to set power state
 16/05 15:05:43,107 ERROR [281473446666304] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
Could not create device - failed to set power state . Check SDK logs for details
No device detected. Is it plugged in?

-----------------

I'm still in the process of testing the camera's functionalities, so are the errors something i should fix immediately or can i ignore them as long as RViz's output is normal? Thank you!

MartyG-RealSense, 2024-05-16:
As long as RViz's output is normal then it is usually safe to ignore log warnings.

MartyG-RealSense, 2025-01-06:
Case closed due to no further comments received.

closed 2025-01-06 by MartyG-RealSense

aws-ahmad, 2025-05-14:
Hello,
No RealSense Device Detected on Orange Pi 5 Plus (Kernel 5.10.160) - Help Needed
I am currently trying to set up my Intel RealSense D455 camera on an Orange Pi 5 Plus running Ubuntu 22.04 with kernel version 5.10.160-rockchip. However, I am encountering an issue where the system does not detect the RealSense camera.