realsenseai/librealsense #12907: Examples/Save-to-disk: cannot find -ltclap: No such file or directory

BOOSTMKII, 2024-05-07:
/usr/bin/ld: cannot find -ltclap: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rs-save-to-disk.dir/build.make:97: rs-save-to-disk] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/rs-save-to-disk.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

It seems missing lib but don't know how to fix it.

MartyG-RealSense, 2024-05-08:
Hi @BOOSTMKII  Have you copied the code of the C++ save-to-disk project into your own project, please?  If you have then using the code of the RealSense SDK examples in this way can cause the new project to have missing linkages.  This can be resolved by adding additional details to the CMakeLists.txt file. 

A starter project at the link below demonstrates this principle.  It uses the same C++ script code as the **rs-hello-realsense** SDK example program but has further lines in its CMakeLists.txt file that the CMakeLists.txt file of the official version of rs-hello-realsense example does not have.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

https://github.com/zivsha/librealsense/blob/getting_started_example/examples/getting-started/CMakeLists.txt

BOOSTMKII, 2024-05-09:
> Hi @BOOSTMKII Have you copied the code of the C++ save-to-disk project into your own project, please? If you have then using the code of the RealSense SDK examples in this way can cause the new project to have missing linkages. This can be resolved by adding additional details to the CMakeLists.txt file.
> 
> A starter project at the link below demonstrates this principle. It uses the same C++ script code as the **rs-hello-realsense** SDK example program but has further lines in its CMakeLists.txt file that the CMakeLists.txt file of the official version of the rs-hello-realsense example does not have.
> 
> https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started
> 
> https://github.com/zivsha/librealsense/blob/getting_started_example/examples/getting-started/CMakeLists.txt

Hi Marty. I have successfully implemented save-to-disk in getting_started_example, thank you!

MartyG-RealSense, 2024-05-09:
You are very welcome.  I'm pleased to hear that you achieved a successful outcome.  Thanks very much for the update!

closed 2024-05-11 by BOOSTMKII