realsenseai/librealsense #12910: Plz help.I cant use Intel RealSense Viewer / Depth Quality Tool

1193361084, 2024-05-08:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_create_context_ex(api_version:25501, json_settings:{):
    failed to convert special folder: errno=42

```

MartyG-RealSense, 2024-05-08:
Hello, a problem with the 'special folder' might occur if your Windows PC does not have a **Documents** folder in your user files directory.  If for example your Windows user name was 'Fred' then your Documents folder might be at the following location:

**C: > Users / Fred / My Documents**

If there is not a My Documents folder in the C: > Users directory then try creating a new folder called My Documents in the Users directory.

Please see https://github.com/IntelRealSense/librealsense/issues/5707#issuecomment-602511768 and the comments beneath it for more information about this problem.

MartyG-RealSense, 2024-05-28:
Case closed due to no further comments received.

closed 2024-05-28 by MartyG-RealSense

Dm1stry, 2024-08-06:
I have folder named My Documents, but it doesn't works for me neither. Previous version (2.54) worked just fine, but when i installed new version i keep getting error like op. 

[image]

Windows 10

Maybe problem tha is my username in Cyrillic, but then is must not work in 2.54 too

MartyG-RealSense, 2024-08-06:
Hi @Dm1stry  Advice about the **errno=42** error can be found at https://github.com/IntelRealSense/librealsense/issues/13037

Dm1stry, 2024-08-06:
> Hi @Dm1stry  Advice about the **errno=42** error can be found at https://github.com/IntelRealSense/librealsense/issues/13037

Thank you!

MartyG-RealSense, 2024-12-04:
My Intel RealSense colleagues have provided some further advice about the errno=42 error.

Given that the problem seems to be related to certain charcters being used in a Windows user account name, I asked whether creating a secondary Windows user account with a name that has no special characters might help.  This was their response. 

****************

You can try to run it as Administrator instead of switching the account.  On Windows 11, it also has a feature to run the app as a different user.  That option can be enabled with the instructions at the link below:
 
https://winaero.com/how-to-run-apps-as-different-user-in-windows-11/

AviaAv, 2024-12-29:
Hey @Dm1stry @1193361084 

We've recently fixed an issue about the "errno=42" error, which would happen if the user's name on that machine contained Unicode characters. Could you please check the development branch to see if this fix resolves the issue?

The PR with the fix is: #13597 

The development branch can be found here: https://github.com/IntelRealSense/librealsense/tree/development