realsenseai/librealsense #12912: pyrealsense via Debian packages on Jetson Nano

jackson-smith-a, 2024-05-09:
I’ve been attempting to get pyrealsense working on a Jetson Nano running Ubuntu 18. I’ve attempted to just use `pip` but it seems binaries aren’t available for it. I’ve attempted to compile from source, but my version of JetPack is unsupported (https://github.com/IntelRealSense/librealsense/issues/12131) and I’m already at the max version allowed for my Nano. I was able to successfully install the precompiled version via Debian packages following the guide [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) but this doesn’t seem to provide the Python wrapper. I can successfully compile C++ programs and get camera feed over `realsense-viewer`, but I can’t `import pyrealsense2` on Python 2 or Python 3. Is there a step I'm missing?

MartyG-RealSense, 2024-05-09:
Hi @jackson-smith-a  Which JetPack are you using, please?  JetPack 6.0 is supported in librealsense 2.55.1 onwards.

The JetPack version does not matter though if you compile from source on Jetson with the CMake flag **-DFORCE_RSUSB_BACKEND=ON** included in the CMake build instruction.

Use of this flag is demonstrated in a Nano-compatible guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 for compiling librealsense and the pyrealsense2 wrapper at the same time so that you do not need to build the wrapper separatey afterwards.

jackson-smith-a, 2024-05-10:
I successfully compiled following the guide at #6964 but still can't import it in python. I've followed steps 7 and 8 and tried `import pyrealsense2.pyrealsense2 as rs` with no success. Trying it within wrapper/python instead fails while trying to `import .pyrealsense2`.

Notably, while there are some files named librealsense2.so and similarly in /usr/local/lib, /usr/local/lib/python3.6 seems to be empty.

MartyG-RealSense, 2024-05-10:
Since librealsense 2.54.1 Python 3.6 has no longer been supported by the SDK, with 3.7 being the minimum supported version.

jackson-smith-a, 2024-05-11:
How do I get it to compile with a different version of python? I changed the CMake command to `-DPYTHON_EXECUTABLE=/usr/bin/python3.8` but /usr/local/lib/python3.8 is still empty, so I'm assuming I need to adjust something else as well.

MartyG-RealSense, 2024-05-11:
If you are building the Python wrapper from source code then I would expect the build's compiled files to be in the librealsense source code folder at **librealsense > build > wrappers >python**

hs3296, 2024-05-11:
I too was trying pyrealsense2 with python3.6 on Jetson Tx2. After reading this article, I installed python3.8. I'm trying to download nvidia pytorch and torchvision, but jetson tx2 says jetpack4.6 is the latest. So, can’t I install pytorch and torchvision on python3.8? Also, is this the same for pyrealsense2?

MartyG-RealSense, 2024-05-11:
Hi @hs3296  The librealsense SDK is compatible with JetPack 4.6 when using Ubuntu 18.04.  The pyrealsense2 wrapper is compatible with Python 3.8.

The Nvidia documentation link below indicates that Pytorch can be used with JetPack 4.6.1.

https://docs.nvidia.com/deeplearning/frameworks/install-pytorch-jetson-platform/index.html

Whilst the link below indicates that Pytorch version 1.11 should be used with JetPack 4.6.1.

https://docs.nvidia.com/deeplearning/frameworks/install-pytorch-jetson-platform-release-notes/pytorch-jetson-rel.html#pytorch-jetson-rel__section_tzk_3gv_52b

hs3296, 2024-05-11:
Thank you! I have jetpack 4.6.4, is this only available for 4.6.1?
<python3 -m pip install --upgrade pip; python3 -m pip install numpy=='1.26.1' python3 -m pip install --no-cache https://developer.download.nvidia.com/compute/redist/jp/v464/pytorch/1.11.0a0+bfe5ad28 > I entered it like this.

As a result, I received the reply <torch-1.11.0-cp38-cp38-linux_aarch64.whl is not a supported wheel on this platform>.

MartyG-RealSense, 2024-05-11:
4.6.1 is the only 4.6 version supported by the RealSense SDK's kernel patch script **patch-realsense-ubuntu-L4T.sh**.  However, you can bypass this and use a non-supported JetPack version if you build the SDK from source code with CMake with the build flag **-DFORCE_RSUSB_BACKEND=ON** included in the CMake build instruction and do not apply a kernel patch script.

In regard to Pytorch's specified requirement for JetPack 4.6.1, it appears from the link below that the error message might be caused by the Python version, as it states that Pytorch only supports a maximum of Python 3.6 when using JetPack 4.  Python 3.8 support was introduced in the Pytorch wheel files for JetPack 5.

https://forums.developer.nvidia.com/t/pytorch-for-jetson/72048

jackson-smith-a, 2024-05-11:
After running `make` and `sudo make install` inside of build/wrappers/python, I got some interesting messages: `Up-to-date: /usr/local/OFF/pybackend... Up-to-date: /usr/local/OFF/pyrealsense2...`

After adding `export PYTHONPATH=$PYTHONPATH:/usr/local/OFF` to my .bashrc and sourcing it, it works.

closed 2024-05-11 by jackson-smith-a

MartyG-RealSense, 2024-05-11:
It's great to hear that you achieved a solution, @jackson-smith-a - thanks very much for the update!