realsenseai/librealsense #12917: keep popping 16:15:26 [Info] /home/greg988/Downloads/realsense/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video2

b07902036, 2024-05-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | intelrealsense l515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   ubuntu | 
| Kernel Version (Linux Only)    |  22.04                                        | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | realsense-viewer| 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
when I am trying using realsense viewer to preview my camera's video, it always got message on realsense viewer message box like the topic.
I have try redownload or using other camera, but it didn't fix.
### message in dmesg -wT

[五  5月 10 16:06:02 2024] hid-sensor-hub 0003:8086:0B64.000C: No report with id 0xffffffff found
[五  5月 10 16:06:06 2024] hid-sensor-hub 0003:8086:0B64.000C: No report with id 0xffffffff found
[五  5月 10 16:06:11 2024] usb 2-2: USB disconnect, device number 9
[五  5月 10 16:06:11 2024] hid-sensor-hub 0003:8086:0B64.000C: No report with id 0xffffffff found
[五  5月 10 16:06:11 2024] hid-sensor-hub 0003:8086:0B64.000C: No report with id 0xffffffff found
[五  5月 10 16:14:21 2024] usbcore: deregistering interface driver uvcvideo
[五  5月 10 16:14:21 2024] mc: Linux media interface: v0.10
[五  5月 10 16:14:21 2024] videodev: Linux video capture interface: v2.00
[五  5月 10 16:14:21 2024] usbcore: registered new interface driver uvcvideo
[五  5月 10 16:15:15 2024] usb 2-2: new SuperSpeed USB device number 10 using xhci_hcd
[五  5月 10 16:15:15 2024] usb 2-2: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.58
[五  5月 10 16:15:15 2024] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[五  5月 10 16:15:15 2024] usb 2-2: Product: Intel(R) RealSense(TM) 515
[五  5月 10 16:15:15 2024] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Camera
[五  5月 10 16:15:15 2024] usb 2-2: SerialNumber: 00000000F0245900
[五  5月 10 16:15:15 2024] usb 2-2: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[五  5月 10 16:15:15 2024] usb 2-2: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[五  5月 10 16:15:16 2024] hid-sensor-hub 0003:8086:0B64.000D: No report with id 0xffffffff found
[五  5月 10 16:15:16 2024] hid-sensor-hub 0003:8086:0B64.000D: No report with id 0xffffffff found
[五  5月 10 16:15:20 2024] hid-sensor-hub 0003:8086:0B64.000D: No report with id 0xffffffff found
[五  5月 10 16:15:30 2024] usb 2-2: USB disconnect, device number 10
[五  5月 10 16:15:31 2024] hid-sensor-hub 0003:8086:0B64.000D: No report with id 0xffffffff found

rs-enumerate-devices :
No device detected. Is it plugged in?

MartyG-RealSense, 2024-05-10:
Hi @b07902036  Which version of the RealSense SDK / RealSense Viewer are you using, please?  Support for the L515 camera model was removed in the current latest version 2.55.1, so if you have 2.55.1 installed then you will need to change it to the previous version 2.54.2.

If you are already using 2.54.2 or older, does it make a difference if you unplug the micro-sized end of the USB cable from the base of the camera, reverse the connector's orientation and plug it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).

b07902036, 2024-05-12:
oh I may install the newest version, as tell 2.55.1, I will try to install the 2.54.2 version and keep working. Thanks.

b07902036, 2024-05-12:
I'm sorry, but It seems that when I reinstall the librealsense in version 2.54.2, the realsense viewer is still in 2.55.1 version. how can I install the correct version of it?
greg988@greg988-System-Product-Name:~/Downloads$ sudo apt install librealsense2=2.54.2-0~realsense.10773 librealsense2-dev=2.54.2-0~realsense.10773
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  librealsense2-udev-rules
The following NEW packages will be installed:
  librealsense2 librealsense2-dev librealsense2-udev-rules
0 upgraded, 3 newly installed, 0 to remove and 13 not upgraded.
Need to get 4,217 kB/4,229 kB of archives.
After this operation, 14.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://librealsense.intel.com/Debian/apt-repo jammy/main amd64 librealsense2 amd64 2.54.2-0~realsense.10773 [2,844 kB]
Get:2 https://librealsense.intel.com/Debian/apt-repo jammy/main amd64 librealsense2-dev amd64 2.54.2-0~realsense.10773 [1,373 kB]
Fetched 4,217 kB in 5s (768 kB/s)            
Selecting previously unselected package librealsense2:amd64.
(Reading database ... 249945 files and directories currently installed.)
Preparing to unpack .../librealsense2_2.54.2-0~realsense.10773_amd64.deb ...
Unpacking librealsense2:amd64 (2.54.2-0~realsense.10773) ...
Selecting previously unselected package librealsense2-dev:amd64.
Preparing to unpack .../librealsense2-dev_2.54.2-0~realsense.10773_amd64.deb ...
Unpacking librealsense2-dev:amd64 (2.54.2-0~realsense.10773) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.55.1-0~realsense.12474_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.55.1-0~realsense.12474) ...
Setting up librealsense2:amd64 (2.54.2-0~realsense.10773) ...
Setting up librealsense2-udev-rules:amd64 (2.55.1-0~realsense.12474) ...

MartyG-RealSense, 2024-05-12:
Did you build the ROS wrapper again after installing 2.54.2 please?  The wrapper has to be built each time that the librealsense version is changed.

b07902036, 2024-05-14:
I'm sorry, I don't get it on build the ROS wrapper. I didn't do that before, can you give me more information about how or why build the ROS wrapper? Thanks.

MartyG-RealSense, 2024-05-14:
My apologies, I mis-read your log.  You do not need to install the ROS wrapper.  Your problem is actually that you installed 2.54.2 after earlier having installed 2.55.1 but the Viewer is still the 2.55.1 version.  You need 2.54.2 for your L515 to work.  Thanks very much for your patience!

When installing on a PC from packages, the two packages that need to be installed are librealsense2-dkms and librealsense2-utils.  The librealsense2-dev package is optional.  So to install a specific version of librealsense you would find the package names for that version with the **apt-cache showpkg librealsense2** command.  It looks as though you have likely used showpkg.

In your installation instruction for 2.54.2 it looks as though you have not installed the `librealsense2-utils` package for 2.54.2.  The RealSense Viewer is installed by the utils package.

b07902036, 2024-05-14:
ok, I have install the 2.54.2 version, do I need to do something special that my realsense-viewer are still 2.55.1? Do I need to do  anything to uninstall the 2.55.1 version of the realsense-viewer?

MartyG-RealSense, 2024-05-14:
Can you find realsense-viewer in the **usr/local/bin** folder of Ubuntu?  If you can, are you able to delete it and then run the 2.54.2 version of the **librealsense2-utils** package to install realsense-viewer back as the 2.54.2 version?

b07902036, 2024-05-15:
sorry about that, but I can't reach it after deleted the realsense-viewer in the usr/local/bin folder
Did I do anything wrong?
> greg988@greg988-System-Product-Name:\~$ sudo apt-get purge librealsense2-utils=2.54.2-0\~realsense.10773
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  librealsense2-utils*
0 upgraded, 0 newly installed, 1 to remove and 12 not upgraded.
After this operation, 23.2 MB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 356357 files and directories currently installed.)
Removing librealsense2-utils:amd64 (2.54.2-0\~realsense.10773) ...
greg988@greg988-System-Product-Name:\~$ sudo apt install librealsense2-utils=2.54.2-0\~realsense.10773
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  librealsense2-utils
0 upgraded, 1 newly installed, 0 to remove and 12 not upgraded.
Need to get 9,053 kB of archives.
After this operation, 23.2 MB of additional disk space will be used.
Get:1 https://librealsense.intel.com/Debian/apt-repo jammy/main amd64 librealsense2-utils amd64 2.54.2-0\~realsense.10773 [9,053 kB]
Fetched 9,053 kB in 4s (2,462 kB/s)              
Selecting previously unselected package librealsense2-utils:amd64.
(Reading database ... 356320 files and directories currently installed.)
Preparing to unpack .../librealsense2-utils_2.54.2-0\~realsense.10773_amd64.deb .
..
Unpacking librealsense2-utils:amd64 (2.54.2-0\~realsense.10773) ...
Setting up librealsense2-utils:amd64 (2.54.2-0\~realsense.10773) ...
greg988@greg988-System-Product-Name:\~$ realsense-viewer
bash: /usr/local/bin/realsense-viewer: No such file or directory

MartyG-RealSense, 2024-05-15:
I do not think that you have done anything wrong.

Can the L515's video be shown if you launch **rs-capture** instead of realsense-viewer?

b07902036, 2024-05-17:
unfortunately, no.

`greg988@greg988-System-Product-Name:/~$ rs-capture
 17/05 14:24:28,379 ERROR [125023893341504] (rs.cpp:237) [rs2_pipeline_start( pipe:0x5c6114d249a0 ) UNKNOWN] No device connected
 17/05 14:24:28,379 ERROR [125023893341504] (rs.cpp:237) [rs2_delete_pipeline_profile( profile:nullptr ) UNKNOWN] null pointer passed for argument "profile"
RealSense error calling rs2_pipeline_start(pipe:0x5c6114d249a0):
    No device connected`

MartyG-RealSense, 2024-05-17:
What information do you get about the camera if you use the command **lsusb** in the Ubuntu terminal?

The lsusb command does not require the librealsense SDK to be installed in order to work.  If the RealSense camera is detected then the information should contain a line like the one below.

`Bus 001 Device 008: ID 8086:0b3a Intel Corp.`

b07902036, 2024-05-20:
yes, it have detect the service

> greg988@greg988-System-Product-Name:~$ lsusb
Bus 002 Device 002: ID 8086:0b64 Intel Corp. Intel(R) RealSense(TM) 515
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0b05:19af ASUSTek Computer, Inc. AURA LED Controller
Bus 001 Device 004: ID 0461:4de2 Primax Electronics, Ltd USB Optical Mouse
Bus 001 Device 002: ID 0461:4e6e Primax Electronics, Ltd Wired USB Keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

MartyG-RealSense, 2024-05-20:
Did you perform the reveral of the L515's micro-sized end of the USB cable that was suggested above in https://github.com/IntelRealSense/librealsense/issues/12917#issuecomment-2104420340 please?

b07902036, 2024-05-20:
yes, I have also tried another l515 i have, and all get the same result.

MartyG-RealSense, 2024-05-20:
Is there any difference if you launch the 2.54.2 SDK's example programs in sudo admin permissions mode?  

```
sudo realsense-viewer
sudo rs-capture
```

b07902036, 2024-05-21:
oh, I can get the version 2.54.2 of realsense-viewer by sudo, but not in re-capture. rs-capture still can't act properly.

MartyG-RealSense, 2024-05-21:
If realsense-viewer works okay when run with sudo power then you should not need to use rs-capture.

b07902036, 2024-05-22:
But when I run `modinfo uvcvideo | grep "version:"`, it doesn't have realsense string. Is it a normal condition?

MartyG-RealSense, 2024-05-22:
If you are using a recent kernel version such as 6.5 then there have been reports of the **realsense** string not being displayed but the camera is still able to function correctly.

b07902036, 2024-05-28:
Sorry, but I have another question about my code will pop the error about ne device connected, can it be fixed?

```
import pyrealsense2 as rs
import numpy as np
import cv2
pipe = rs.pipeline()
cfg  = rs.config()
cfg.enable_stream(rs.stream.color, 640,480, rs.format.bgr8, 30)
cfg.enable_stream(rs.stream.depth, 640,480, rs.format.z16, 30)
pipe.start(cfg)
while True:
    frame = pipe.wait_for_frames()
    depth_frame = frame.get_depth_frame()
    color_frame = frame.get_color_frame()
    depth_image = np.asanyarray(depth_frame.get_data())
    color_image = np.asanyarray(color_frame.get_data())
    depth_cm = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha = 0.5), cv2.COLORMAP_JET)

    gray_image = cv2.cvtColor(color_image, cv2.COLOR_BGR2GRAY)

    cv2.imshow('rgb', color_image)
    cv2.imshow('depth', depth_cm)

    if cv2.waitKey(1) == ord('q'):
        break 
pipe.stop()
```

> Traceback (most recent call last):
  File "/home/greg988/Desktop/test.py", line 11, in <module>
    pipe.start(cfg)
RuntimeError: No device connected

MartyG-RealSense, 2024-05-28:
Is the camera stll undetected if you try `import pyrealsense2.pyrealsense2 as rs`

b07902036, 2024-05-28:
unfortunately, no. Still the same result as above.

MartyG-RealSense, 2024-05-28:
Did you run your script with sudo admin power using **sudo python3 test.py**

b07902036, 2024-05-29:
it said that 
`ModuleNotFoundError: No module named 'pyrealsense2'`
but I'm sure that I have installed pyrealsense2?

MartyG-RealSense, 2024-05-29:
Please try `import pyrealsense2.pyrealsense2 as rs` again and running the script in sudo mode with `sudo python3 test.py` to see whether this resolve the **ModuleNotFound** error now that the camera can be detected.

b07902036, 2024-05-30:
[image]
same as above

> greg988@greg988-System-Product-Name:~/Desktop$ sudo python3 test.py
> Traceback (most recent call last):
>   File "/home/greg988/Desktop/test.py", line 1, in <module>
>     import pyrealsense2.pyrealsense2 as rs
> ModuleNotFoundError: No module named 'pyrealsense2'

MartyG-RealSense, 2024-05-30:
Which method did you use to install the pyrealsense2 wrapper, please?

b07902036, 2024-05-31:
pip install pyrealsense2

> greg988@greg988-System-Product-Name:~/Desktop$ pip install pyrealsense2
> Defaulting to user installation because normal site-packages is not writeable
> Requirement already satisfied: pyrealsense2 in /home/greg988/.local/lib/python3.10/site-packages (2.55.1.6486)

MartyG-RealSense, 2024-05-31:
On the **Requirement already satisfied** line in the comment above, it says that the pyrealsense2 version installed with pip is 2.55.1.6486.  But you had changed your librealsense version from 2.55.1 to 2.54.1 so that it would work with your L515 camera.  The version of the pip package that should be installed should be the one for the 2.54 librealsense version that you currently have installed.

What happens if you try installing the **2.54.2.5684** version of the pip package with the command below?

`pip install pyrealsense2==2.54.2.5684`

b07902036, 2024-06-04:
ok, fix it. Thank you.

closed 2024-06-18 by MartyG-RealSense