realsenseai/librealsense #12920: D455F and Depth Module D450 + Vision Processor D4 Board V3 Bundle (8 Pack)

lesreaper, 2024-05-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel® RealSense™ Depth Module D450 + Vision Processor D4 Board V3 Bundle (8 Pack) | 
| Operating System & Version |   Jetson Jetpack  | 
| Kernel Version (Linux Only)    |  5.1.3                                       | 
| Platform                                 | nVidia Jetson  |
| SDK Version                            |  {2.x }                          | 
| Language                            |  python                          | 
| Segment			|  AI Platfom                         | 

### Issue Description
We think we might have messed up on our order. We used the [455F](https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d455f.html) for our prototype.

We want to get a bunch of these [Modules and Vision Processor Boards](https://store.intelrealsense.com/buy-intel-realsense-depth-module-d450-and-d4-board-bundle-d.html), because we're not going to building our own boards yet, but don't want to pay $450+ per camera. We purchased the 8-Pack bundle, along with some interposers, still yet to deliver. We also are using the USB-C external power splitter in our prototype. 

It seems this 8-pack is not the 455F however, which comes with an IR filter and longer range. 

1. Is there a way to purchase the 455F in bulk properly?
2. How do we handle the extra splitter? Does Intel have something specific for that?
3. How do we go about changing an order if need be?

lesreaper, 2024-05-11:
As far as the USB power splitter, no, I'm referring to a different piece, but I got the answer on that in another place. 

We're kind of in a hurry to buy, and it looks like we're going to have to wait and pray that someone sees our message, then set up an appointment, etc. 

Is there a faster way to get our pre-sales questions answered and talk to someone?

MartyG-RealSense, 2024-05-11:
If your questions are technical rather than about changing an order that has been placed then I will be happy to provide assistance if possible, as I am an Intel RealSense Support Engineer.

The D450 board in the multipack that does not have the IR filters is long-range too (up to 20 m depth sensing) so the D455f would not provide an additional range advantage over the D450 + D4 pack that you have ordered.  The advantage that the filter provides is in dealing with light reflections (such as when directly facing sunlight) and when processing scenes that have a row of similar looking objects such as a row of fence posts or floor / ceiling tiles.

lesreaper, 2024-05-13:
Thank you!! That's the primary answer I needed.

It's the filter I was wondering about. We're installing in buildings that may have monochromatic walls in the backgrounds at various depths, and we could use that for sure. Is the CLAREX NIR-75N filter available pre-cut somewhere from the RealSense team, or so I have to buy it somewhere else and cut it myself?

MartyG-RealSense, 2024-05-13:
You are very welcome!

The filter is not available for purchase as a ready-made RealSense product.  A senior RealSense team member once recommended a company called Astra Products as a reliable purchasing source for optical filters for use with RealSense cameras though (other suppliers are of course available).  Astra Products' page about the CLAREX filter is below.

https://astraproducts.com/info-acrylic-nir-filters.asp

MartyG-RealSense, 2024-05-20:
Hi @lesreaper  Do you require further assistance with this case, please?  Thanks!

lesreaper, 2024-05-20:
I don't think so at this point, thank you @MartyG-RealSense !

I'm speaking with Astra now, and will test out the cameras. Appreciate the pre-sale support!

MartyG-RealSense, 2024-05-20:
You are very welcome.  Thanks very much for the update.  As you do not require further assistance at this point, I will close this case.  Please feel free to re-open it at a future date or create a new issue if you have further questions.  Thanks again!

lesreaper, 2024-05-20:
Ah, sorry @MartyG-RealSense , looks like there is a question. Astra doesn't seem to have a record of the material. Do we know which wavelength is the right one for the new 455f?

Type No. | Color Tone | Near Infrared transmission
-- | -- | --
NIR-70N | Black | Transmit from 670nm
NIR-75N | Black | Transmit from 730nm
NIR-80N | Black | Transmit from 760nm
NIR-85N | Black | Transmit from 810nm
NIR-90N | Black | Transmit from 860nm
NIR-911N | Black | Transmit from 910nm

MartyG-RealSense, 2024-05-20:
D455F uses the NIR-75N filter.