realsenseai/librealsense #12921: External power supply for Intel Realsense d415

tjak99, 2024-05-11:
Hi,

I’ve been using a Jetson Nano with an Intel Realsense D415 camera and have encountered a problem related to power consumption. The camera’s power demands seem to push the Jetson to its limits, especially when processing data. Additionally, the Jetson does not recognize the camera on boot, requiring me to replug it. Sometimes, the Jetson even fails to boot due to the camera’s power draw. Is there a solution to power the camera from an external source while continuing to transmit data to the Jetson?

Thank you for your time.

MartyG-RealSense, 2024-05-12:
Hi @tjak99  You can attach the camera to a mains electricity powered USB hub if your project is one where the camera does not need to move around, like on a mobile robot.  Because a powered USB hub draws from the mains instead of the computer's power supply, camera connections are typically more stable than when the camera is plugged directly into the computer's USB port.

Intel also have specific guidance about Nano, strongly suggesting to enable the **barrel jack** connector if your model of NAno board has one.  Instructions for doign so can be found at the link below.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

tjak99, 2024-05-12:
Thank you for the prompt response. The problem is that the camera is mounted on a mobile robot, so I cannot connect it to mains electricity. I considered a quick solution, but it seems I may need to design an external circuit for this application. If you have any more ideas, I would be grateful.

MartyG-RealSense, 2024-05-12:
The Nano has a micro-USB port that it can be powered by.  I wonder if it would make a difference to the Nano's USB power stability if it was powered by a USB power bank.  This is like a battery pack that you can charge up on mains power and then unplug and transport it, attaching a USB device to it to power it as though it was plugged into the mains.

There is an example of a discussion about powering a Nano with a power bank at the link below.

https://www.chiefdelphi.com/t/jetson-nano-power/379667

MartyG-RealSense, 2024-05-23:
Hi @tjak99  Do you require further assistance with this case, please?  Thanks!

tjak99, 2024-05-24:
I wanted to let you know that I managed to get it working. I used an external battery with a custom-made PCB and connected the output from the PCB to the Jetson header. This setup provides up to 6 Amps. Thanks for your rapid responses, great service

MartyG-RealSense, 2024-05-24:
You are very welcome.  That's great to hear that you succeeded.  Thanks very much for the update and for sharing the details that worked for you!

MartyG-RealSense, 2024-05-30:
Hi @tjak99  After achieving a solution, do you require further assistance with this case please?  Thanks!

closed 2024-06-06 by MartyG-RealSense