realsenseai/librealsense #12922: Error to update firmware

ibarramiguel119, 2024-05-12:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|
I am working with a realsense D457 camera, it was working fine but at one point it stopped reading the port as 3.2 and changed to 2.1, a firmware update message appeared in relasense Viewer to which I accepted, now it no longer recognizes the camera tells me that I do not have a device connected. I hope for your valuable help, greetings!!

MartyG-RealSense, 2024-05-12:
Hi @ibarramiguel119  Could you first try unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end). Does  doing so make any difference?

ibarramiguel119, 2024-05-12:
No, I try to do it like that but it doesn't work. I get those errors
[image]

ibarramiguel119, 2024-05-12:
I'm switching to Windows 11 to check what happened and the drivers. and I get the following
[image]

MartyG-RealSense, 2024-05-12:
I will discuss your problem with my Intel RealSense colleagues.  May I ask which country you are located in, please?

ibarramiguel119, 2024-05-12:
Thank you very much, I appreciate it very much, I'm from Mexico.

MartyG-RealSense, 2024-05-12:
Thank you very much, and thanks for your patience.

MartyG-RealSense, 2024-05-30:
Hi @ibarramiguel119  Thanks very much again for your patience.  Please send me an email at the address below so that we can discuss the next step in restoring your D457 camera.

martyx.grover@intel.com

MartyG-RealSense, 2024-06-06:
Hi @ibarramiguel119  Please email me at the address in the comment above if you require further assistance with restoring your D457 camera.  Thanks!

MartyG-RealSense, 2024-06-14:
Hi @ibarramiguel119  Please send me an email at the address at https://github.com/IntelRealSense/librealsense/issues/12922#issuecomment-2140246238 if you would like to continue the process of having your D457 restored.

closed 2024-07-09 by MartyG-RealSense