realsenseai/librealsense #12925: Can not run libuvc_installation.sh normally

DiopWang, 2024-05-13:
Hello,When I'm running realsense-viewer,it shows like this:
[image]

`21:12:39 [Info] /tmp/sandbox/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/platform/5fb80000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/usb1/1-2/1-2:1.0/video4linux/video0
21:12:39 [Info] /tmp/sandbox/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video1 realpath=/sys/devices/platform/5fb80000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/usb1/1-2/1-2:1.0/video4linux/video1
21:12:39 [Info] /tmp/sandbox/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/platform/5fb80000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/usb1/1-2/1-2:1.0/video4linux/video2
21:12:39 [Info] /tmp/sandbox/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/platform/5fb80000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/usb1/1-2/1-2:1.0/video4linux/video3
21:12:39 [Info] /tmp/sandbox/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/platform/5fb80000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/usb1/1-2/1-2:1.3/video4linux/video4
21:12:39 [Info] /tmp/sandbox/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/platform/5fb80000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/usb1/1-2/1-2:1.3/video4linux/video5
21:12:39 [Info] /tmp/sandbox/src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)`

Then I search online for the problems,I try to run libuvc_installation.sh,but it didn't run normally

`
linaro@bm1684:/data$ sudo ./libuvc_installation.sh 
+ exec
+ exec
+ '[' 0 -ne 0 ']'
+ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04 LTS
Release:	20.04
Codename:	focal
++ uname -r
+ echo 'Kernel version 5.4.217-bm1684-g70a3e20060f6-dirty'
Kernel version 5.4.217-bm1684-g70a3e20060f6-dirty
+ sudo apt-get update
Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease                    
Hit:2 https://librealsense.intel.com/Debian/apt-repo bionic InRelease         
Hit:3 https://librealsense.intel.com/Debian/apt-repo focal InRelease
Reading package lists... Done
+ cd /root/
+ sudo rm -rf ./librealsense_build
+ mkdir librealsense_build
+ cd librealsense_build
++ sudo swapon --show
++ wc -l
+ '[' 0 -eq 0 ']'
+ echo 'No swapon - setting up 1Gb swap file'
No swapon - setting up 1Gb swap file
+ sudo fallocate -l 2G /swapfile
+ sudo chmod 600 /swapfile
+ sudo mkswap /swapfile
mkswap: /swapfile: warning: wiping old swap signature.
Setting up swapspace version 1, size = 2 GiB (2147479552 bytes)
no label, UUID=8ea18b91-7bd9-4148-9e59-7aa2566480ba
+ sudo swapon /swapfile
swapon: /swapfile: swapon failed: Invalid argument
`

for more information

`
linaro@bm1684:/data$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04 LTS
Release:	20.04
Codename:	focal
linaro@bm1684:/data$ uname -r
5.4.217-bm1684-g70a3e20060f6-dirty
`

Can you give me some advice ,please

DiopWang, 2024-05-13:
I am using D435i

MartyG-RealSense, 2024-05-13:
Hi @DiopWang  This error does not usually occur when using the **libuvc_installation.sh** script.  My research of non-RealSense Linux cases indicates that when it does occur, it is caused by the instruction `sudo fallocate -l 2G /swapfile`.  In the libuvc script it can be found on line 26. 

Replacing the sudo fallocate command with the one below can resolve the error.

`sudo dd if=/dev/zero of=/swapfile bs=1024 count=1048576`

DiopWang, 2024-05-13:
> Hi @DiopWang This error does not usually occur when using the **libuvc_installation.sh** script. My research of non-RealSense Linux cases indicates that when this error does occur, it is caused by the instruction `sudo fallocate -l 2G /swapfile`. In the libuvc script it can be found on line 26.您好，使用 libuvc_installation.sh 脚本时通常不会发生此错误。我对非 RealSense Linux 案例的研究表明，当确实发生此错误时，它是由指令 `sudo fallocate -l 2G /swapfile` 引起的。在 libuvc 脚本中，可以在第 26 行找到它。
> 
> Replacing the sudo fallocate command with the one below can resolve the error.将 sudo Fallocate 命令替换为以下命令可以解决该错误。
> 
> `sudo dd if=/dev/zero of=/swapfile bs=1024 count=1048576`

I really thank you for your response,however,I follow your advice to replace `sudo fallocate -l 2G /swapfile` as `sudo dd if=/dev/zero of=/swapfile bs=1024 count=1048576`,but it did not work.I have met this problem before also
like this,it shows `swapon: /swapfile: swapon failed :Invalid argument`,Can you tell me what can I do next.Thank you so much
[image]

DiopWang, 2024-05-13:
> Hi @DiopWang This error does not usually occur when using the **libuvc_installation.sh** script. My research of non-RealSense Linux cases indicates that when this error does occur, it is caused by the instruction `sudo fallocate -l 2G /swapfile`. In the libuvc script it can be found on line 26.
> 
> Replacing the sudo fallocate command with the one below can resolve the error.
> 
> `sudo dd if=/dev/zero of=/swapfile bs=1024 count=1048576`

@MartyG-RealSense Hi,I am really thank you for your response.However,I did what you told me,it still show `swapon: /swapfile: swapon failed: Invalid argument`. Can you tell me what can I do in other way?Thank you so much
[image]

MartyG-RealSense, 2024-05-13:
If you comment out the entire swapfile mechanism in lines 26 to 30 then is the libuvc script able to complete sucessfully?

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L26-L30

DiopWang, 2024-05-13:
> If you comment out the entire swapfile mechanism in lines 26 to 30 then is the libuvc script able to complete sucessfully?
> 
> https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L26-L30

@MartyG-RealSense Yes,I comment out it,and it can work,and `Librealsense script completed`
But when I am running rs-enumerate-devices,it still did not work.
To be honest,I have tried almost every methods in last 4 days

DiopWang, 2024-05-13:
> > If you comment out the entire swapfile mechanism in lines 26 to 30 then is the libuvc script able to complete sucessfully?
> > https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L26-L30
> 
> @MartyG-RealSense Yes,I comment out it,and it can work,and `Librealsense script completed` But when I am running rs-enumerate-devices,it still did not work. To be honest,I have tried almost every methods in last 4 days ![cfc29177543431522a7896173ccebb4](https://private-user-images.githubusercontent.com/132583657/330003372-ee87c872-345f-4f6e-9e4c-913b25c3d5aa.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU1OTM5ODQsIm5iZiI6MTcxNTU5MzY4NCwicGF0aCI6Ii8xMzI1ODM2NTcvMzMwMDAzMzcyLWVlODdjODcyLTM0NWYtNGY2ZS05ZTRjLTkxM2IyNWMzZDVhYS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxM1QwOTQ4MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMjFhOTVmMGMzNGI2MmI4NGRhYjc5OTJiNTZkNDQ3MDEzNjZkMmZkNDczZWJlMzBiMmVkMzFmZWYwMGIxMTI2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UMdfdPSUsJzS4sfRYIOqW9k4DBCFhTxW3okPX08VVME)
> 
> ![761878c5a24e997e3215582b7d78d61](https://private-user-images.githubusercontent.com/132583657/330004011-0d6e150b-7d43-49ec-b3f6-fbcfba857b2b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU1OTM5ODQsIm5iZiI6MTcxNTU5MzY4NCwicGF0aCI6Ii8xMzI1ODM2NTcvMzMwMDA0MDExLTBkNmUxNTBiLTdkNDMtNDllYy1iM2Y2LWZiY2ZiYTg1N2IyYi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxM1QwOTQ4MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNGM3NDg3NjhkYzlmYTdhNjUwYTIxNjgyNDA3MWE3YmNhZjc3ZmJjMTNjZjg4NjgyZDE2ZDkyMGUyY2IzMzNiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.VGxV9H_XrYLV0P_d0KOXFgeXaivtdU0cLO348M8tsMU)

术语库
Could this be the cause of the system kernel, such as no corresponding drivers

MartyG-RealSense, 2024-05-13:
When using a libuvc installation of librealsense the kernel should not matter as librealsense bypasses the kernel.

Are you using the official 1 meter USB cable supplied with the camera or a longer USB cable of your own choice?

What is the computer that you are using your D435i on, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc)

Is the cable plugged into a built-in USB port on the computer or a USB interface such as a hub or PCIe?

DiopWang, 2024-05-13:
> When using a libuvc installation of librealsense the kernel should not matter as librealsense bypasses the kernel.当使用 librealsense 的 libuvc 安装时，内核应该无关紧要，因为 librealsense 会绕过内核。
> 
> Are you using the official 1 meter USB cable supplied with the camera or a longer USB cable of your own choice?您使用相机随附的官方 1 米 USB 线还是您自己选择的更长 USB 线？
> 
> What is the computer that you are using your D435i on, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc)请问您使用 D435i 的计算机是什么（Raspberry Pi、Nvidia Jetson、台式机/笔记本电脑等）
> 
> Is the cable plugged into a built-in USB port on the computer or a USB interface such as a hub or PCIe?电缆是否插入计算机上的内置 USB 端口或 USB 接口（例如集线器或 PCIe）？

I am using official cable,and plugged USB3.0 port,but ubuntu shows it only used USB2.0 port,
It is using a development board called "Shaolinpai" which designed by sophon company

MartyG-RealSense, 2024-05-13:
Are you using Sophon's SLKY01 Shaolin Pai board at the link below?

https://en.sophgo.com/product/introduce/slky01.html

It appears to use a Sophon chip called BM1684 as its CPU.  The librealsense SDK is intended for use with x86 / x64 CPUs (like those in desktop and laptop PCs) or Arm CPUs (such as those in Raspberry Pi and Nvidia Jetson boards).  If the BM1684 CPU is neither x86/x64 or Arm chip architecture then librealsense may not know how to make use of it.  It looks as though the BM1684 chip was designed to specialize in processing deep learning activities rather than general computing activities.

DiopWang, 2024-05-13:
> Are you using Sophon's SLKY01 Shaolin Pai board at the link below?您是否使用下面链接中的 Sophon SLKY01 少林牌板？
> 
> https://en.sophgo.com/product/introduce/slky01.html
> 
> It appears to use a Sophon chip called BM1684 as its CPU. The librealsense SDK is intended for use with x86 / x64 CPUs (like those in desktop and laptop PCs) or Arm CPUs (such as those in Raspberry Pi and Nvidia Jetson boards). If the BM1684 CPU is neither x86/x64 or Arm chip architecture then librealsense may not know how to make use of it. It looks as though the BM1684 chip was designed to specialize in processing deep learning activities rather than general computing activities.它似乎使用名为 BM1684 的 Sophon 芯片作为 CPU。 librealsense SDK 适用于 x86 / x64 CPU（例如台式机和笔记本电脑中的 CPU）或 Arm CPU（例如 Raspberry Pi 和 Nvidia Jetson 主板中的 CPU）。如果 BM1684 CPU 既不是 x86/x64 也不是 Arm 芯片架构，那么 librealsense 可能不知道如何使用它。看起来 BM1684 芯片的设计目的是专门处理深度学习活动，而不是一般计算活动。

yes,it is a TPU chip.Thanks for your help

closed 2024-05-26 by MartyG-RealSense