realsenseai/librealsense #12926: SOFTWARE DOESN'T RECOGNIZE REALSENSE SR300 CAMERA

Scotty866211, 2024-05-13:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Please provide a description of the problem
Quite simply, the camera is not seen by the software. I have struggled for weeks to locate and install all of the needed drivers only to run into this issue. I am using the Intel Realsense viewer v2.50.0 and it simply doesn't see the camera when plugged in. Am I not using the correct software for this camera? The original software recommended by 3D Systems doesn't recognize it either.

MartyG-RealSense, 2024-05-13:
Can you check the camera privacy settings in Windows please to ensure that permission for applications to access the camera is set to **On**.  Instructions for doing so can be found at 
https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

Scotty866211, 2024-05-13:
I have checked the settings you mentioned and both were set to "on". I have also tried multiple fixes recommended on various forums to no avail as I have been working on this off and on for weeks now. I think it is fair to assume that I have already tried the obvious ... things like "please look at power button on your computer to verify it is on" are not going to be helpful or appreciated. Nothing personal, I am just very frustrated. This morning after requesting assistance from you but before I heard back, I tried another "recommended fix" and seem to have lost most of my drivers for the camera. I had the RGB, depth, camera, virtual coms and bus enumerator drivers (sorry for abbreviating, I'm sure you know the drivers I mean) but now I have lost the virtual and bus drivers again and can't seem to figure out how to get them back. I have to wonder if this damned scanner is worth the effort? I got the scanner with a 3d printer that I purchased second hand and figured it would be easier in creating 3d object for reproduction than actually having to learn an entire cad program. I am now questioning that decision. Not ready to give up just yet....but it's tempting.
Scott

MartyG-RealSense, 2024-05-14:
As you mention "scanner", may I confirm whether or not your device is actually a **Sense 2**, a handheld scanner that Intel did not produce but was based on Intel's SR300 technology and used Intel's SR300 camera drivers?

[image]

Scotty866211, 2024-05-14:
> As you mention "scanner", may I confirm whether or not your device is actually a **Sense 2**, a handheld scanner that Intel did not produce but was based on Intel's SR300 technology and used Intel's [...]

You are correct, that is exactly the item I am in need of help with. Is this a problem?
Scott

Scotty866211, 2024-05-14:
> Intel does not provide technical support for RealSense-equipped hardware that was manufactured by another company as it is the responsibility of the manufacturer to provide that support.
> 
> As the Sensor 2 uses SR300 drivers though, based on your kindly provided information above it sounds as though the missing drivers are likely to be impairing your camera. The SR300 is especially likely to not be operable if the Virtual Driver is absent.
> 
> Could you open your Windows 'Device Manager' interface please and check in its **Camera** and **Imaging Devices** categories to see if any SR300 drivers are listed there? There should be a Depth, RGB and Virtual driver. They may be spread across both categories or all located in one of them.
> 
> If you have some of the drivers there but not all, you could try right-clicking on each SR300 driver and selecting **Uninstall Device** from the menu. When all SR300 drivers listed in Camera or Imaging Devices has been removed, unplug the camera, wait a couple of seconds and re-insert the camera. If you have an active internet connection then all three drivers should automatically re-install and be displayed in the Device Manager.

I have tried that several times to no avail. The most recent attempt has some how now lost the the virtual driver and I can't seem to find it. Before that happened I did right click on each of them and was told by device manager that the device was working properly. Don't really know where to go from here. I thought I had made real progress when I was finally able to locate and install all of the drivers. But even once I'd done that the various software packages I tried refused to see the camera/scanner.
Scott

MartyG-RealSense, 2024-05-14:
In the 2.50.0 Viewer, you could try clicking the **Add Source** button at the top of the side-panel (which can list non-RealSense cameras such as webcams) and see if at least its RGB camera is listed and selectable as a Platform Camera.

The link below suggests some software that may be compatible with Sense 2, such as **Skanect.**

https://www.reddit.com/r/3DScanning/comments/h8fxjj/3d_systems_sense_2_software/

A download of the official **UnitySense** Windows 10 software for the scanner can be launched in your browser by clicking the link below:

https://s3.amazonaws.com/dl.3dsystems.com/binaries/support/sense-scanner/UnitySenseSetup-2.0.321.exe

MartyG-RealSense, 2024-05-24:
This product is very old, dating back to 2016.  Even with Intel's official SR300 camera the software that made use of it became increasingly incompatible with Windows 10 as time went on and Windows 10 continued to be updated whilst the software that used the camera was no longer updated to keep up with Windows changes.

Scotty866211, 2024-05-25:
> This product is very old, dating back to 2016. Even with Intel's official SR300 camera the software that made use of it became increasingly incompatible with Windows 10 as time went on and Windows [...]

That being said, what operating system WAS this camera compatible with? Windows 7? Windows XP?

MartyG-RealSense, 2024-05-25:
Windows 8 and Windows 10.  In the 2016 era that Sense 2 was released in, the Windows 10 version was Build 14393, codenamed Anniversary Update.

Scotty866211, 2024-06-03:
> Hi @Scotty866211 Do you require further assistance with this case, please? Thanks!

I have downloaded the early version of Windows 10 recommended by you earlier. I have managed to install that version onto my labtop but am now once again having difficulty aquiring all four of the drivers. The virtual com driver seems to be alluding me once again... I will post my progress in the next day or three.

MartyG-RealSense, 2024-06-04:
Okay, thanks very much for the update.  I look forward to your next report.  Good luck!

closed 2024-06-16 by MartyG-RealSense