realsenseai/librealsense #12930: RealSense BGRA Buffer Doesn't Work with ZXing

ffreality, 2024-05-14:
Hi, I am the creator of Unofficial Unreal Engine 5 plugin for Intel RealSense.
I can get BGRA buffer on FRunnableThread and generate UTexture2D on game thread in UE5 without problem but I would like Integrate ZXing library to decode visible QR codes.

If I create a texture and read buffer from it with render thread, everything works.
But this is expensive operation. I just want to directly feed RS stream to ZXing with FRunnableThread. So, it will be much more performance optimized.

But ZXing doesn't decode that direct buffer.
I will also open this issue on ZXing, too but I am curious what can be the problem ?

Btw. OpenCV works with direct feed but we can't use it on Android platform because it gives a typedef clash error.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D455| 
| Operating System & Version |W11 and Android| 
| Platform                                 |W11 and Android|
| SDK Version                            |2.54.1| 
| Language                            |C/C++| 
| Segment			|Smartphone, VR, AR|

MartyG-RealSense, 2024-05-14:
Hi @ffreality  There is a mention of the development of the RealSense UE5 wrapper and ZXing at the January 2024 Unreal forum link below.  Is this posting by you, please?

https://forums.unrealengine.com/t/image-tracking-in-vr-passthrough/1623474/2

If the librealsense2 SDK is built from source code on Windows with CMake with the build flag **-DFORCE_RSUSB_BACKEND=TRUE** then the SDK's backend should be based on a WinUSB rewrite of the UVC protocol (the SDK's backend on Linux) instead of the default backend on Windows, which is Microsoft Media Foundation.  This may make setup of the UE5 wrapper though and I cannot guarantee that it will make a difference to ZXing will be able to decode directly.

ffreality, 2024-05-14:
Yes, I was the one who answering it.
Couple months ago before the question asked, I could read A4 sized QR codes from correct angle but right now I can't read with direct buffer. On the other hand, if read buffer from texture, decoding won't problem.

Thank you very much for your suggestion. I want to update RealSense library as well. I can also try it.

ffreality, 2024-05-14:
@MartyG-RealSense I updated the library. 

**Good thing:**
I downloaded released WIN10 SDK and everything works correctly including ZXing.

**Bad Thing:**
CMake description said that `-DFORCE_RSUSB_BACKEND=TRUE` was mandatory for Win builds and also It asked `LibUsb` library (include and lib.) I built and pointed them. Both RealSense and LibUsb gave no error but right now everything is broken.
I can't even list my attached D455
(If I use my custom built)

These are my CMake Config
[image]
[image]

MartyG-RealSense, 2024-05-14:
I do not see the flag **CMAKE_BUILD_TYPE** in the lists.  This would typically be used when building from source with CMake in Linux and is given the status **Release**.  If the SDK is not set to build in Release mode then by default it builds in Debug mode for SDK development instead.

ffreality, 2024-05-14:
I set it as `Release` on Visual Studio 2022.

closed 2024-05-14 by ffreality