realsenseai/librealsense #12931: Frame didn't arrive within 5000 on macOS

webbershaw, 2024-05-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D420 | 
| Firmware Version                   |  | 
| Operating System & Version |   macOS 14.4  | 
| Kernel Version (Linux Only)    |                                 | 
| Platform                                 | MacBook Pro M1 late 2020 |
| SDK Version                            |                        | 
| Language                            |  Python 3.10                      | 
| Segment			|                      | 

### Issue Description
The code runs well in any Windows but can't run in my mac.

It used to be fine to use it in my mac but no longer available since a failed firmware update using realsense-viewer

The original problem is:
(base) xxx@MacBook-Pro test % sudo python3 realsense_test.py
Traceback (most recent call last):
  File "/Users/xxx/Downloads/CaremaManagement/CaremaManagement/test/realsense_test.py", line 38, in <module>
    profile = pipeline.start(config)
RuntimeError: 
Failed to resolve the request: 
        Format: BGR8, width: 640, height: 480
Into:
failed to set power state

Then I googled the problem and someone said they solve the problem by brew upgrade/reinstall libusb librealsense.
After I reinstalled libusb and librealsense the problem became:
Traceback (most recent call last):
  File "/Users/xxx/Downloads/CaremaManagement/CaremaManagement/test/realsense_test.py", line 63, in <module>
    frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000

I tried to use realsense viewer to see if it's Ok with that but realsense viewer just crashed if camera connected upon i open it.
Below is what Problem Report of macOS said:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               realsense-viewer [70540]
Path:                  /opt/homebrew/*/realsense-viewer
Identifier:            realsense-viewer
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [70539]
Responsible:           Terminal [1082]
User ID:               0

Date/Time:             2024-05-15 15:55:00.7316 +0800
OS Version:            macOS 14.4.1 (23E224)
Report Version:        12
Anonymous UUID:        118CEFF8-1233-072F-052F-DE00AD327390

Sleep/Wake UUID:       EEB29B4D-3151-4665-B10D-DD2EA4982348

Time Awake Since Boot: 150000 seconds
Time Since Wake:       2098 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000028
Exception Codes:       0x0000000000000001, 0x0000000000000028

[... 408 lines cut ...]
}
}

Model: MacBookPro17,1, BootROM 10151.101.3, proc 8:4:4 processors, 16 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Display: TE24-10, 1920 x 1080 (1080p FHD - Full High Definition), MirrorOff, Online
Display: Duet, 4096 x 2304, MirrorOff, Online
Memory Module: LPDDR4, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Jan 13 2024 03:06:59 version 18.20.440.9.7.8.167 FWID 01-7be24857
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB3.1 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB 2.0 Hub
USB Device: USB2.1 Hub
USB Device: USB C Video Adaptor
USB Device: USB31Bus
USB Device: Intel(R) RealSense(TM) Depth Camera 415
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

I'd be appreciate it if you'd help me with this issue. THX so much!

MartyG-RealSense, 2024-05-15:
Hi @webbershaw  If the camera works fine on Windows then there likely is not a problem with the firmware (which is stored inside the camera hardware and not on the computer).

How long has it been since you last used RealSense successfully on Mac, please?  The RealSense SDK frequently has problems with MacOS Ventura and Sonoma. So if you had not used RealSense since updating to Sonoma, that could explain why the camera worked before but does not work now on Mac.

webbershaw, 2024-05-15:
Thanks! I can hardly recall when i upgraded to Sonoma but i do agree that it might relate to macOS update. Last time I successfully use it was about 4 month ago. 2 month ago it worked with my code only 1 times in about every 10 attempts so i have to run my code over and over again until it work. Today i realized that it completely doesn't work no matter how i tried.

What do you think I shall do

MartyG-RealSense, 2024-05-15:
There is not a clear solution for RealSense installation problems with Ventura and Sonoma at present, though a couple of RealSense users have reported that inputting the two commands below worked for them.

```
brew install libusb
brew install librealsense
```

webbershaw, 2024-05-16:
Thanks anyway! I'll try.

closed 2024-06-03 by MartyG-RealSense