realsenseai/librealsense #12932: Android App Freeze At Second Launch

ffreality, 2024-05-15:
### Issue Description
Hi, I built librealsense2.so for Android, integrated to UE5, packaged app as APK and installed to my Android device (Samsung Galaxy S23). I can open app for the first time and use it without problem. But after closing it, I can't open it again without restarting my Android device.

Unreal Engine just stuck at splash screen.

In order to test it, I set some delay before running actual C/C++ codes to get device list, device and stream.
I found out that RsContext initialization has problem. I am not Java programmer. 
So, I could did some mistakes. I think I need somehow close or destruct RsContext at app close.

I call `RealSense_Destruct` with JNI at Event Destruct but it doesn't help at all.

This is my plugin and it's Java side.
https://github.com/FF-Plugins-Active/FF_RealSense/blob/main/Source/FF_RealSense/Private/Java/FF_RealSense.java

Do you have any suggestion ?

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Android  | 
| Platform                                 | Android  |
| SDK Version                            | v2.55.1 | 
| Language                            | C/C++ and Java | 
| Segment			|  Smartphone / AR / VR |

MartyG-RealSense, 2024-05-15:
Hi @ffreality  My research of the RealSense Android wrapper's Java source code indicates that when a Close void is used, the instruction `nRelease (mHandle);` is usually used.  For example:

```
@Override
public void close() {
nRelease (mHandle);
}
```

Here is an example of using this mechanism in the Android wrapper's source code.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/librealsense/src/main/java/com/intel/realsense/librealsense/Frame.java#L62-L66

ffreality, 2024-05-15:
https://github.com/IntelRealSense/librealsense/blob/ff8a9fb213ec1227394de4060743b0ed61171985/wrappers/android/librealsense/src/main/java/com/intel/realsense/librealsense/RsContext.java#L50

But shouldn't this close RsContext ?

MartyG-RealSense, 2024-05-15:
If an app is closed down then RsContext should be able to be initialized again on the next launch with `RsContext.init` like in the official RealSense Java example program.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/readme.md#on-create

ffreality, 2024-05-15:
Yes but it somehow affect whole engine's activity. I can't even start my app for second time after I use RealSense

ffreality, 2024-05-15:
Okay something weird happened. Samsung Galaxy S23 Android 14 has this problem but on Oculus/Meta Quest Pro (Android 12) I can start app as much as I want. There is no freeze after first usages.

I think there is problem with Android 14. What do you think ?

MartyG-RealSense, 2024-05-16:
There is precedent from 2021 for incompatibility between the RealSense Android wrapper and the chipset in some Samsung models, as described at https://github.com/IntelRealSense/librealsense/issues/8442#issuecomment-785798027

In the current latest 2.55.1 SDK, support for Android 12 was added.  Official Android 13 and 14 support has not yet been implemented in the SDK.  Please see https://github.com/IntelRealSense/librealsense/pull/10965 for notes about the Android 12 support addition.

There have been a couple of past reports of unofficial Android 13 use.  For one user it worked on their Android 13 device and for the other it did not work.

ffreality, 2024-05-16:
I packaged for SDK API Level 30 and tried to run on Samsung Android 14. I got second startup stuck problem. It's good to know that it is Samsung related.

Btw. if I package project with API level 32, I got crash at when I try to run Rs_Context init function. But it can be related with Samsung, too.

When Meta release 5.4.1 version of their specialized branch, I will try to package a mixed reality project with API level 32 and look logs.

MartyG-RealSense, 2024-05-16:
Thanks very much for the information!

ffreality, 2024-05-17:
Okay, problem was about Samsung.
I packaged project with SDK API Level 32 and deployed to the Quest Pro. Works like a charm with multiple usage.

closed 2024-05-17 by ffreality

MartyG-RealSense, 2024-05-17:
It's great to hear that you were successful.  Thanks so much for your ongoing work on the UE5 plugin!