realsenseai/librealsense #12934: Cannot link to new camera

AmrGaballah29, 2024-05-15:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2024-05-16:
Hi @AmrGaballah29  There was a past case at https://github.com/IntelRealSense/librealsense/issues/11338 where this error occurred when trying to access the RGB camera from the **Add Source** option of the RealSense Viewer tool.  Are you using the Add Source button in the Viewer when the error happens, please?

AmrGaballah29, 2024-05-16:
Hi
Thank you for your reply. I think my case is related to the error described in this post https://github.com/IntelRealSense/librealsense/issues/6254#issuecomment-618688110

The viewer  always detected the camera to be connected to a usb2 port although it is connected to a USB 3 port

MartyG-RealSense, 2024-05-16:
The chances of the camera being detected as USB 3.2 instead of 2.1 increase if the connector is inserted into the USB port with a quick, firm motion instead of a slow insertion.

Also, if you are using your own choice of USB cable instead of the official cable then ensure that the chosen cable is a USB 3 cable.  If a USB hub is being used, ensure that it is a USB 3 hub and not a USB 2 one.

closed 2024-06-03 by MartyG-RealSense