realsenseai/librealsense #12940: Realsense-viewer does not detect L515 camera on ubuntu 20.04

multiplexcuriosus, 2024-05-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 
| Firmware Version                   | (Open RealSense Viewer | 
| Operating System & Version | Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15.0-107                                        | 
| Platform                                 | PC |
| SDK Version                            |  v2.55.1                          | 
| Language                            |  -                         | 
| Segment			|  Vision                       | 

### Issue Description
Hi
I have the following weird situation. On my laptop (Ubuntu 20.04) I installed the RS SDK v2.54.2 and the realsense-viewer works without any issues.

On my PC however, the same camera, with the same OS (Ubuntu 20.04) is not detected by the realsense-viewer.

On my PC, I have:
- tried the installation process with the kernel patch
- tried the installatoin process without the kernal patch
- checked that the output of `dmesg -wT` show the same few lines concerning the usb detection of the camera on the laptop and on the PC.
- reinstalled ubuntu 
but nothing made the camera show up.

I installed the SDK on the laptop a couple of weeks ago, back then it was still version v2.54.2. Hoping that the issue had to do with the new version I tried to reinstall the SDK on my PC with version v2.54.2 but failed, since I was not linux-competent enough to handle the dependency issues which arose. 

Can anyone give me advice on how to proceed?

My goal in the end is to use the python API for the SDK, which on the laptop already works. 

Thank you!

MartyG-RealSense, 2024-05-18:
Hi @multiplexcuriosus  Support for the L515 has been removed from the current latest 2.55.1 version of the RealSense SDK as L515 is a retired camera model.  You will need to use 2.54.2 on your PC too.  Sincere apologies for the inconvenience.

multiplexcuriosus, 2024-05-18:
Hi @MartyG-RealSense 

Thank you a lot for your answer!
Now I at least know what the problem was.
No need to apologize, it's not your fault I didn't check if the camera version is still supported. 

Can you point me to a resource or briefly summarize the steps necessary to install the 2.54.2 version? As mentioned I did not succeed in doing that the first time I tried.

MartyG-RealSense, 2024-05-18:
You can find the source code for a particular librealsense version from the SDK's **Releases** page and compile it with CMake using instructions at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

multiplexcuriosus, 2024-05-18:
Perfect, now it works. Thank you so much!

multiplexcuriosus, 2024-05-18:
Before we close the issue, one more question.

The realsense-viewer works no, but the python API pyrealsense2 still seems to have problems.

The first few lines of the python script I am trying to run are:

`import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()

# Get device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))`

which results in the error:

`Traceback (most recent call last):
  File "/home/jau/Documents/SP/shelf_localization/myScripts/rs_tuts.py", line 18, in <module>
    pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected`

This is the same error I got when I was still using 2.55.1.

I already reinstalled pyrealsense2 to no avail.

Do you have a suggestion on how to get pyrealsense2 to work again?

multiplexcuriosus, 2024-05-18:
Never mind, I just had to downgrade the 'pyrealsense2' as well. All is working well. You may close the issue if you want.

MartyG-RealSense, 2024-05-18:
It's great to hear that you succeeded both with the SDK and pyrealsense2.  As you suggested, I will close the issue. Thanks!

closed 2024-05-18 by MartyG-RealSense