realsenseai/librealsense #12941: window 10 can't install realsense viewer

fendiling, 2024-05-20:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_create_context_ex(api_version:25501, json_settings:{):
    failed to convert special folder: errno=42

```

MartyG-RealSense, 2024-05-20:
Hi @fendiling  I hope that the information about this error at https://github.com/IntelRealSense/librealsense/issues/12910 will be helpful to you.

closed 2024-06-03 by MartyG-RealSense

MartyG-RealSense, 2024-06-14:
My Intel RealSense colleagues have provided further advice about the errno=42 error at https://github.com/IntelRealSense/librealsense/issues/13023#issuecomment-2167504813

AviaAv, 2024-12-29:
Hey @fendiling 

We've recently fixed an issue about the "errno=42" error, which would happen if the user's name on that machine contained Unicode characters. Could you please check the development branch to see if this fix resolves the issue?

The PR with the fix is: #13597 

The development branch can be found here: https://github.com/IntelRealSense/librealsense/tree/development