realsenseai/librealsense #12947: How to correct the distortion in D405, I found that when the object is too large or not in the center, the x,y coordinates of the detected point are not quite correct

ybefe, 2024-05-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D400| 
| Firmware Version                   | | 
| Operating System & Version |   Win(10) | 
| Kernel Version (Linux Only)    |                                | 
| Platform                                 | PC |
| SDK Version                            |                    | 
| Language                            |  python                         | 
| Segment			|                           | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ybefe, 2024-05-22:
I did use get_distance() to get the 3D coordinates of the point, so how do I fix this? @MartyG-RealSense

MartyG-RealSense, 2024-05-22:
Are you performing depth to color alignment before calling the get_distance() instruction?

MartyG-RealSense, 2024-05-22:
Yes, it is correct to use depth to color alignment.  Either the aligned intrinsics or the color intrinsics should be used, as using the depth intrinsics can cause the inaccuracy at the outer edges of the image that you experienced.

ybefe, 2024-05-22:
So how to use the aligned intrinsics or the color intrinsics

MartyG-RealSense, 2024-05-22:
A Python get_distance() script at the link below that uses color intrinsics should be a helpful reference.

https://support.intelrealsense.com/hc/en-us/community/posts/23541262437011/comments/23566066750355

ybefe, 2024-05-22:
align = rs.align(rs.stream.color)->align = rs.align(rs.stream.depth)
dis, camera_coordinate = get_3d_camera_coordinate(depth_pixel, aligned_depth_frame, depth_intrin)->
dis, camera_coordinate = get_3d_camera_coordinate(depth_pixel, aligned_depth_frame, color_intrin)
It's probably just using color alignment, and then using the camera parameters to find the 3D coordinates,right?

MartyG-RealSense, 2024-05-22:
When using **align_to()**, the alignment calculations are handled by a software routine in the SDK called the **align processing block** that performs automated adjustments, such as adjusting for differences in resolution between the streams or resizing the depth's field of view size to match the field of view size of color.

Things are a little more complicated on the D405 model, as it does not have a dedicated RGB sensor and its color is provided by the depth sensor and passed through an Image Signal Processor (ISP) chip to produce the final color image.  The  color intrinsics should still be different from the depth intrinsics though.

ybefe, 2024-05-22:
So I should use align = rs.align(rs.stream.color) and  color_intrin ,right?

MartyG-RealSense, 2024-05-22:
Yes, align to color and use color intrinsics.

ybefe, 2024-05-22:
Thank you very much!

MartyG-RealSense, 2024-05-28:
Hi @ybefe  Do you require further assistance with this case, please?  Thanks!

ybefe, 2024-05-30:
camera_coordinate = rs.rs2_deproject_pixel_to_point(color_intrin, depth_pixel, dis)
With the D405, should I use rs2_deproject_pixel_to_point or rs2_project_pixel_to_point to handle the distortion?

MartyG-RealSense, 2024-05-30:
Do you mean **rs2_project_point_to_pixel** please?  If you do, then this instruction is for converting 3D real-world values back into 2D pixel values after having first converted 2D to 3D with **rs2_deproject_pixel_to_point**.  

In a project where 3D values are being converted back to 2D though, there may be a small amount of inaccuracy caused by the depth-color alignment.  If you are converting 3D XYZ values back to 2D XY values using rs2_project_point_to_pixel then there is not much that can be done to improve accuracy if you are already using the color intrinsics.

A more accurate method is to generate a depth-color aligned pointcloud with **map_to()** and **pc.calculate()** and then store the coordinates in **vertices** that XYZ values can be retrieved from and displayed.

closed 2024-06-14 by MartyG-RealSense