realsenseai/librealsense #12951: patch-realsense-ubuntu-lts-hwe.sh Unsupported kernel version 6.5.0-35-generic

amburkoff, 2024-05-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435, T265| 
| Firmware Version                   |  | 
| Operating System & Version |   Linux (Ubuntu 22.04)   | 
| Kernel Version (Linux Only)    |  6.5.0-35-generic                                       | 
| Platform                                 | PC |
| SDK Version                            | master in github and 2.53.1 }                          | 
| Language                            |  python                       | 
| Segment			|  Robot                         | 

### Issue Description
`./scripts/patch-realsense-ubuntu-lts-hwe.sh`

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
bc is already the newest version (1.07.1-3build1).
bc set to manually installed.
build-essential is already the newest version (12.9ubuntu3).
build-essential set to manually installed.
git is already the newest version (1:2.34.1-1ubuntu1.10).
The following packages were automatically installed and are no longer required:
  linux-headers-5.15.0-60 linux-headers-5.15.0-60-generic ros-humble-diagnostic-updater
  ros-humble-xacro
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  linux-headers-5.15.0-107 linux-headers-5.15.0-107-generic
The following NEW packages will be installed:
  linux-headers-5.15.0-107 linux-headers-5.15.0-107-generic linux-headers-generic
0 upgraded, 3 newly installed, 0 to remove and 104 not upgraded.
Need to get 15.2 MB of archives.
After this operation, 103 MB of additional disk space will be used.
Get:1 http://ru.archive.ubuntu.com/ubuntu jammy-updates/main amd64 linux-headers-5.15.0-107 all 5.15.0-107.117 [12.3 MB]
Get:2 http://ru.archive.ubuntu.com/ubuntu jammy-updates/main amd64 linux-headers-5.15.0-107-generic amd64 5.15.0-107.117 [2,887 kB]
Get:3 http://ru.archive.ubuntu.com/ubuntu jammy-updates/main amd64 linux-headers-generic amd64 5.15.0.107.107 [2,354 B]
Fetched 15.2 MB in 35s (435 kB/s)                                                                       
Selecting previously unselected package linux-headers-5.15.0-107.
(Reading database ... 470234 files and directories currently installed.)
Preparing to unpack .../linux-headers-5.15.0-107_5.15.0-107.117_all.deb ...
Unpacking linux-headers-5.15.0-107 (5.15.0-107.117) ...
Selecting previously unselected package linux-headers-5.15.0-107-generic.
Preparing to unpack .../linux-headers-5.15.0-107-generic_5.15.0-107.117_amd64.deb ...
Unpacking linux-headers-5.15.0-107-generic (5.15.0-107.117) ...
Selecting previously unselected package linux-headers-generic.
Preparing to unpack .../linux-headers-generic_5.15.0.107.107_amd64.deb ...
Unpacking linux-headers-generic (5.15.0.107.107) ...
Setting up linux-headers-5.15.0-107 (5.15.0-107.117) ...
Setting up linux-headers-5.15.0-107-generic (5.15.0-107.117) ...
/etc/kernel/header_postinst.d/dkms:
 * dkms: running auto installation service for kernel 5.15.0-107-generic

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
unset ARCH; [ ! -h /usr/bin/cc ] && export CC=/usr/bin/gcc; env NV_VERBOSE=1 'make' -j8 NV_EXCLUDE_BUILD_MODULES='' KERNEL_UNAME=5.15.0-107-generic IGNORE_XEN_PRESENCE=1 IGNORE_CC_MISMATCH=1 SYSSRC=/lib/modules/5.15.0-107-generic/build LD=/usr/bin/ld.bfd CONFIG_X86_KERNEL_IBT= modules...........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.0-107-generic (x86_64)
Consult /var/lib/dkms/nvidia/525.147.05/build/make.log for more information.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.15.0-107-generic -C /lib/modules/5.15.0-107-generic/build M=/var/lib/dkms/virtualbox/6.1.50/build.....
Signing module:
 - /var/lib/dkms/virtualbox/6.1.50/5.15.0-107-generic/x86_64/module/vboxnetflt.ko
 - /var/lib/dkms/virtualbox/6.1.50/5.15.0-107-generic/x86_64/module/vboxnetadp.ko
 - /var/lib/dkms/virtualbox/6.1.50/5.15.0-107-generic/x86_64/module/vboxdrv.ko
Secure Boot not enabled on this system.
cleaning build area...

vboxdrv.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-107-generic/updates/dkms/

vboxnetadp.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-107-generic/updates/dkms/

vboxnetflt.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-107-generic/updates/dkms/

depmod...
   ...done.
Setting up linux-headers-generic (5.15.0.107.107) ...
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Unsupported kernel version 6.5.0-35-generic . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only
```
I tried the script from versions branch master and 2.53.1, but the error is always: Unsupported kernel version 6.5.0-35-generic . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only

```
uname -r
6.5.0-35-generic
```

Please provide a working script and instructions for a patch to compile the sdk version from sources that support camera metadata containing all timestamps: backend_timestamp, frame_timestamp, sensor_timestamp, time_of_arrival for ubuntu 22.04 kernel 6.5.0-35-generic

MartyG-RealSense, 2024-05-21:
Hi @amburkoff  Support for kernel 6.5 was added to patch-realsense-ubuntu-lts-hwe.sh in SDK version 2.55.1 (the current master at the time of writing this) - please see https://github.com/IntelRealSense/librealsense/commit/d54529b66c6abafe8ceae13caaf76e0a32101e87

If you are still experiencing problems with kernel 6.5 in the 2.55.1 version though then there is the option available of building the SDK from source code with CMake with the flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction.  For example:

`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

When the SDK is built with RSUSB = true then the kernel is bypassed, meaning that the SDK build is not dependent on Linux versions or kernel versions and does not need to have a kernel patch script applied.  Hardware metadata support is also enabled.  So the method will work with 2.53.1 too as it is not dependent on the patch script.

amburkoff, 2024-05-23:
Thanks for the detailed instructions, it worked. But not all metadata is available in the t265 camera, compared to the d435, for example, there is no sensor time. Tell me how does the kernel patch affect the time of frames.get_timestamp() or does it exist, regardless of the kernel patch, with the data frames.get_frame_metadata(rs.frame_metadata_value.sensor_timestamp) and frames.get_frame_metadata(rs.frame_metadata_value.frame_timestamp)?
[image]

MartyG-RealSense, 2024-05-23:
My understanding from https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803 is that Frame Timestamp is a **hardware metadata timestamp** and is dependent on support for hardware metadata that is provided either by patching the kernel or building the SDK in RSUSB = true mode.

Time of Arrival is a **software timestamp** based on the computer's host clock and so is not dependent on support for hardware metadata being enabled.

MartyG-RealSense, 2024-05-30:
Hi @amburkoff  Do you require further assistance with this case, please?  Thanks!

amburkoff, 2024-06-04:
thanks for your help, the main issue is resolved

MartyG-RealSense, 2024-06-04:
You are very welcome.  I'm pleased that I could help.  :)

closed 2024-06-10 by MartyG-RealSense