realsenseai/librealsense #12954: E: Unable to locate package librealsense2-dkms in ubuntu 20.04.5 LTS version.

kotran88, 2024-05-22:
D455
Ubuntu 20.04.5 LTS
Jetson nano

I just follow link
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

and the problem is that when I type 
sudo apt-get install librealsense2-dkms
then It cause error 

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package librealsense2-dkms

how can I solve problem ? 
I already connected D455 to jetson nano with usb

MartyG-RealSense, 2024-05-22:
Hi @kotran88  The packages on the **distribution_linux.md** instruction page are not designed for Arm-based devices such as Nvidia Jetson.  Intel provide dedicated package installation instructions for Jetson at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

kotran88, 2024-05-23:
@MartyG-RealSense 
Thank you for quick response!
with new link you provide with, I also face with new problem 
error log : 
```
sudo apt-get install librealsense2-utils
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 librealsense2-utils : Depends: libgtk-3-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
er@er:~$ 
```

what is weird is that I try to remove libgtk-3-dev with command 
```
sudo apt-get remove --purge libgtk-3-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'libgtk-3-dev' is not installed, so not removed

```
it says there is no installed libgtk-3-dev,, 

how may i can install it or try with new method ?

MartyG-RealSense, 2024-05-23:
Please try the instruction below to install libgtk-3-dev.

`sudo apt install libgtk-3-dev`

kotran88, 2024-05-23:
@MartyG-RealSense 
wow so difficult to get realsense view...
I did try install libgtk-3-dev but this time cause different error 
The following packages have unmet dependencies:
 libgtk-3-dev : Depends: libatk-bridge2.0-dev but it is not going to be installed
                Depends: libepoxy-dev (>= 1.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

should I go over every step  installing unmet one? 

I go over sources.list 
and at the last, 

deb https://librealsense.intel.com/Debian/apt-repo focal main
# deb-src https://librealsense.intel.com/Debian/apt-repo focal main

is it normal? 
someone says I should add below to install libgtk-3-dev...
deb http://archive.ubuntu.com/ubuntu focal-updates main

MartyG-RealSense, 2024-05-23:
Another RealSense user who experienced this problem shared the solution that worked for them at https://github.com/IntelRealSense/realsense-ros/issues/2512#issuecomment-1287235695

MartyG-RealSense, 2024-06-06:
Case closed due to no further comments received.

closed 2024-06-06 by MartyG-RealSense

gfx73, 2024-09-16:
Hi @MartyG-RealSense !

https://github.com/IntelRealSense/librealsense/issues/12954#issuecomment-2125114030

Do I understand correctly that librealsense installation with dpkg on arm based system is only possible with CUDA compatible devices? 

My device does not have CUDA GPU, so this [instruction](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) is not applicable for me

MartyG-RealSense, 2024-09-16:
Hi @gfx73  You do not need CUDA to build the librealsense SDK on Arm devices.  However, Nvidia Jetson is the only Arm-based system that has packages provided.    

For other Arm devices (for example, Raspberry Pi), librealsense should be built from source code.  An easy way to do so is to use the **libuvc backend** build script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md