realsenseai/librealsense #12955: L515 camera   significant misalignment on the left side

DongLZY, 2024-05-23:
When using the Intel RealSense L515 camera model and the pyrealsense2 library in Python to align the depth map with the color map, I have noticed that the right side aligns properly, but there is a significant misalignment on the left side, preventing a complete alignment. This issue has been tested on both Windows 11 and Ubuntu systems, with consistent results.

Required Info 	Value
Camera Model 	Intel RealSense L515
Operating System & Version 	Windows 11 and Ubuntu22.04
Platform 	PC
SDK Version 	pyrealsense2  2.54.2.5684
Language 	Python

MartyG-RealSense, 2024-05-23:
Hi @DongLZY  Did you test with the RealSense SDK's **align_depth2color.py** alignment example program?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

If you did use this script then support for L515 was removed from it when overall support in the SDK for L515 was removed in SDK 2.55.1.  There is an older version of the example at https://github.com/IntelRealSense/librealsense/issues/11329#issue-1537754614 that still contains L515 support.

MartyG-RealSense, 2024-05-30:
Hi @DongLZY  Do you require further assistance with this case, please?  Thanks!

DongLZY, 2024-05-30:
Hi, thank you for your reply.
I found that I set the short range mode in the script, 
'''  preset = rs.l500_visual_preset.short_range
  depth_sensor.set_option(rs.option.visual_preset, int(preset))
'''
and in this mode, the depth map cannot be fully aligned with the color map. Is there a solution to this issue? Thank you.

DongLZY, 2024-05-31:
Setting the low_ambient_light does not affect the alignment, but it is not fully aligned in the short range mode. Is it necessary to perform some calibration operations in this mode?

MartyG-RealSense, 2024-05-31:
It should not be necessary to perform an calibration operation when using presets.

The main difference between Short Range and Low Ambient Light is that Short Range reduces the value of the Laser Power setting slightly from its maximum of 100 to '93', whilst Low Ambient Light maximizes the laser power at 100.

You could try setting the Low Ambient Light preset and then using the Python code below to set the Laser Power to '93' like the Short Range preset does in order to see whether it affects alignment.

`depth_sensor.set_option(rs.option.laser_power, 100)`

DongLZY, 2024-06-03:
Hello, I have conducted more tests. The alignment still fails towards the left side, while the right side is aligned. Could it be possible that there is a hardware issue?

MartyG-RealSense, 2024-06-03:
If alignment is correct when using the Low Ambient Light preset and only has a problem when using Short Range then that suggests that the camera hardware does not have an issue.

DongLZY, 2024-06-03:
I just tested using the Low Ambient Light preset, and the same effect occurred—it was not aligned. when not using the Short Range , the situation remains the same.

MartyG-RealSense, 2024-06-03:
Alignment is calculated in the SDK software and not on the camera hardware.  

If you installed the librealsense SDK with tools and examples included, are you able to launch the pre-built **rs-align** example program to see how alignment behaves when generated by that program?  Examples are installed to the **usr/local/bin** folder on Linux and **C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools** on Windows.

DongLZY, 2024-06-06:
Yes, I tested another L515 camera, and the same issue occurred—the left side was not aligned. I think I need to find other way to solve this problem because it's important for me to obtain the depth of finer targets.

closed 2024-06-21 by MartyG-RealSense