realsenseai/librealsense #12957: Reduced fps in low-light conditions

avadosf, 2024-05-24:
Hi, I have an intel realsense d455 camera
Noticed a decrease in fps in low light conditions, what could be the problem?
I have an assumption that the reason is the complexity of scene texture detection and because of this the processing time of each frame increases. is this correct?

MartyG-RealSense, 2024-05-24:
Hi @avadosf  If you have auto-exposure enabled then it is possible that the low lighting condition is causing the auto-exposure to change the exposure value to a range where FPS drops (since there is a relationship between the exposure value and FPS speed).

Enabling the **HDR Enabled** option (High Dynamic Range) may help, as it allows the camera to alternate between two different exposure values and therefore be able to see in both very dark and very bright lighting conditions.

Intel provide a guide to HDR at the link below.

https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras#31-hdr-with-intel-realsense-viewer

avadosf, 2024-05-24:
I am more interested in the moment of speed of depth map calculation at different degrees of illumination. And does illumination affect the speed of depth map calculation?

MartyG-RealSense, 2024-05-24:
The illumination level should not affect the speed of depth map calculation.

avadosf, 2024-05-24:
can affect the accuracy of the calculation?

avadosf, 2024-05-24:
[image]
such lighting

MartyG-RealSense, 2024-05-24:
Lighting can have an effect on accuracy, yes.  It is only going to be an issue on the depth image though, since accuracy is not a factor on the infrared and RGB streams.  In dark conditions, having the camera's IR emitter component enabled can help a lot with providing a high quality depth image as the emitter projects invisible IR light and an invisible dot pattern that is cast over objects in the scene to help the camera with analyzing the surfaces for depth.

The D455 model can stream RGB color from the left infrared sensor and so take advantage of illumination provided by the IR emitter.  The normal color image from the RGB sensor is not affected by the IR emitter though and so relies on external illumination in the scene.

You can access RGB from the infrared sensor by enabling the infrared stream and using RGB8 or BGR8 as its format instead of the infrared stream's default Y8 format.

avadosf, 2024-05-24:
Yeah, I meant the depth map. Thank you

closed 2024-05-24 by avadosf